xoseperez/espurna #2500: wrong temperature value with 100k NTC

hamed-ta, 2022-01-21:
Hello,

I use 100 NTC and a downstream resistor of 10K ohm, to read the LED board temperature, my configs and parameters are as below:

#define SENSOR_SUPPORT     1
#define NTC_SUPPORT        1
#define NTC_BETA           3950
#define NTC_R_UP           0
#define NTC_R_DOWN         10000
#define NTC_R0             100000

but what I read in 25C room temperature is about 80 Celsius, what is the problem?

here is the NTC that I use:
[image]

here is my circuit:
[image]

and what I read in room temp:

[image]

mcspr, 2022-04-30:
^ see the godbolt link above with the code reproducing the values from the sensor, and how things change depending on the rdown value (constants at the top)
shouldn't it be 100k down as well?

hamed-ta, 2022-05-02:
> ^ see the godbolt link above with the code reproducing the values from the sensor, and how things change depending on the rdown value (constants at the top) shouldn't it be 100k down as well?

No, the R down shouldn't be 100K because in that way the divided voltage would be higher than 1V which ADC can measure.

mcspr, 2022-05-02:
What voltage are you seeing right now, with 10k and with 100k?

hamed-ta, 2022-05-02:
> What voltage are you seeing right now, with 10k and with 100k?

In room temp (26 Celcius), it's about 0.39V which is correct, and the corresponding ADC value is about 416 which is correct so the problem only can be the temp calculation.

[image]

commit 1b49326e (2022-07-21): ntc: scale based on input voltage

resolves #2500
resolves #2523

- fixing default config values, make sure we take inputs into
  an account and not assume we stay in the 0.0...1.0 range
- use shelly values from tasmota analog sensor impl
- handle edgecase when both resistances are zeroed
- clearer description of available connection schemes

in theory, there is also an option of having *both* up and down
resistors at the same time; right now we simply deduce NTC position by
making it an opposite of resistance. ldrsensor implementation currently
allows to explicitly set sensor position
closed 2022-07-21 by mcspr