xoseperez/espurna #2501: add support of INA219 Current/Power Monitor sensor

hamed-ta, 2022-01-21:
Hello,

I need to add INA219 Current/Power Monitor sensor support to Espurna, it's an I2C sensor, looked at the sensors codes but it's a little time-consuming to find out from the sources how to add support of another sensor, is there any documentation for it? I need to monitor the battery voltage, current, and power that INA219 reports and then show them on the web console also show some graphics for battery and some real-time processing based on battery and current.

can you please guide me on how to do this?

Thanks.

mcspr, 2022-01-26:
Sensor model remained the same for a long time though:
- something needs to implement BaseSensor interface and be accessible from sensor.cpp (see at the top, either as include sensors/BlahSensor.h or just write it right there). 
- one needs to manually push into sensors vector (/`_sensors.push_back(new ...())`/). required methods are enforced as 'virtual ... = 0;', so those can't really be missed at build time. DummySensor would be a minimal example
- begin() is called at boot, typically we search the bus for the device and initialize it somehow via config register. see I2CSensor, ADE7953Sensor, EmonADC121Sensor. at minimum, we need to update _error = SENSOR\_ERROR\_OK and \_count to number of available measurement slots so that we actually get initialized to allowed to be read from
- type(slot) returns something MAGNITUDE\_... and value(slot) returns a numeric (double) value of the latest measurement. sensor.cpp
- each READING time, we call sensor->pre() then sensor->value(slot), pre() would be the method to read and cache i2c results in this case. webui receives data when reading ends
- each REPORT time, aggregated values are sent via api modules like mqtt, thingspeak, dcz, etc.

Basic idea would be to add begin() with I2C search & lock, then do initialization via config register or otherwise (digital pin writes, reads, etc.). After that, pre() can load voltage register(s) and calculate the rest for the output.

hamed-ta, 2022-01-26:
I made one by checking other sources, and it's working, is this in the correct format?

`#if SENSOR_SUPPORT && EMON_INA219_SUPPORT

#pragma once
/******************************************************************************
 *
 * This is a library for the INA219 Current Sensor Module
 *
 * You'll find several example sketches which should enable you to use the library.
 *
 * You are free to use it, change it or build on it. In case you like it, it would
 * be cool if you give it a star.
 *
 * If you find bugs, please inform me!
 *
 * Written by Wolfgang (Wolle) Ewald
 * https://wolles-elektronikkiste.de/en/ina219-current-and-power-sensor (English)
 * https://wolles-elektronikkiste.de/ina219 (German)
 *
 * Modified by Hamed Taheri for Espurna (https://hamed-taheri.com)
 ******************************************************************************/
#include "Arduino.h"
#include <Wire.h>
#include "BaseAnalogEmonSensor.h"
#include "I2CSensor.h"

/* registers */
#define INA219_ADDRESS 0x40
#define INA219_CONF_REG 0x00    // Configuration Register
#define INA219_SHUNT_REG 0x01   // Shunt Voltage Register
#define INA219_BUS_REG 0x02     // Bus Voltage Register
#define INA219_PWR_REG 0x03     // Power Register
#define INA219_CURRENT_REG 0x04 // Current flowing through Shunt
#define INA219_CAL_REG 0x05     // Calibration Register

/* parameters */
#define INA219_RST 0x8000

typedef enum INA219_ADC_MODE
{
    BIT_MODE_9 = 0b00000000,
    BIT_MODE_10 = 0b00000001,
    BIT_MODE_11 = 0b00000010,
    BIT_MODE_12 = 0b00000011,
    SAMPLE_MODE_2 = 0b00001001,
    SAMPLE_MODE_4 = 0b00001010,
    SAMPLE_MODE_8 = 0b00001011,
    SAMPLE_MODE_16 = 0b00001100,
    SAMPLE_MODE_32 = 0b00001101,
    SAMPLE_MODE_64 = 0b00001110,
    SAMPLE_MODE_128 = 0b00001111
} ina219AdcMode;

typedef enum INA219_MEASURE_MODE
{
    POWER_DOWN = 0b00000000,
    TRIGGERED = 0b00000011,
    ADC_OFF = 0b00000100,
    CONTINUOUS = 0b00000111,
    CONTINOUS = 0b00000111 // keeping misspelled enum for backwards compatibility.
} ina219MeasureMode;

typedef enum INA219_PGAIN
{
    PG_40 = 0x0000,
    PG_80 = 0x0800,
    PG_160 = 0x1000,
    PG_320 = 0x1800
} ina219PGain;
[... 369 lines cut ...]
    void pre()
    {
        _error = SENSOR_ERROR_OK;

        _voltage = _ina219.getBusVoltage_V();
        _current = _ina219.getCurrent_mA();
        _power = _ina219.getBusPower();

        espurna::time::blockingDelay(espurna::duration::Milliseconds(100));
    }

    double value(unsigned char index)
    {
        if (index == 0)
            return _voltage;
        if (index == 1)
            return _current;
        if (index == 2)
            return _power;
        return 0;
    }
};

#endif
`

mcspr, 2022-01-26:
Pretty much, yes.
The rest of the port would be tweaking the original code though; like using i2c.h funcs, address & rest of options; enum-classing the c style enums (or at least hiding them inside of the class{};) and minor typecast tweaks

...and using baseemonsensor so volt-current-power scaling works.

hamed-ta, 2022-01-27:
Ok, I will do so, another problem is units, how can I define other units such as mA and mW? and why the voltage value is rounded?

[image]
 
Another thing is I need to have a battery level indicator according to the battery voltage how to do so? and also how to access the sensor values in other places? I need to adjust the light and some other parameters based on the battery voltage and led board temperature?

Thanks.

mcspr, 2022-01-27:
> Ok, I will do so, another problem is units, how can I define other units such as mA and mW?

see sensor::Unit in sensor.h. w and kw, j and kwh, etc. are distinct enumeration types instead of something that scales with metric prefixes. in case of mA and mV, probably just divide by 1000 to get a 0.985A instead of 958A?

> and why the voltage value is rounded?

see BaseSensor::decimals(sensor::Unit) and _sensorUnitDecimals(sensor::Unit). Voltage is set to 0, so there's nothing after the dot. overriding decimals() would do the trick

> and also how to access the sensor values in other places? I need to adjust the light and some other parameters based on the battery voltage and led board temperature?

https://github.com/xoseperez/espurna/wiki/RPN-Rules ?

> Another thing is I need to have a battery level indicator according to the battery voltage how to do so?

https://developer.mozilla.org/en-US/docs/Web/HTML/Element/progress ?
there are no 'widgets' per-se, one needs to manually add this element and logic to the webui

hamed-ta, 2022-02-11:
> 

I tried to change the enums to scoped enumerations but as in the code there are some places that bit-wise operations have been done on enums, the code breaks and I think for the solution some operators should be overloaded makes things complicated, so I just moved the enums in the private scope of the class in that enough?

and could you please explain where the typecast tweaks are needed? I'm asking because I'm going to add this to the Espurna then.

mcspr, 2022-02-14:
(sorry for the delay, missed in the inbox)

Enumerations in C++ are a bit different, bitwise won't work with scoped enumerations since those are not really ints but almost a separate type. And also the fact that we still need to manually compose enum to be shifted one bit each element.

May not be the best idea though, and simple const uint*_t numbers may suffice. Or hide them in the class, which also works. Although, in system.{h,cpp} the heartbeat enumerations were migrated way back to class enums and custom operators with enum & int as lhs and rhs.

Re. types - I was referring to the old C type casting done with {u,}int16_t and maybe unintentional comparison. I'd expect compiler to raise warnings for some things in the above code where we work with unsigned and signed type. Yes, it kind-of works, but is not really a correct C++ code.

hamed-ta, 2022-04-08:
`

#if SENSOR_SUPPORT && EMON_INA219_SUPPORT

#pragma once
/******************************************************************************
 *
 * This is a library for the INA219 Current Sensor Module
 *
 * You'll find several example sketches which should enable you to use the library.
 *
 * You are free to use it, change it or build on it. In case you like it, it would
 * be cool if you give it a star.
 *
 * If you find bugs, please inform me!
 *
 * Written by Wolfgang (Wolle) Ewald
 * https://wolles-elektronikkiste.de/en/ina219-current-and-power-sensor (English)
 * https://wolles-elektronikkiste.de/ina219 (German)
 *
 * Modified by Hamed Taheri for Espurna (https://hamed-taheri.com)
 ******************************************************************************/
#include "Arduino.h"
#include <Wire.h>
#include "BaseAnalogEmonSensor.h"
#include "I2CSensor.h"

/* registers */
#define INA219_ADDRESS 0x40
#define INA219_CONF_REG 0x00    // Configuration Register
#define INA219_SHUNT_REG 0x01   // Shunt Voltage Register
#define INA219_BUS_REG 0x02     // Bus Voltage Register
#define INA219_PWR_REG 0x03     // Power Register
#define INA219_CURRENT_REG 0x04 // Current flowing through Shunt
#define INA219_CAL_REG 0x05     // Calibration Register

/* parameters */
#define INA219_RST 0x8000

    enum class INA219_ADC_MODE : uint8_t
    {
        BIT_MODE_9 = 0b00000000,
        BIT_MODE_10 = 0b00000001,
        BIT_MODE_11 = 0b00000010,
        BIT_MODE_12 = 0b00000011,
        SAMPLE_MODE_2 = 0b00001001,
        SAMPLE_MODE_4 = 0b00001010,
        SAMPLE_MODE_8 = 0b00001011,
        SAMPLE_MODE_16 = 0b00001100,
        SAMPLE_MODE_32 = 0b00001101,
        SAMPLE_MODE_64 = 0b00001110,
        SAMPLE_MODE_128 = 0b00001111
    };

    enum class INA219_MEASURE_MODE : uint8_t
    {
        POWER_DOWN = 0b00000000,
        TRIGGERED = 0b00000011,
        ADC_OFF = 0b00000100,
        CONTINUOUS = 0b00000111,
    };

    enum class INA219_PGAIN : uint16_t
    {
        PG_40 = 0x0000,
        PG_80 = 0x0800,
        PG_160 = 0x1000,
        PG_320 = 0x1800
    };
    
[... 382 lines cut ...]

        _voltage = _ina219.getBusVoltage_V();
        _current = _ina219.getCurrent_mA() / 1000.0;
        _power = _ina219.getBusPower() / 1000.0;

        espurna::time::blockingDelay(espurna::duration::Milliseconds(100));
    }

    double value(unsigned char index)
    {
        if (index == 0)
            return _voltage;
        if (index == 1)
            return _current;
        if (index == 2)
            return _power;
        return 0;
    }
};

#endif

`

What more things do I need to add to the code in order to add to the repository?

mcspr, 2022-04-08:
pls let me work out how we are dealing with PIO 5.3.0 first :) there are some breaking changes that need to be done to the builder scripts

if you intend to change the source above, I would suggest either work with a
- fork. either locally, or via web interface. (notice that file upload works via drag-n-drop!)
- [gist](https://gist.github.com). same git updates, but without the rest of the repo

hamed-ta, 2022-04-08:
> pls let me work out how we are dealing with PIO 5.3.0 first :) there are some breaking changes that need to be done to the builder scripts
>
> if you intend to change the source above, I would suggest [...]

I know, I will make a pull request from my fork but just wanted to make sure before that.

hamed-ta, 2022-04-08:
> see sensor::Unit in sensor.h. w and kw, j and kwh, etc. are distinct enumeration types instead of something that scales with metric prefixes. in case of mA and mV, probably just divide by 1000 to [...]

I studied the RPN rules, which seem interesting but how to use them in the code? I wonder if the is any normal way to access the value of sensors and change the value of lights or relays in a for example custom loop function?

l

mcspr, 2022-04-11:
> I studied the RPN rules, which seem interesting but how to use them in the code? I wonder if the is any normal way to access the value of sensors and change the value of lights or relays in a for example custom loop function?

Not many, besides modifying the RPN module code to expose / change certain things about the processing. C++ code changes require OTA update, RPN rule changes do not :)
(...unless we want to heavily split the resulting .elf file to make it more or less modular, allowing separate code section uploads... I have no real experience in that though...)

If you writing something outside of the sensor.cpp main module, access happens through the global index of the magnitude.
i.e. the order in which we processed the sensors and 'created' these magnitude objects
https://github.com/xoseperez/espurna/blob/02581c6988f9d6b52a74c82fb2f6b3d5f895627c/code/espurna/sensor.h#L148-L170
Access is still by the index, but I suppose you can write a proxy object similar to the Value one, hiding the actual implementation behind a pointer and various functions that operate on it.

hamed-ta, 2022-04-15:
> Not many, besides modifying the RPN module code to expose / change certain things about the processing. C++ code changes require OTA update, RPN rule changes do not :) (...unless we want to heavily [...]

Got it, for temp value I use this 

`        sensor::Value temp = magnitudeValue(3);

        DEBUG_MSG_P(PSTR("Temp: %s C\n\n"), String(temp.get()).c_str());`

but it's not straightforward, can you give me an example of the RPN rule in which if the temp goes higher than 70C it turns off the light and invoke a function that produces an alarm? and the same thing for voltage and current as well? I have a loop and extra setup function that I want to implement these codes there, how to do that?

mcspr, 2022-04-15:
ref. operators are described at https://github.com/mcspr/rpnlib#operators and names of these variables can be seen with `rpn.vars` in the terminal

check out the ``rpnrules::operators::relay::init(ctx)`` and `rpnrules::operators::light::init(ctx)` in the rpnrules sources. lights were not taken care of properly, there no function to change the status :/ need to take a mental note of that

but, suppose we have a virtual relay active, so the 1st relay can change the lights status
```forth
$temperature0 70.0 ge false 0u relay
```
rpn rules are executed *every time* something changes the variables, so think of it as ensuring the relay stays off until the temperature drops below 70
(but, another mental note would be to add relay locking support to the operators, since we are still technically allowed to turn the relay back ON only to immediately turn back OFF via the rule)

hamed-ta, 2022-04-15:
> ref. operators are described at https://github.com/mcspr/rpnlib#operators and names of these variables can be seen with `rpn.vars` in the terminal
> 
> check out the `rpnrules::operators::relay::init(ctx)` and `rpnrules::operators::light::init(ctx)` in the rpnrules sources. lights were not taken care of properly, there no function to change the status :/ need to take a mental note of that
> 
> but, suppose we have a virtual relay active, so the 1st relay can change the lights status
> 
> ```forth
> $temperature0 70.0 gt end false 0u relay
> ```
> 
> _edit: added proper condition to end the rule before..._
> 
> rpn rules are executed _every time_ something changes the variables, so think of it as ensuring the relay stays off until the temperature drops below 70 (but, another mental note would be to add relay locking support to the operators, since we are still technically allowed to turn the relay back ON only to immediately turn back OFF via the rule)

Got it, thanks. 

Is this the correct way of changing relay status?

`relayStatus(0, false);`

mcspr, 2022-04-15:
> Got it, thanks.
>
>Is this the correct way of changing relay status?
>
>relayStatus(0, false);

Yes. Also `lightState(false);`

mcspr, 2022-05-15:
On the topic... What is up with SADC and BADC bits? Our version has something different from the way  datasheet describes ADC operating mode, where these values are set separately.

https://www.ti.com/lit/ds/symlink/ina219.pdf

per 8.6.2.1 we have
- [3...6] bits SADC (Shunt ADC Resolution/Averaging)
- [7...10] bits BADC (Bus ADC Resolution/Averaging)

The code above, however, simply copies 4bit into both locations.
```cpp
// I2CPort::setADCMode(mode)
uint16_t adcMask = mode << 3;
currentConfReg |= adcMask;
adcMask = mode << 7;
currentConfReg |= adcMask;
```

I'd think we want them set separately? But, whatever is described in the BADC and SADC table does not look like bitmasks in the other ones :shrug:
Suppose they literally mean numbers 9,10,11 and 12... What represents samples then? Both as a 7bit number?

mcspr, 2022-05-17:
Ooh, nvm. Those are bits. Will push as a PR some time today.

hamed-ta, 2022-05-17:
> 

Sorry for the late answer, only on Fridays, do I get some time to work on this.

closed 2022-05-18 by mcspr

mcspr, 2022-05-18:
Note that the updated version defaults to datasheet values.

Not sure I got the scaling right, though.
I have not included the ratio / shunt res / idk. I suppose the idea was to offset the result in program itself, but shouldn't the calibration be changed instead? No idea yet of the API, for what would make sense to configure here.

hamed-ta, 2022-05-20:
> Note that the updated version defaults to datasheet values.
>
> Not sure I got the scaling right, though. I have not included the ratio / shunt res / idk. I suppose the idea was to offset the result in [...]

I am not sure what you mean, but in other libraries, you can set the calibration manually based on your exact value of the resistor and other configurations, especially the resolution, so you can calculate the calibration value with the software pack available on the TI website and set it on config time. but in this library, I found that this works great even without setting the calibration manually.

hamed-ta, 2022-05-20:
this should be only 'voltage' to make it more general!

https://github.com/xoseperez/espurna/blob/c980acbd10870dc5620e8399aaed3e90fe89e5c1/code/espurna/sensors/INA219Sensor.h#L459

mcspr, 2022-05-20:
By config I meant the part in `code/espurna/config/sensors.h`
(which is also not very descriptive of what `PG_...` does in relation to other values)

Right now I just used the library's version of pga gain value associated with a fixed set of params - calibration, current and power factor and a fixed Rshunt of 0.1
[image]
So, aren't we missing expected current and R shunt configuration? From expected current, we know current LSB. From current LSB and resistance, we know calibration. Power LSB is just scaled current LSB (times 20). Seemed like a weird approach to do math when device already does it :)

hamed-ta, 2022-05-20:
> By config I meant the part in `code/espurna/config/sensors.h` (which is also not very descriptive of what `PG_...` does in relation to other values)
>
> Right now I just used the library's version of [...]

Yes, I just updated my library and noticed that you missed the Rshunt... factors which is really needed, I will try to get them back!

hamed-ta, 2022-05-20:
I added the methods and tested the library it works well, yet I noticed a few other bugs which I will make some issues for them.

mcspr, 2022-05-21:
> This should be only 'voltage' to make it more general!
> 
> https://github.com/xoseperez/espurna/blob/c980acbd10870dc5620e8399aaed3e90fe89e5c1/code/espurna/sensors/INA219Sensor.h#L459

Removing would be better? In WebUI, 'Description' is on the right, while we already have 'Voltage' to the left anyway as the name of the measurement. Default method simply shows sensor name.