xoseperez/espurna #2503: Regression on "Do not quote numbers in MQTT JSON payloads"

alexmick, 2022-02-06:
**Bug description**
There seems to be an unexpected change between 1.14 and 1.15 : numeric values are now quoted in the JSON payloads sent over MQTT.

**Steps to reproduce**
Upgrade to espurna 1.15 and activate MQTT JSON mode. Message produced will look like this, with quoted numeric values :
```json
{
   "datetime":"2022-02-06 14:18:27",
   "mac":"*********",
   "host":"espurna-1",
   "ip":"192.168.1.XX",
   "id":172618,
   "energy":"1.351",
   "energy_delta":"230",
   "factor":"58",
   "apparent":"38",
   "reactive":"30",
   "power":"22",
   "voltage":"222",
   "current":"0.172"
}
```

**Expected behavior**
Messages sent with version 1.14 had numeric values instead of strings where applicable :
```json
{
   "current":0,
   "voltage":232,
   "power":0,
   "reactive":0,
   "apparent":0,
   "factor":0,
   "energy":44,
   "time":"2022-02-06 12:33:34",
   "mac":"********",
   "host":"espurna-1",
   "ip":"192.168.1.XX",
   "id":2394
}
```

**Device information**
Manufacturer: BLITZWOLF
Device: BWSHPX_V2
SDK version: 2.2.2-dev(38a443e)
Core version: 2.7.4
Version : 1.15.0-dev.gitf01ea1ce+nightly20220116

**Additional context**
On line [663 of CHANGELOG.md](https://github.com/xoseperez/espurna/blob/dev/CHANGELOG.md?plain=1#L663) a change mentions fixing these quoted numbers so I'm under the impression that this new 1.15 behaviour is unintended , maybe introduced by @mcspr's change #2405. If someone confirms this is unwanted, I'd be happy to further investigate what might have caused it !

mcspr, 2022-02-06:
> Additional context
On line [663 of CHANGELOG.md](https://github.com/xoseperez/espurna/blob/dev/CHANGELOG.md?plain=1#L663) a change mentions fixing these quoted numbers [here](https://github.com/xoseperez/espurna/commit/dd12251e43174281c737ca0868d5e6fcb8711558#diff-adf1d7b355d1037ad971dd9bf6ca48bce5d8d033d2f1e77073951c5553a2997cR577) so I'm under the impression that this new 1.15 behaviour is unintended , maybe introduced by @mcspr's [change here](https://github.com/xoseperez/espurna/commit/c01fbeb8ae6318b22e570d2d7b7a933c6975c4c4). If someone confirms this is unwanted, I'd be happy to further investigate what might have caused it !

Changelog refers to this modification btw
dd12251e43174281c737ca0868d5e6fcb8711558

Breaking happened here, moving some MQTT things around there is no longer `isNumber(...)` block converting string that looks like either float or int to the respective type inside of JSON
7ba1283024de11aeed569cdb5c636bc163f882b4
Which seemed backwards at the time, to receive a serialized data and immediately de-serialize it ...and I'd rather it did something like websockets and directly modified the JSON tree. Just not sure what to do with the rest of MQTT JSON things.

Also note we expose prometheus formatted data via `/api/metrics`
```
$ curl -H'Api-Key: ...' 'http://x.x.x.x/api/metrics'
relay0 0
current 0.000
voltage 210
power 0
reactive 0
apparent 0
factor 0
energy_delta 0
energy 0.054
```

closed 2022-02-11 by mcspr

alexmick, 2022-02-11:
Wonderful thanks for the fix ! Will test it this week-end 🚀

alexmick, 2022-02-21:
All good, works like a charm, thanks again for the wonderful work !

commit ce7de7d8 (2022-02-11): mqtt: once again, do not quote numbers in MQTT JSON payload

Restore behaviour from dd12251e43174281c737ca0868d5e6fcb8711558
7ba1283024de11aeed569cdb5c636bc163f882b4 unintentionally removed it
Instead of reverting to the 'atof' code, just use the provided string
as-is via RawJson (aka `serialized()` in the v6)

Note that the old approach may have caused some silent data loss b/c the
data could have been stored in 'double' or 'u64', where the ArduinoJson
only stores 'u32' and 'float' in it's internal buffer and does not (really)
warn us about any type incompatibilities.
(at least by default. bu [... cut]