xoseperez/espurna #2509: Websocket error in new devices

hamed-ta, 2022-04-15:
When I flash a new device before configuring the wifi client I get this error,

[image]

what is this for?

mcspr, 2022-04-15:
WebUI script failed somewhere. As the message suggests, take a look at Developer Tools. Console will have the specific code line of the exception, un-minifying the code you will see the context inside of  code/html/custom.js that calls forEach

davebuk, 2022-04-18:
Are you trying this on a small screen, phone maybe? I have this on certain pages when on my phone. When I've tried on a full browser or 'view as desktop' on my phone it's fine.

mcspr, 2022-04-18:
Developer tools include 'remote debugging' in case this happens on a phone
https://developer.chrome.com/docs/devtools/remote-debugging/
https://firefox-source-docs.mozilla.org/devtools-user/remote_debugging/debugging_firefox_for_android_with_webide/index.html

I'd assume the issue is related to an empty WiFi config.
btw... What browser, version and platform / OS? Which pages?

hamed-ta, 2022-04-18:
> Are you trying this on a small screen, phone maybe? I have this on certain pages when on my phone. When I've tried on a full browser or 'view as desktop' on my phone it's fine.

No on a desktop computer and with a Chrome browser. will send you a picture of the developer console.
دخ

davebuk, 2022-04-18:
Sorry, didn't look closely on my error. Mines to do with the toggle menu :-)

```
TypeError: Cannot read properties of null (reading 'classList')
    at HTMLDivElement.toggleMenu (http://192.168.1.34/:17:31055)

For more info see the Developer Tools console.
```

mcspr, 2022-04-18:
pretty sure the @davebuk issue was solved some time around January this year (https://github.com/xoseperez/espurna/commit/4c178948b5f5af48a4710548ba7c962f57f59b66)

@hamed-ta what commit are you running? can you also show the output of the c/p and running this snippet in the console?
```js
for (let style of document.styleSheets) { for (let rule of style.cssRules) { if (rule.cssText.includes("module-")) { console.log(rule.selectorText); } } }
```

hamed-ta, 2022-04-18:
> pretty sure the @davebuk issue was solved some time around January this year ([4c17894](https://github.com/xoseperez/espurna/commit/4c178948b5f5af48a4710548ba7c962f57f59b66))
> 
> @hamed-ta what commit are you running? can you also show the output of the c/p and running this snippet in the console?
> 
> ```js
> for (let style of document.styleSheets) { for (let rule of style.cssRules) { if (rule.cssText.includes("module-")) { console.log(rule.selectorText); } } }
> ```

I use this commit [c9ba70a](https://github.com/xoseperez/espurna/commit/c9ba70a7f0d795fcc2254d77f9bd6142ee5d66ac)

mcspr, 2022-04-18:
Try with the latest dev version

hamed-ta, 2022-04-18:
> Try with the latest dev version

I have made several changes in different files, it is not easy to merge with the new version, what tools do you recommend to compare all the files of in whole?

I also added another button action to put the device to sleep and wake it up with the interrupt of the same button, do you thing it is good to be added to the project?

mcspr, 2022-04-18:
> I have made several changes in different files, it is not easy to merge with the new version, what tools do you recommend to compare all the files of in whole?

https://www.sublimemerge.com/
https://code.visualstudio.com/
(or, just git cli :)

> I also added another button action to put the device to sleep and wake it up with the interrupt of the same button, do you thing it is good to be added to the project?

Sure. Preferably as a pull request.

hamed-ta, 2022-04-19:
> Try with the latest dev version

Yes, It is solved in the latest version.

closed 2022-04-19 by hamed-ta