xoseperez/espurna #2510: How to compensate battery voltage drop with light PWM value

hamed-ta, 2022-04-22:
I made custom hardware that has 5 PWM channels to control 5 light channels, this device uses a one-cell lithium battery which when is fully charged the voltage is about 4.2 volts and when it's getting empty it can go down to 3 volts. so when the user sets the light brightness to 50% when the battery is full the PWM value should be different than when the battery voltage is 3.7 and when it's 3.2. I should mention that I measure the battery voltage, current, and power of the device with a power meter sensor. so where and how do you think is the better approach to adjust the PWM value based on brightness and battery voltage?

mcspr, 2022-04-24:
Could be the same approach as with colored lights? With the most basic brightness, we simply walk the existing channels and apply it as a 0...100% scaling
https://github.com/xoseperez/espurna/blob/02581c6988f9d6b52a74c82fb2f6b3d5f895627c/code/espurna/light.cpp#L732-L736
https://github.com/xoseperez/espurna/blob/02581c6988f9d6b52a74c82fb2f6b3d5f895627c/code/espurna/light.cpp#L713-L730

You could either replace the implementation completely and make it dependent on the PWM value at that moment, or chain another object into the 'apply()` of the light channel. Like colored funcs do
https://github.com/xoseperez/espurna/blob/02581c6988f9d6b52a74c82fb2f6b3d5f895627c/code/espurna/light.cpp#L875
(i.e. it applies channel filtered through rgb object first, then brightness one)

And some periodic timer to force-apply channel values periodically

hamed-ta, 2022-04-25:
ok got it, or maybe adding another module to the project called for example battery support?

any idea of the approach of adjusting the PWM value according to the battery level? when for example the nominal voltage of LEDs is 3.3 volts and the battery voltage range is between 3 to 4.2 volts? which means there should be an upper limit for PWM value according to the exceeded voltage (battery voltage - 3.3) and a limited brightness value when the battery voltage is less than 3.3 to show to the user?

mcspr, 2022-04-26:
I'd think about the separation when we already have a clear api. It would also depend where are you getting the battery info; is it analog pin, is it some digital interface, etc. But if there is, sure.

re. values... just encode it as a 2-value table with specific voltage level <-> multiplier? for example, with 0.1v step (or, through experimentation, where voltage makes the most difference)

hamed-ta, 2022-04-28:
Or another approach could be, as I have the current, that can be possible to store the corresponding current value when the brightness is set and then adjust the PWM value when the battery level changes but in this way, I still need to limit the MAX PWM value to prevent over current problem...!