xoseperez/espurna #2511: Apple HomeKit support

hamed-ta, 2022-04-25:
Is there any plan to add the Apple homeKit support to espurna?

mcspr, 2022-11-12:
If we have the means to run it, yes.

Just to write down existing projects, there is a mgos homekit lib port for Shelly devices
* https://github.com/mongoose-os-apps/shelly-homekit
* https://github.com/mongoose-os-libs/homekit-adk

Build system skips whatever built-in TLS lib in favour of mbedtls bundled with the homekit sdk
* https://github.com/mongoose-os-apps/shelly-homekit/blob/e9f95b65e44afd1248be6bf5dcb1597c9547b204/mos.yml#L215-L216
* https://github.com/mongoose-os-libs/mbedtls

We also seem to need DNS-SD support (Bonjour?), iirc we have that already but just to note the external one
* https://github.com/mongoose-os-libs/dns-sd

mcspr, 2022-11-15:
+list
* https://github.com/RavenSystem/esp-homekit-devices

(thx to Github home page feed)