xoseperez/espurna #2512: Brigness and color change on light doesn't work with Alexa

hamed-ta, 2022-05-02:
I have one RGBWW light device with Espurna. when I ask alexa to search for devices it finds 5 different light bulbs, it seems one for each color! but only cold white and warm white work as a switch and the brightness and color changes doesn't work at all. what is the problem?

[image]
[image]

mcspr, 2022-05-05:
It is the way it is implemented in the fauxmoESP.
Also, duplicate of https://github.com/xoseperez/espurna/issues/1636

I am not using Echo, and (still) don't really understand it's inner workings, so can't really help there. You can check out the proposed colour control branch at https://github.com/vintlabs/fauxmoESP/pull/180 (https://github.com/vintlabs/fauxmoESP/tree/Colour) which seems to do what you want.

closed 2022-05-05 by mcspr

hamed-ta, 2022-05-09:
I will try to use Espalexa which seems more promising.