xoseperez/espurna #2513: I2C bus not working with build 3275f49.

davebuk, 2022-05-07:
Possibly similar issue to:
[#1672 ](https://github.com/xoseperez/espurna/issues/1672#issue-428368037) 

After building using the following, the I2C bus is not finding my two sensors and reporting 'ERROR 6' in debug.

```
#elif defined(DB_DEV_BOARD_1RY)

    // Info
    #define MANUFACTURER        "DEVBOARD"
    #define DEVICE              "ESP12F_BMX-BH_RELAY_X1_RFRX"

    //My config

    #define DEBUG_SERIAL_SUPPORT    0
    #define ALEXA_SUPPORT           0
    #define DOMOTICZ_SUPPORT        0
    #define HOMEASSISTANT_SUPPORT   0
    #define THINGSPEAK_SUPPORT      0
    #define NTP_TIMEZONE            TZ_Europe_London
    
    
    // Buttons
    #define BUTTON_MQTT_SEND_ALL_EVENTS     1
    #define BUTTON1_PIN         4
    #define BUTTON1_CONFIG      BUTTON_PUSHBUTTON | BUTTON_DEFAULT_HIGH | BUTTON_SET_PULLUP
    #define BUTTON1_RELAY       1
    #define BUTTON1_PRESS       BUTTON_ACTION_ON
    #define BUTTON1_CLICK       BUTTON_ACTION_OFF
    #define BUTTON1_DBLCLICK    BUTTON_ACTION_ON
    #define BUTTON1_LNGCLICK    BUTTON_ACTION_ON
    #define BUTTON1_LNGLNGCLICK BUTTON_ACTION_ON

    // Relays
    #define RELAY1_PIN          5    // Default GPIO 5 via R14
    #define RELAY1_TYPE         RELAY_TYPE_NORMAL

    // LEDs
    #define LED1_PIN            2
    #define LED1_PIN_INVERSE    1
    #define LED2_PIN            16
    #define LED2_PIN_INVERSE    1
    
    //Extras
    #define BMX280_SUPPORT 1
    #define BMX280_ADDRESS 0x76  
    #define BH1750_SUPPORT 1
    #define BH1750_ADDRESS 0x00
    #define I2C_SDA_PIN 12
    #define I2C_SCL_PIN 14
    // RF Support
    #define RFB_SUPPORT              1
    #define RFB_PROVIDER    RFB_PROVIDER_RCSWITCH
    #define RFB_RX_PIN              13
```

davebuk, 2022-05-07:
I tried setting BMX280_ADDRESS to 0x00, no change. I also tried having only one sensor defined and that didn't work either.

I've returned to a pre-built version using 1.15.0-dev.gitd5273513 and I2C sensors are shown again.

mcspr, 2022-05-07:
What `i2c.scan` command is saying?

Single bme280 should still work?
```
i2c.scan
found 0x76
+OK
magnitudes
0 * temperature/0 @ BME280 @ I2C (0x76) (read:23.980 reported:24.061 units:°C)
1 * humidity/0 @ BME280 @ I2C (0x76) (read:18.66 reported:18.69 units:%)
2 * pressure/0 @ BME280 @ I2C (0x76) (read:998.5729 reported:998.4582 units:hPa)
+OK
```

Not a lot of changes to the i2c & sensors here, but you might find the commit that broke it
(from the code/)
```
> git log d5273513...HEAD -- espurna/sensors/BMX280Sensor.h espurna/sensors/BH1750Sensor.h espurna/i2c.cpp
> # add -p after 'log' to see the actual changes in diff format
```

davebuk, 2022-05-07:
Odd, I've rebuilt using the same as before, flashed that file and its working now!

```
i2c.scan
found 0x23
found 0x76
```

Closed. 😁

closed 2022-05-07 by davebuk

mcspr, 2022-05-07:
Are you using -latest-base env?

davebuk, 2022-05-07:
Yes,

`extends = env:esp8266-4m-latest-base` in my `platformio_override.ini`

mcspr, 2022-05-07:
I would strongly suggest to git-pull again and rebuild everything. Barring possible hw issues, there's a fatal bug in the toolchain there that happens almost at random (...which I was just reminded of in the esp8266/Arduino repo...)

```
PACKAGES:
 - framework-arduinoespressif8266 @ 3.30002.0 (3.0.2)
 - tool-esptool @ 1.413.0 (4.13)
 - tool-esptoolpy @ 1.30000.201119 (3.0.0)
 - toolchain-xtensa @ 5.100300.211127 (10.3.0) <<< this thing is new
```

davebuk, 2022-05-24:
I have just built again and had an `ERROR 4` failure with I2C. Closed VSCode/platformIO, re-built and that file works. The one that failed was 575kB and the working one was 597kB. I have the two .bin files but not the .elf if the .bins are of any use?

mcspr, 2022-05-24:
.bin is already stripped of useful info, so not really.
sensor code related to i2c had a subtle bug with two or more devices though, git pull should fix those

plus, PIO turned out to have a build flags issue that sometimes happens b/c of the way libraries are built, you'd need to `pio upgrade --dev` yet again