xoseperez/espurna #2519: elevated temperature on SHT3X with newer code

drc38, 2022-05-21:
**Bug description**
For newer code there appears to be a difference in temperature reported of around +3C when using a SHT3X device. 

**Steps to reproduce**
Flashing the current `dev` code with i2c and sht3x enabled results in an elevated temperature being reported. The previous commit I compared to is `24a2c94e962c53467023ae4bfe619f68bcb3f579`. Firmware is being built in Arduino on a Wemos D1 Mini.

The calculation in `SHT3XI2CSensor.h` has not changed so assume it may be something upstream?

drc38, 2022-05-21:
The change on the RHS of the graph shows the impact of changing to newer firmware and back again:
[image]

mcspr, 2022-05-21:
Reported value is filtered, and filter code did just change. Perhaps it was not ported correctly, if there's a mistake when reporting and not reading from the device, I'd look there.
(ofc, if actual device is reporting the same values. `SENSOR_DEBUG=1` prints every read, or just print those values in pre() of the sensor)

drc38, 2022-05-21:
Enabling `SENSOR_DEBUG=1` gives this error:
`sensor.cpp:3358:30: error: 'using MagnitudeReadHandlers = class std::forward_list<void (*)(const String&, unsigned char, double, const char*)>' {aka 'class std::forward_list<void (*)(const String&, unsigned char, double, const char*)>'} has no member named 'push_back'`

mcspr, 2022-05-21:
Something wrong with your local tree? CI now builds with debug, it was also something fixed somewhat recently in 1c9553176d6a71c3c8ecac3d5104eb6c694d1df8

drc38, 2022-05-21:
Thanks, I was trying it in my older branch first to be able to compare to dev

mcspr, 2022-05-21:
Or, reducing report count to one reading would've also worked :)

drc38, 2022-05-21:
Ok, looks like the Corrections aren't recognised, when trying to save it says "No changes detected"

mcspr, 2022-05-21:
Hm, yeah. Correction settings key was wrong...
```diff
diff --git a/code/espurna/sensor.cpp b/code/espurna/sensor.cpp
index 2810c43d..e43094e9 100644
--- a/code/espurna/sensor.cpp
+++ b/code/espurna/sensor.cpp
@@ -783,7 +783,7 @@ SettingsKey get(::settings::StringView prefix, ::settings::StringView suffix, si
     String key;
     key.reserve(prefix.length() + suffix.length() + 4);
     key.concat(prefix.c_str(), prefix.length());
-    key.concat(suffix.c_str(), prefix.length());
+    key.concat(suffix.c_str(), suffix.length());

     return SettingsKey(std::move(key), index);
 }
```

mcspr, 2022-05-21:
b5e467d4513d46f2dc739fcea951a210eec56c5e for the above
683347c8d746a0223ba18105b6647549f0bbd265 for what I noticed in the median filter that applies to temperature... only seems to really matter in case report count is 1. but, when greater than one it still keeps the very last value, causing the last reading in the next cycle to be ignored. unlike the previous version with read-count-plus-one list of values. won't cause the large temperature difference though (i think)

drc38, 2022-05-21:
Thanks for fixing, it was the temperature correction value being ignored causing my issue. Also my sht31 sensors may have an issue as they require a large offset when they are supposed to be factory calibrated to within +-0.5C

closed 2022-05-21 by drc38

drc38, 2022-05-22:
@mcspr with the latest build when enabling `SENSOR_DEBUG` the sensor values output are corrected rather than raw values from the sensor, is that intended? I was expecting raw values.

drc38, 2022-05-22:
Another oddity the MQTT reading is always the previous reading rather than the current reading?
```
[840392] [SENSOR] temperature -> 21.8°C
[840395] [MQTT] Sending ROOM2/temperature => 21.9 (PID 1)
[840396] [SENSOR] humidity -> 57%
[840397] [MQTT] Sending ROOM2/humidity => 57 (PID 1)
[870400] [SHT3X] Status crc:0 cmd:0 htr:0
[870402] [SENSOR] temperature -> 21.9°C
[870405] [MQTT] Sending ROOM2/temperature => 21.8 (PID 1)
[870406] [SENSOR] humidity -> 57%
[870408] [MQTT] Sending ROOM2/humidity => 57 (PID 1)
```

mcspr, 2022-05-22:
> with the latest build when enabling SENSOR_DEBUG the sensor values output are corrected rather than raw values from the sensor, is that intended? I was expecting raw values.

debug log comes from yet another 'read' handler, also used for other things (like web). but I suppose it could just be a debug statement nearby raw & processed value.

> Another oddity the MQTT reading is always the previous reading rather than the current reading?

so we have a chain of value, starting from there
https://github.com/xoseperez/espurna/blob/09273b1bc1c6efa5ffb59a11291affd74a8bf329/code/espurna/sensor.cpp#L3670

raw value is fed into https://github.com/xoseperez/espurna/blob/dev/code/espurna/filters/MedianFilter.h(`magnitude.filter->update(value.raw)`) and report uses not the previously fed value, but the result from the filter (`magnitude.filter->value()`). `value.processed

mcspr, 2022-05-22:
> with the latest build when enabling SENSOR_DEBUG the sensor values output are corrected rather than raw values from the sensor, is that intended? I was expecting raw values.

debug log comes from yet another 'read' handler, also used for other things (like web). but I suppose it could just be a debug statement nearby raw & processed value.

> Another oddity the MQTT reading is always the previous reading rather than the current reading?

median filter keeps last value on reset(), moving it to the front
https://github.com/xoseperez/espurna/blob/09273b1bc1c6efa5ffb59a11291affd74a8bf329/code/espurna/filters/MedianFilter.h#L27
and the value() call would be the first value in the list
https://github.com/xoseperez/espurna/blob/09273b1bc1c6efa5ffb59a11291affd74a8bf329/code/espurna/filters/MedianFilter.h#L57

result comes from the filter, not the value directly. plus, depends on how much values are in the filter storage.