xoseperez/espurna #2522: Web based multiple device management.

davebuk, 2022-05-27:
Does anyone know of a webUI based interface that could be used to view the status of all espurna devices on a local network? It would be useful to scan the network and find known devices and add them to an interface. Maybe it could use the REST API to get device RELAY status, have the option to toggle those RELAYS, show the outputs of any sensors and report the MQTT status messages if those are being used.

I have found that basic webUIs can be made to run on a wemos D1 type device and control LEDs etc on that device, but if something exists that could be used to run a basic web server and build a page to show all devices I believe that could be quite useful.

I have made basic html/css pages in the past but wouldn't know what the best place is to look at building something to interact with espurna devices.

Could anyone offer any advice on known options? Would the 4MB memory of a wemos device be large enough to run the needed services to run the interface and server etc?

davebuk, 2022-06-03:
Would something along the lines of https://github.com/hivemq/hivemq-mqtt-web-client work?

MQTT over websocket seems to be a possible option?

mcspr, 2022-06-04:
Yes, MQTT broker just needs websockets endpoint enabled.
(since it is a separate thing from the normal tcp/1883)

davebuk, 2022-06-15:
I've made a start on a project at: https://github.com/davebuk/davebuk-Wemos-MQTT-websocket-PubSub-client
Its very early days but the core idea works. As long as your MQTT broker supports websockets, this can be run on a wemos d1 mini. 

By subscribing to: `+/ip` you can see the ip addresses of all devices and you can then go its web config.

`/+relay/0` will show all the relay#0 states of connected devices and you should be able to publish updates.

I'll look to add things like a wifi config page and possibly a better way of holding the file system as it uses LittleFS and you need to upload the files from plaformIO into the ESP storage. I can't get the images to load at the moment so I'll need to do some more work on why they aren't being served.

Any constructive help is always appricated. :-)

davebuk, 2022-06-17:
I now have the core implementation of what I wanted at: https://github.com/davebuk/davebuk-Wemos-MQTT-websocket-PubSub-client

I'll build on it over time and look to pre-define some of the more useful topics like `+/ip` `+/relay/0` etc.

Thanks for the tips.

closed 2022-06-17 by davebuk