xoseperez/espurna #2526: New board version R2 V.2.2

Seeelefant, 2022-07-24:
Unfortunately there is a new board version V2.2 as described [here](https://github.com/arendst/Tasmota/discussions/13283) . Can you add this to your wiki?

mcspr, 2022-07-24:
Reading the discussion... does it work with the current fw? As I understood it, protocol of communication with the 2nd MCU stayed the same.

Seeelefant, 2022-07-25:
I haven´t tried it yet. Currently I only have this window closed sensor 
[image]
Now assuming that your firmware works what message can I expect from this sensor? Do I go through a binding procedure? If yes, how?

mcspr, 2022-07-25:
No binding, 443 sensor will simply broadcast the message to whomever listens to it. SONOFF uses simple 24bit message that you could see in our logs; last 6 characters aka 3 bytes is a static code that does not change.

Then, we either associate with a relay action, [use RPN rule to match it and do something](https://github.com/xoseperez/espurna/wiki/RPN-Rules#rfbridge), or simply subscribe to `+/rfin` in MQTT to do something with it on a different device.

Seeelefant, 2022-07-30:
Its working! Shall I send you the photos of the new/modified board?

mcspr, 2022-07-30:
Sure! You could grag & drop them right here, then I could re-upload them to the wiki repository.

Seeelefant, 2022-07-30:
Here it is
[image]
By the way:  Do you know which GPIO Pins to set for RF_LED and the Buzzer?

mcspr, 2022-07-30:
https://github.com/xoseperez/espurna/wiki/Hardware-Itead-Sonoff-RF-Bridge---Direct-Hack#board-r2-v22
(assuming the rest from r2 v1.0 is still valid for the new one)

> By the way: Do you know which GPIO Pins to set for RF_LED and the Buzzer?

afaik, both are attached to the 8051. you could check out https://github.com/Portisch/RF-Bridge-EFM8BB1 source which sets up those pins (but, that may have changed between revisions)

closed 2022-07-30 by mcspr