xoseperez/espurna #2528: PZEM004T V3 software serial doesn't appear to work on latest dev build.

davebuk, 2022-08-03:
I have tried updating a device to the latest `dev` build but it can't now find the PZEM004TV3 device connected to it. I have the original build `1.15.0-dev.gitd5273513` working fine with the following definitions:

Using the latest dev build the sensor output shows all values as `undefined` and the debug shows:
 
`[SENSOR] Could not read from PZEM004T V3.0 @ SwSerial, 0xf8 (Other / Unknown error)`

I thought it might be an error at:
https://github.com/xoseperez/espurna/blob/e9de3d44354b4b6b2e3c9a22310485ebc6859ebf/code/espurna/sensors/PZEM004TV30Sensor.h#L39 
but changing to `{ PZEM004TV30_TX_PIN }` doesn't make a difference.

Is there anything that could have changed since Sep 2021 for the V3 PZEM004T?

davebuk, 2022-08-10:
I don't suppose it's as simple as adding that line back into the code?

mcspr, 2022-08-11:
See the commit, above is invalid b/c `port` no longer points to anything at that point in the code :oops:

commit 396f8a77 (2022-08-11): pzem004tv3: init serial port

fix a bug introduced in e4b068ec7d7173b457c5e616814069a3a55cde8b
resolve #2528