xoseperez/espurna #2530: Can "Energy #0" be reset over MQTT?

davebuk, 2022-08-11:
I'm current using a rule in openHAB to calculate kWh from kW and time between updates. It would be easier to just take the "Energy #0" value and reset at midnight.

Can this be done via MQTT?

mcspr, 2022-08-12:
Looking at the current code, energy can be reset by publishing anything to `<root>/energy/<id>/set`.
See debug log; `mqtt.reset` will re-connect to the broker, and print every subscription.

davebuk, 2022-08-12:
Sending `0` via openHAB comandTopic to `D1mini002/energy/0/set` seems to either crash device or `reboot`. `Crash` in the de-bug gives 'reason of restart 3'.

Total energy stays at previous value.

mcspr, 2022-08-12:
Crash reason does not mean much by itself. `crash` will show stack, exception decoder mentioned in the issue template should be able to decode it using `-s` switch.

iirc there is also PlatformIO built-in decoder that must be called while inside of the build dir using the env specific to the device, and it works over tcp through our 'telnet' port
```
> platformio device monitor --filter esp8266_exception_decoder --echo --port socket://192.168.4.1:23
```
(also see what happens via <kbd>Ctrl</kbd>+<kbd>F</kbd> and <kbd>H</kbd>, e.g. `esp8266_exception_decoder` or any other filter can be enabled / disabled from there)

davebuk, 2022-08-12:
I'll have a look and see what I can find.

davebuk, 2022-08-12:
Does the following mean anything to you? :-)

```
PS C:\VSCode\ESPurna\espurna\code> platformio device monitor --environment db-wemos-pzemv3 --filter esp8266_exception_decoder --port socket://192.168.1.55:23 --echo
--- Terminal on socket://192.168.1.55:23 | 115200 8-N-1
--- Available filters and text transformations: colorize, debug, default, direct, esp8266_exception_decoder, hexlify, log2file, nocontrol, printable, send_on_enter, time
--- More details at https://bit.ly/pio-monitor-filters
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H
[741505] [TELNET] Client #0 connected
last reset reason: Software Watchdog
extra info: Fatal exception:4 flag:3 (Software Watchdog) epc1:0x40214641 epc2:0x00000000 epc3:0x00000000 excvaddr:0x00000000 depc:0x00000000

latest crash was at 260706 ms after boot
Reason of restart: 3

>>>stack>>>

ctx: todo
sp: 3fffec60 end: 3fffffb0 offset: 0000
3fffec60:  3ffefe0c 3fff6914 3fff691b 4021463e
3fffec70:  3fff6914 0008001f 80000000 3fffec00 
3fffec80:  3fffec90 00000004 00000000 00000001
3fffec90:  3ffefe0c 3fff1f64 3fff5a5c 4023d548
3fffeca0:  00000000 3fff72ad 3fffc228 40100110 
3fffecb0:  00000002 3fff1f64 3ffefe0c 00000001 
3fffecc0:  3ffefe0c 3fff1f64 00000001 40209a41
3fffecd0:  00000000 00000000 00000001 00000006 
3fffece0:  3ffef16a 00000000 0000000c 3fff35a4
3fffecf0:  3fff72ac 3fff1f64 00000000 4023d5ac
3fffed00:  3ffef130 3fff1f64 00000000 4022d2b9 
3fffed10:  3fffed30 00000000 00000000 3fffee80 
<<<stack<<<

0x4021463e in ArduinoJson::Internals::StringTraits<__FlashStringHelper const*, void>::equals(__FlashStringHelper const*, char const*) at libraries\ArduinoJson\src/ArduinoJson/Deserialization/../StringTraits/FlashString.hpp:36
 (inlined by) ArduinoJson::Internals::ListIterator<ArduinoJson::JsonPair> ArduinoJson::JsonObject::findKey<__FlashStringHelper const*>(__FlashStringHelper const*) at libraries\ArduinoJson\src/ArduinoJson/JsonObject.hpp:270
 (inlined by) bool ArduinoJson::JsonObject::set_impl<__FlashStringHelper const*, unsigned int const&>(__FlashStringHelper const*, unsigned int const&) at libraries\ArduinoJson\src/ArduinoJson/JsonObject.hpp:293
 (inlined by) bool ArduinoJson::JsonObject::set<unsigned int, __FlashStringHelper const>(__FlashStringHelper const*, unsigned int const&) at libraries\ArduinoJson\src/ArduinoJson/JsonObject.hpp:114
 (inlined by) ArduinoJson::Internals::EnableIf<!ArduinoJson::Internals::IsArray<unsigned int>::value, ArduinoJson::Internals::JsonObjectSubscript<__FlashStringHelper const*>&>::type ArduinoJson::Internals::JsonObjectSubscript<__FlashStringHelper const*>::operator=<unsigned int>(unsigned int const&) at libraries\ArduinoJson\src/ArduinoJson/JsonObjectSubscript.hpp:41
 (inlined by) _relayWebSocketSendRelays at espurna/relay.cpp:2190
0x4023d548 in ClientContext::_write_from_source(char const*, unsigned int) at C:\Users\David\.platformio\packages\framework-arduinoespressif8266\libraries\ESP8266WiFi\src/include/ClientContext.h:490 (discriminator 2)
0x40100110 in std::function<bool (ApiRequest&, ArduinoJson::JsonObject&)>::operator()(ApiRequest&, ArduinoJson::JsonObject&) const at c:\users\david\.platformio\packages\toolchain-xtensa\xtensa-lx106-elf\include\c++\10.3.0\bits/std_function.h:623
0x40209a41 in espurna::led::(anonymous namespace)::Led::toggle() at espurna/led.cpp:364
0x4023d5ac in ClientContext::state() const at C:\Users\David\.platformio\packages\framework-arduinoespressif8266\libraries\ESP8266WiFi\src/include/ClientContext.h:375
 (inlined by) ClientContext::wait_until_acked(int) at C:\Users\David\.platformio\packages\framework-arduinoespressif8266\libraries\ESP8266WiFi\src/include/ClientContext.h:357
 (inlined by) ClientContext::_write_from_source(char const*, unsigned int) at C:\Users\David\.platformio\packages\framework-arduinoespressif8266\libraries\ESP8266WiFi\src/include/ClientContext.h:499
0x4022d2b9 in tryParseId(char const*, unsigned int (*)(), unsigned int&) at espurna/utils.cpp:27

[754002] [MQTT] Sending D1mini002/voltage => 246 (PID 1)
[754005] [MQTT] Sending D1mini002/frequency => 50.0 (PID 1)
[754006] [MQTT] Sending D1mini002/current => 3.029 (PID 1)
[754008] [MQTT] Sending D1mini002/power => 595 (PID 1)
[754013] [MQTT] Sending D1mini002/energy => 10796.433 (PID 1)
[754018] [MQTT] Sending D1mini002/factor => 80 (PID 1)
```

mcspr, 2022-08-12:
Looks weird, nowhere near the sensor code :shrug: Thanks nonetheless.

Looking at MQTT callback, there is simply a recursion happening with 'energy::set' calling it with the same argument over and over until it resets due to watchdog timeout.
Need to also fix PZEM code itself, it needs specific internal index which is not correctly selected (or used at all)

mcspr, 2022-08-15:
Latest dev fixed the above. Also noticed the way we set up energy tracking may be interfering with the desired use case.
https://github.com/xoseperez/espurna/blob/2d38efcd6b372cfe90683499c3a6093134a1cb95/code/espurna/sensor.cpp#L2913-L2926

Since the index <-> energy is fixed, here it will try to read either our internal kvs or rtcmem for energy and in case of PZEM, it will simply reset everything to 0 every time firmware boots
(since it does not know what to do with any value)

davebuk, 2022-08-15:
That all seems to be working. As you said, `energy` is re-set on boot (not desirable in case of crash and loss of that data).

If I send `0` to `D1mini002/energy/0/set` the device doesn't crash :-) and `energy` is re-set.

What has changed that a re-boot now re-sets the `energy` value? I thought that data was held in the PZEM memory and read into the esp?

mcspr, 2022-08-16:
> What has changed that a re-boot now re-sets the energy value? I presumed that data was held in the PZEM memory and read into the esp?

This is part of the original code dealing with 'simple' sensors where we take care of the energy persistence. Original PZEM had invalid numeric ID <-> ENERGY association, which is now restored to the correct value so energy reset through API works. The issue here is that both API and boot code (above) use the same exact energy reset API, where 'simple' sensor simply starts from zero or using RAM value. Here it is not the case.

mcspr, 2022-08-18:
Latest dev should ignore the request to set energy on boot, but still allow CLI / MQTT / HTTP API calls to reset it. There is also an ENERGY DELTA value stored each reading; but, since the sensor only reports in Wh, it will tick up in 3600 Ws / J increments.

davebuk, 2022-08-18:
Yep, that all works. ENERGY no longer re-sets on boot/re-boot.

Even though ENERGY units is currently set to 'kWh', the ENERGY DELTA is always in 'J'.

I'm not sure if ENERGY DELTA is reporting correctly. During low demand times the ENERGY is around 480W (say 500W). That should be 0.5 kWh if predicted to draw 500W for an hour. During that current draw, 'J' changes between ~3600 and ~7200. I believe that is 0.001-0.002 kWh (or is that 0.1-0.2 kWh)? The sensor is set to read every 10 seconds and I would have thought for an approx 0.5 kWh I would see 18,000 J (or 18,000,000 J)?

mcspr, 2022-08-18:
Yeah, it should probably have kWh option (and Wh, too). Note that display unit config is specific to measurement, not overall 'this is value for every energy thing'. It is literally $E_{current} - E_{last}$ every reading, then summed for the report (N readings for report setting).

Internally, value for these inputs is the exact same one. In case J value is this
$E = 3\ 600\ 000\mathrm{J}$
Wh conversion divides by 3 600
$E = 1\ 000\mathrm{Wh}$
And kilo conversion divides by 1 000
$E = 1\mathrm{kWh}$

davebuk, 2022-08-23:
Was there any more you wanted to do with this or shall I close the question?

mcspr, 2022-08-25:
> Was there any more you wanted to do with this or shall I close the question?

Sure, just checking for any problems while using these new energy counters.
BTW, are latest commits working ok as well, not just the one that updated the energy thing?

closed 2022-08-25 by mcspr

davebuk, 2022-08-25:
1.15.0-dev.gitc86b4c37 is working well.