xoseperez/espurna #2531: Nous A1T OTA flashing from Tasmota to ESPURNA?

Homer-Sim, 2022-08-18:
Hello,
I bought some Nous A1T which come flashed already with Tasmota.
I am more used to ESPURNA and all my other plugs running it.

Might it be possible to "give" Tasmota just a espurna firmware to reflash it?
Which firmware would be fitting?

Thanks

mcspr, 2022-08-18:
We have a slightly different approach at configuration, but most things stay the same. If the idea is to flash something pre-built, take a look at hardware.h for a compatible pin selection and grab the matching .bin from release page (preferably, the snapshot one, in case it somehow breaks during or after the process b/c of slight incompatibilities and SDK differences)

https://nous.technology/product/a1t/template.html
GPIO0 is a pushbutton
GPIO4, GPIO5 and GPIO12 are used for HLW8012-compatible BL0937 chip. Note SELi aka SEL(ect) i(nverted) 
GPIO13 is an inverted LED
GPIO14 is a relay

e.g. this AVATTO one
https://github.com/xoseperez/espurna/blob/844f20003ded9e00c26c8c845e097ec50308bc31/code/espurna/config/hardware.h#L2906-L2937
or KMC 70012
https://github.com/xoseperez/espurna/blob/844f20003ded9e00c26c8c845e097ec50308bc31/code/espurna/config/hardware.h#L2043-L2071

or, just build it using these values

Homer-Sim, 2022-08-18:
Great, I will check that.

Can I use the upload new firmware option within tasmota to flash espurna firmware?

mcspr, 2022-08-19:
I don't remember any problems doing that the last time I tried it (but, it has been a while). iirc Tasmota updater may not like anything *but* Tasmota binaries, but recent builds introduced a way to circumvent that.

Homer-Sim, 2022-08-20:
Hello,

is this the right firmware bin for the Avatto?
[espurna-1.15.0-dev.git97a4e0a8+github20220819-avatto-power-plug-wifi.bin](https://github.com/xoseperez/espurna/releases/download/github20220819/espurna-1.15.0-dev.git97a4e0a8+github20220819-avatto-power-plug-wifi.bin)

This is how it looks like in tasmota:
[image]

Homer-Sim, 2022-08-21:
Ok, I dared to flash it.
First with a minimalistic tasmota and then with the above written espurna bin.

I see a WIFI espurna but with fibonacci I can not connect.
Connection refused ...

Any ideas? Help is highly appreciated.

mcspr, 2022-08-21:
Connection refused on WiFi or the Web page? Even after device power off -> on cycle?
I have noticed captive portal issue, so automatic login might not work correctly. WebUI needs admin as username and fibonacci as password

Homer-Sim, 2022-08-21:
connecting the WIFI is refused.
I plugged it out/in no changes. LED is flashing 3 times then after a while once slow and 3 times faster .... if this helps

Homer-Sim, 2022-08-21:
So it seems it is bricked?! I am not able to open and solder something to the plug.....

mcspr, 2022-08-21:
Does the button work, switching the relay?

Blinking LED just means that the WiFi access point is working
Does it connect to WiFi on a different device?

Homer-Sim, 2022-08-21:
Button seems to work, I can hear a click but with a delay

I have different plugs which are working fine (with ESPURNA) and previously with this plug it worked with tasmota

Homer-Sim, 2022-08-21:
Sometimes there appears a second WIFI:
It is not working to connect too
Button seems to work, I can hear a click but with a delay

[image]

mcspr, 2022-08-21:
The issue here is I have no idea why it would not connect :/
Click is the relay, yes, but does it *stay on*? If, for example, we have a software crash it would not connect due to that. Just to understand the state the device is currently in.

mcspr, 2022-08-26:
> Which would give us the indicator for a rebooting loop, right?!

Exactly. And the button behaviour, too.

Homer-Sim, 2022-08-28:
OK, I did the test.
after plugging in it takes ~20-30 seconds then the button is switching and light (I attached a simple light to it) is shining.
After additional ~max 5 sec light is going out
Once I was able to connect to the plug and the logindata window appeared but then it switched off again.

Anything else I can do except throwing it into the bin?

mcspr, 2022-08-30:
There are at least two issues at play and I have not noticed / considered
- energy monitor counter has a bug that resets the device (and a not really obvious one, even with all of compilation warnings enabled). e704efffb42d1ad667ce3833c0d4025912b05a5d will fix that
- because the reset is due to the hardware watchdog trigger, current implementation erased our internal 'reboot counter' value causing effectively a non-stop reboot cycle without a way to recover and it will never trigger SAFE mode. 7659a63441a7fafff30a274caad65db44de7983a removes this handling of hwdt resets

The script approach may work. I suppose directly uploading stuff to `/upgrade` endpoint would do the trick with a fixed .bin, but I am not really sure if it will upload everything in one go or not. At least everything related to updates happens separately from the sensor code that fails.

Will need to trigger a new build here.

Homer-Sim, 2022-08-30:
>  _edit: reverting to -core .bin will definitely work in the mean time_
Which one would you suggest? And, sorry for this maybe silly question,  how do I upload to /upgrade folder?
Is there a howto any where?

mcspr, 2022-08-30:
> > _edit: reverting to -core .bin will definitely work in the mean time_
> Which one would you suggest? And, sorry for this maybe silly question,  how do I upload to /upgrade folder?
> Is there a howto any where?

See https://github.com/xoseperez/espurna/wiki/OTA#updating-via-http

Homer-Sim, 2022-09-02:
I have the issue that I am not able to  write a script to connect to ESPURNA WIFI and if established to upload the core bin.

Is there something I can use?

mcspr, 2022-09-03:
`netsh wlan` on Windows, `nmcli` (or `wpa_cli`) on Linux, `networksetup` on macos?
all seem to have a common SSID + PASSPHRASE connection option

Homer-Sim, 2022-09-04:
I managed to do a connection to espurna with nmcli

Which core bin should I use?

This one?

espurna-1.14.1-espurna-core-4MB.bin

mcspr, 2022-09-04:
1.15 one for 1MB, latest snapshot; to avoid version downgrades

Homer-Sim, 2022-09-04:
espurna-1.15.0-dev.git0e48dab5+github220830-espurna-core-4MB.bin 
This one?

Homer-Sim, 2022-09-04:
I tried it and received this:

```
Device 'wlx74da386523be' successfully activated with 'b630acbc-216d-4263-a6be-ed6cc53fb8d6'.
PING 192.168.4.1 (192.168.4.1) 56(84) bytes of data.
64 bytes from 192.168.4.1: icmp_seq=1 ttl=255 time=75.5 ms

--- 192.168.4.1 ping statistics ---
1 packets transmitted, 1 received, 0% packet loss, time 0ms
rtt min/avg/max/mdev = 75.544/75.544/75.544/0.000 ms
ERROR: No magic byte / invalid flash config
```

What does the ERROR Message means?

mcspr, 2022-09-04:
See above, you need 1MB - the same as the one used to build avatto .bin
https://github.com/xoseperez/espurna/blob/8bfc0c4882fb1e4a2cbd3174db54c19d039dcf94/code/platformio.ini#L777-L779

You could still use curl to download, just add `-L` so redirects work. And `-O` to replicate wget behaviour setting download output

Homer-Sim, 2022-09-04:
I tried again, but it stoppes for hours
[image]

This is my bash script:

```
#!/bin/bash

nmcli dev wifi connect ESPURNA-A11CFB password "fibonacci"
ping -c 1 192.168.4.1
curl -XPOST --digest -uadmin:fibonacci \
  -H "Content-Type: multipart/form-data" \
  -F "filename=@/home/h/espurna-1.15.0-dev.git0e48dab5+github220830-espurna-core-1MB.bin" \
  http://192.168.4.1/upgrade
```

Homer-Sim, 2022-09-05:
How do I know the file is uploaded and it will be flashing the plug with the core version?
Do I need to unplug/plug to start this update process?

mcspr, 2022-09-05:
-core does not have led configured, so missing wifi notification and a longer working AP when its done and already running that new version.
no unplug necessary, process happens immediately after updater receives the file. only question is whether it has time to proceed with actually writing it over the existing app. I wonder if espota.py might work better (see desc on the OTA wiki page)

Homer-Sim, 2022-09-05:
It seems that it worked now.

What should be the next steps?

mcspr, 2022-09-05:
Upload the latest snapshot instead of the old and broken one :)

Homer-Sim, 2022-09-05:
Just to be sure, 
this one I used previously: espurna-1.15.0-dev.git97a4e0a8+github20220819-avatto-power-plug-wifi.bin
Is this the broken one?

And this one I should try now?:
espurna-1.15.0-dev.git0e48dab5+github220830-avatto-power-plug-wifi.bin

mcspr, 2022-09-05:
> And this one I should try now?:
espurna-1.15.0-dev.git0e48dab5+github220830-avatto-power-plug-wifi.bin

Yes. Same device config, with the included fixes I mentioned in https://github.com/xoseperez/espurna/issues/2531#issuecomment-1231024234 (as seen in commit log)

Homer-Sim, 2022-09-05:
GREAT and big thanks for your support 
It is working.

BUT one question if I am changing hostname under GENERAL it is not saving it.
Bug or Feature?

Homer-Sim, 2022-09-05:
This made it work:

curl -F "data={\"app\":\"ESPURNA\",\"hostname\":\"Dose-WZ-LampeSO\"};type=application/json" --digest -u admin:cde3VFR$ http://192.168.100.213/config

mcspr, 2022-09-06:
> BUT one question if I am changing hostname under GENERAL it is not saving it.
Bug or Feature?

Definitely a bug in WebUI scripts since `/config` works, will check it out. And probably not the only text entry not working.

closed 2022-09-06 by mcspr