xoseperez/espurna #2537: Support for ZMPT101B AC Single Phase Voltage Sensor Module

amarendrapramanik, 2022-09-20:
### Description

Hay there,

Kindly extend support for ZMPT101B AC Single Phase Voltage Sensor Module.

mcspr, 2022-09-20:
Judging by these two documents, don't we already support it as EmonAnalogSensor? We calculate R for the input, and simply read A0 aka ADC voltage
[ZMPT101B specification.pdf](https://github.com/xoseperez/espurna/files/9610137/ZMPT101B.specification.pdf)
[ZMPT101B innovatorsguru.pdf](https://github.com/xoseperez/espurna/files/9610140/ZMPT101B.innovatorsguru.pdf)

amarendrapramanik, 2022-09-23:
You are correct but some significant modification needs to be done to calibrate the sensor with the actual reading.

This portion of the code preventing creating sinusoidal wave at the output 

    unsigned int analogRead() override {
      auto now = TimeSource::now();
      if (now - _last > _interval) {
        _last = now;
        _value = ::analogRead(A0);
      }
      return _value;
    }

Instead we need to hit A0 with minimal time gap.
Also could you please explain more about EMON_CURRENT_RATIO
Current ratio in the clamp (30A/1V) 

What is the context of using here?

mcspr, 2022-09-24:
Interval is there based on the SDK docs and some feedback at the esp8266/Arduino repo. We can't continuously read ADC, WiFi driver will lose its mind and break since we load the system too much.

afaik ESP8266 ADC has two ways of reading it
- [analogRead aka what Arduino originally supported. it is using `system_analog_read()`.](https://github.com/esp8266/Arduino/blob/a0c7a85649a3c48c05e70f3014769af23d01ec3b/cores/esp8266/core_esp8266_wiring_analog.cpp#L35) it does some sampling on its own
- [`system_analog_read_fast()` that is supposedly a somewhat simpler version, where you are supposed to do sampling manually](https://github.com/esp8266/Arduino/blob/502d9469fa022a5ce942bc2f2940f42699664ef7/tools/sdk/include/user_interface.h#L126)

re. ratio code. It was pretty much the same since originally introduced, working as a simple multiplier of the raw readings. Note that I am neither the original author or actually use the sensor in question