xoseperez/espurna #2539: Lights JSON broken on recent home assistant

mcspr, 2022-09-24:
Noticed in gitter
https://gitter.im/tinkerman-cat/espurna?at=632e273b9d3c186299170be1

Based on updated docs
https://www.home-assistant.io/integrations/light.mqtt/#json-schema

`"white_value"` is deprecated and replaced with `["color"]["w"]` and `["color"]["c"]` combined with `"color_mode"`
https://github.com/xoseperez/espurna/blob/75a518c6caaa40289737b0c8faaf8b96e5ce129a/code/espurna/homeassistant.cpp#L414-L430

https://www.home-assistant.io/integrations/light.mqtt/#color_mode
> Flag that defines if the light supports color modes.

https://www.home-assistant.io/integrations/light.mqtt/#supported_color_modes
> A list of color modes supported by the list. This is required if color_mode is True. Possible color modes are onoff, brightness, color_temp, hs, xy, rgb, rgbw, rgbww, white.

mcspr, 2022-09-26:
Some context. Plus, might be important to remember that HASS has a dev blog
* https://github.com/home-assistant/core/pull/71002 (white value marked for deprecation)
* https://developers.home-assistant.io/blog/2022/08/18/light_white_value_removed (removal in 2022.09)
* https://github.com/home-assistant/architecture/discussions/519 (background)
* https://github.com/home-assistant/core/pull/47720 (color modes)

mcspr, 2022-10-31:
Fix is in the tree, just pending some testing for the concurrent WebUI changes for the binary

mattster98, 2023-01-24:
I noticed this issue on a H801 I have had on 1.14.1 for a long time but only now getting into Home Assistant.  I two-step upgraded to the 230112 nightly and it is now being provisioned in HomeAssistant, HOWEVER, it is only showing up as an off/on light.  None of the color or brightness controls are being discovered, even though they are enabled.  I have tried various reboot, enable/disable of HASS integration, mqtt json on/off, etc.. but nothing seems to help.

mcspr, 2023-01-25:
HA might give some hints in the logger, where it processes MQTT subscription to `homeassistant/+`
Entity changes between versions were not really avoidable
* we have switch\_0, as you would with physical relay. we have this one set up through relay provider, if you don't want it upload config with `"relayProv0": "none"` (or, in terminal - `set relayProv0 none` and `reset`)
* lights entity is a brand new one, since HA had breaking changes and it is no longer possible to use some of the old stuff without it completely aborting the setup

mattster98, 2023-01-25:
I think the relay provider attached to the device is coming through the MQTT plugin, and the Light entity is coming via the HASS integration setting on Espurna.  What I can't resolve is the Light isn't associated with the Device, only the Relay.  Maybe that's straightforward to fix in HA (I'm still new at it).  Thanks!

Overnight the Light entity lost its "read only" status on the entity list.. and seems to be staying that way.

Still, I can't trivially update the light color, etc with an automation since it doesn't show as a device.  I'll keep poking at the HA side to see if I can fix that myself.

mcspr, 2023-01-25:
^ `del relayProv0` and reset

*from MQTT side of things*

1.14.1 did not have this single topic, so we had an extra relay to do state switches. This is no longer the case and lights work differently, but it is kept for backwards compatibility (and for people not using HASS; no need to change topics)

Both switch and light are separate entities when we publish to discovery topic.

mattster98, 2023-01-25:
Thanks!  I think what threw me off so much was how straightforward some of the simple relay (even multiple relay) espurnas still on 1.14.1 came into HA, but this one (and the others like it, as I eventually discovered) are quite different.

closed 2024-08-30 by mcspr