xoseperez/espurna #2541: Firmware for a new (unsupported?) switch board

GitJa21, 2022-10-04:
### Device

Unnamed, seems like Geekcreit

### Question

Hello!
There is such a board. (PSF-B85/PSF-B04, RFE274DM and one unknown chip, apparently related to the RF part)
It has two separate relay control buttons, a two-color LED indicating the operation of the board and two single-color LEDs indicating the operation of the relay.
Using the buttons, we can switch wifi to hotspot and to pairing of 433MHz remotes.
I don’t want to try e-Welink (especially register on their website)
I really want to flash ESPurna into it. Since I like it and it is already flashed on different ESP switches in my network.
Ofcourse, I want to be able to control both relays using the RF remote control. (I can't even imagine if this is possible on this board)
Tell me what firmware can be used in finished form. Maybe someone has already tried...
Is there a generic version where I can configure the GPIO and RCs scan-codes via the Web interface?
Or adapt some of the existing firmware.
[image]
[image]

mcspr, 2022-10-05:
PSF-B04 is a ESP8285, so at least that works :)
* https://itead.cc/diy-kits-guides/psf-b04-application-guide/
Quick search through the device db for tasmota, here what looks like it
* https://templates.blakadder.com/2ch_smart_switch.html

I'd expect RF works the same as other ITEAD / SONOFF devices (4CH RF, BASIC RF, wall mounted switch T1, etc.) - RF chip handles all the learning and emulates a button press when it receives a matching signal. Just a matter of finding out what it needs to 'learn' a signal 

While we already have DUAL\_R2 hardware that has matching pins
https://github.com/xoseperez/espurna/blob/35656314aa5f452a5b09fbb3acbe508fde9e280a/code/espurna/config/hardware.h#L638-L670
https://github.com/xoseperez/espurna/blob/35656314aa5f452a5b09fbb3acbe508fde9e280a/code/platformio.ini#L445-L447
I am a bit worried b/c tasmota db has these completely different pins; we might have a mislabeled device
https://templates.blakadder.com/sonoff_dual_R2.html

GitJa21, 2022-10-05:
Thanks a lot for the information.
I flashed the firmware from Sonoff Dual R2. Through WEB everything works perfectly. Relay and LED.
But for some reason, both the physical buttons and the radio began to work incorrectly. When the button is pressed, the state of the relay changes, and after releasing button, relay returns to the previous state.

mcspr, 2022-10-06:
> But for some reason, both the physical buttons and the radio began to work incorrectly. When the button is pressed, the state of the relay changes, and after releasing button, relay returns to the previous state.

Dual configuration specifies that these are 'switches', so 'press' and 'release' events toggle the relay. Could be remedied either by changing both and re-building, or adding a runtime setting that changes them from `switch` to `button`. In the terminal (DEBUG, telnet, serial connection)
```
set btnMode0 pushbutton
set btnMode1 pushbutton
reset
```
Reboot, and also check out what `button 0` and `button 1` say after that

GitJa21, 2022-10-08:
Added parameters. Still no changes.

"btnMode0":"switch",
"btnMode1":"switch",
Button #1:
```
[345476] [BUTTON] Button #1 event 2
[345481] [RELAY] #1 scheduled ON in 0 ms
[345498] [RELAY] #1 set to ON
[345709] [BUTTON] Button #1 event 2
[345712] [RELAY] #1 scheduled OFF in 0 ms
[345727] [RELAY] #1 set to OFF
[346731] [RELAY] Setting relay mask: 0b0
```
Button #0:

```
[438668] [BUTTON] Button #0 event 2
[438673] [RELAY] #0 scheduled ON in 0 ms
[438691] [RELAY] #0 set to ON
[438828] [BUTTON] Button #0 event 2
[438832] [RELAY] #0 scheduled OFF in 0 ms
[438849] [RELAY] #0 set to OFF
[439854] [RELAY] Setting relay mask: 0b0
```
"btnMode0":"pushbutton",
"btnMode1":"pushbutton",

Button #1:
```
[085263] [BUTTON] Button #1 event 2
[085267] [RELAY] #1 scheduled ON in 0 ms
[085280] [RELAY] #1 set to ON
[085563] [BUTTON] Button #1 event 2
[085568] [RELAY] #1 scheduled OFF in 0 ms
[085581] [RELAY] #1 set to OFF
[086584] [RELAY] Setting relay mask: 0b0
```
Button #0:
```
[095240] [BUTTON] Button #0 event 2
[095244] [RELAY] #0 scheduled ON in 0 ms
[095257] [RELAY] #0 set to ON
[095366] [BUTTON] Button #0 event 2
[095370] [RELAY] #0 scheduled OFF in 0 ms
[095382] [RELAY] #0 set to OFF
[096386] [RELAY] Setting relay mask: 0b0
```
WEB, BUTTON0:
On
```
[391598] [WEBSOCKET] Requested action: relay
[391602] [RELAY] #0 scheduled ON in 0 ms
[391610] [RELAY] #0 set to ON
[392614] [RELAY] Setting relay mask: 0b1
```
Off
```
[430593] [WEBSOCKET] Requested action: relay
[430598] [RELAY] #0 scheduled OFF in 0 ms
[430606] [RELAY] #0 set to OFF
[431611] [RELAY] Setting relay mask: 0b10
```
WEB, BUTTON1:
On
```
[411815] [WEBSOCKET] Requested action: relay
[411819] [RELAY] #1 scheduled ON in 0 ms
[411826] [RELAY] #1 set to ON
[412831] [RELAY] Setting relay mask: 0b11
```
Off
```
[462615] [WEBSOCKET] Requested action: relay
[462619] [RELAY] #1 scheduled OFF in 0 ms
[462626] [RELAY] #1 set to OFF
[463631] [RELAY] Setting relay mask: 0b0
```

But sometimes, after multiple clicking, relay finally changed status
```
[967861] [BUTTON] Button #1 event 2
[967866] [RELAY] #1 scheduled OFF in 0 ms
[967883] [RELAY] #1 set to OFF
[968010] [BUTTON] Button #1 event 2
[968015] [RELAY] #1 scheduled ON in 0 ms
[968031] [RELAY] #1 set to ON
[968811] [BUTTON] Button #1 event 2
[968816] [RELAY] #1 scheduled OFF in 0 ms
[968833] [RELAY] #1 set to OFF
[968990] [BUTTON] Button #1 event 2
[968994] [RELAY] #1 scheduled ON in 0 ms
[969012] [RELAY] #1 set to ON
[969733] [BUTTON] Button #1 event 2
[969738] [RELAY] #1 scheduled OFF in 1128 ms
[969860] [BUTTON] Button #1 event 2
[969864] [RELAY] #1 scheduled OFF in 1001 ms
[970017] [RELAY] Setting relay mask: 0b10
[970868] [RELAY] #1 set to OFF
[971875] [RELAY] Setting relay mask: 0b0
```

mcspr, 2022-10-08:
re. deleted message, were there any problems with the setup? I don't think pushbutton setup is wrong, but double-check the 'button 0' and 'button 1' output for the full config

GitJa21, 2022-10-11:
Thanks for the quick and interesting replies!
Yes, I had to delete the message. I realized that I was playing with the old firmware (1.14) Then I reflashed and repeated the experiment (1.15)
But unfortunately, your advice did not work with a new firmware too (of course I reboot the device).
  "btnMode0": "pushbutton",
  "btnMode1": "pushbutton",
```
[488471] [BUTTON] Button #0 event 1 (pressed)
[488493] [RELAY] #0 set to ON
[488979] [BUTTON] Button #0 event 3 (click)
[489000] [RELAY] #0 set to OFF
[489851] [BUTTON] Button #1 event 1 (pressed)
[489872] [RELAY] #1 set to ON
[490352] [BUTTON] Button #1 event 3 (click)
[490373] [RELAY] #1 set to OFF
[491378] [RELAY] Relay mask: 0b0
```
  "btnMode0": "switch",
  "btnMode1": "switch",
```
[015009] [BUTTON] Button #0 event 1 (pressed)
[015031] [RELAY] #0 set to OFF
[015149] [BUTTON] Button #0 event 2 (released)
[015171] [RELAY] #0 set to ON
[016118] [BUTTON] Button #1 event 1 (pressed)
[016139] [RELAY] #1 set to OFF
[016269] [BUTTON] Button #1 event 2 (released)
[016290] [RELAY] #1 set to ON
[017296] [RELAY] Relay mask: 0b11
```

I tried other dual+ firmware and **4CH** firmware **worked** as it should. True, 4 sliders are displayed (This is the lesser evil :) without rebuilding the firmware, of course)

Sorry, some commands from the wiki are not quite clear without examples. Is it possible to change the INPUT / OUTPUT direction of the GPIO and how to reassign it to the relay / button ...

_By the way, RF works the same as other SonOff devices - RF chip handles all the learning and emulates a button press when it receives a matching signal. Indeed._

Here are some differences in settings. But I still can't figure out why it doesn't work correctly.
(I set the switch mode manually after pushbutton, so nothing change anyway)
[image]

mcspr, 2022-10-12:
Ah, since dual declares `pressed` / `btnPress` in build config, we need to set it to `none` to accompany pushbutton mode change. 4ch doesn't have anything there, so it works like you expect.

Relay association is also a setting for button - `btnRelay0`, `btnRelay1`, and etc. numbered from 0
If action does something related to relays, it would use that relay index to `relay-toggle` or `relay-on` or `relay-off`

GitJa21, 2022-10-12:
Wow! I did not think that it is possible to reassign procedures on events. 
So commands  
```
set btnRlse0 none 
set btnRlse1 none 
```
did their job. 
Everything works perfectly! Thanks a lot!

closed 2022-10-12 by GitJa21