xoseperez/espurna #2543: Are DallasSensor temperature readings averaged or median'ed

CraigMarkwardt, 2022-10-05:
### Device

DS18B20

### Version

1.15.0-dev

### Question

I have a group of four DS18B20 sensors attached to a Wemos D1 Mini.  These are all embedded in the ice of an ice rink.  Occasionally, maybe two times per day, sometimes 3-4 times and sometimes never in a day, one of the sensors will report spurious values.  I am requesting values every six seconds, and "averaging" 5 samples.
[image]
(the larger variations at the beginning of the sample set are when the temperature control loop was open loop)

Are the "every 5" values averaged or median'ed?  When I look at the sensor.cpp code I see the default filter for temperature data is median, rather than a running average, but the code is complicated enough that it's hard to say.

When I look at the spurious filtered values, they are exact multiple of the correct value.
    (spurious value) = (true value) x factor
where factor is 0.5, 0.75, 0.833, etc.  The "true value" is found by looking by the near neighbors in time.

Here is a plot of X=(true value) and Y=(spurious value)/(true value) and the multiples are shown with dashed lines.
[image]

My thought is that the averaging is not being done properly, or a spurious sensor value of "0" is being inserted into an averaging filter and propagating through.  I see that checksums seem to be enabled for DallasSensor.h so in principle the raw values inserted into the filter should be correct.  Also, typically "median" filtering should not produce such a multiple effect, and instead should ignore such an outlier.  Just trying to figure out where this could come from.

CraigMarkwardt, 2022-10-06:
And, I think from reading the code, there is a way to change a setting to determine which filter is used, but I don't see how this should be done.

mcspr, 2022-10-06:
> 1.15.0-dev

This does not tell much :) How do you build? You should be able to see git hash throughout our build (WebUI, boot message, `info` in terminal), or when using any desktop client UI or by running `git rev-parse HEAD` while PWD is anywhere in the repo clone directory.

>  I am requesting values every six seconds, and "averaging" 5 samples.

Which means you have sensor reading every 6 seconds and sensor report set to 5 readings? For example,  if you pull data from HTTP API and also use real-time setting, value is really the last reading not the value from the filter. iirc it is the same thing with Prometheus and Influx (...but I would need to re-check that)

> Are the "every 5" values averaged or median'ed? When I look at the sensor.cpp code I see the default filter for temperature data is median, rather than a running average, but the code is complicated enough that it's hard to say.

Filter itself is set up on boot (check out `get tmpFilter0` in the terminal), and then we only control its capacity (`get snsReport` / `SENSOR_REPORT_EVERY` flag).
https://github.com/xoseperez/espurna/blob/35656314aa5f452a5b09fbb3acbe508fde9e280a/code/espurna/sensor.cpp#L4027-L4037
https://github.com/xoseperez/espurna/blob/35656314aa5f452a5b09fbb3acbe508fde9e280a/code/espurna/sensor.cpp#L3872
https://github.com/xoseperez/espurna/blob/35656314aa5f452a5b09fbb3acbe508fde9e280a/code/espurna/sensor.cpp#L3878
(which by the looks of it may cause some wrong overall data as it happens before the value is actually read, and it is also shared between *all* measurements. plus the fact that we may report based on delta settings causing a filter reset)

From the description above, we allocate 5+1 slots for values in the filter and push values every reading. Since there is a +1, latest value is kept for the next report loop.
https://github.com/xoseperez/espurna/blob/35656314aa5f452a5b09fbb3acbe508fde9e280a/code/espurna/sensor.cpp#L3929-L3930
https://github.com/xoseperez/espurna/blob/35656314aa5f452a5b09fbb3acbe508fde9e280a/code/espurna/filters/MedianFilter.h#L15-L19
https://github.com/xoseperez/espurna/blob/35656314aa5f452a5b09fbb3acbe508fde9e280a/code/espurna/filters/MedianFilter.h#L68-L74
https://github.com/xoseperez/espurna/blob/35656314aa5f452a5b09fbb3acbe508fde9e280a/code/espurna/filters/MedianFilter.h#L30-L62
https://github.com/xoseperez/espurna/blob/35656314aa5f452a5b09fbb3acbe508fde9e280a/code/espurna/sensor.cpp#L3957-L3961

You should be able to pick both base filter and the filter code headers and build them using host `CXX`. Or, use the `code/test/unit` facilities to add another test case, if we have some sample data that we could compare with.

CraigMarkwardt, 2023-10-09:
Hi @mcspr , I know I'm responding a year later.  Illness and family issues intervened and then this issue got pushed far down the stack.  But I'm replying because I have found some bugs that can be easily fixed.  Also, a lot has changed in one year so I was able to apply fixes to the current codebase.  

These are a few simple diffs.  Hopefully they can be easily applied.  I  don't frequently develop with github and I know I'll spend hours doing all the dancing to get a pull request prepared, so I hope you can just paste these in.

### Results up front

Before the fixes ("noise spikes" due to invalid CRC data being reported as 0-temperature)
[image]

After the fixes (no invalid CRCs survive to reporting)
[image]

### BUG 1: some sensors don't know their status until value()

The current code checks the sensor's status() after reading but before calling value().  DallasSensor only computes the checksum during value() so it doesn't know about errors until then.

This fix puts a check after value() as well.

```c++
diff --git a/code/espurna/sensor.cpp b/code/espurna/sensor.cpp
index 0d5504d5..666bebf9 100644
--- a/code/espurna/sensor.cpp
+++ b/code/espurna/sensor.cpp
@@ -3969,6 +3970,10 @@ void loop() {

             value.raw = magnitude.sensor->value(magnitude.slot);

+           if (!magnitude.sensor->status()) {
+                continue;
+            }
+
             // But, completely remove spurious values if relay is OFF
 #if RELAY_SUPPORT && SENSOR_POWER_CHECK_STATUS
             switch (magnitude.type) {
``` 

### BUG 2: DallasSensor can report values before first reading

The DallasSensor technique is to pre-allocate zeroed memory for each sensor's scratchpad.  It turns out that "all zeroes" is a legitimate scratchpad, including CRC, so it passes the CRC check.  Until the first reading is successfully pulled from the actual sensor, DallasSensor is happy to return "0" temperature values.  This usually happens for the first 1 or two samples, depending on snsRead.

The following fix adds some salt to the buffers to prevent a CRC clash until a true read can occur.  Also the _error status is set to something not-OK, though I'm not sure that's very useful.

```c++
diff --git a/code/espurna/sensors/DallasSensor.h b/code/espurna/sensors/DallasSensor.h
index 6a4415a4..92e4ff67 100644
--- a/code/espurna/sensors/DallasSensor.h
+++ b/code/espurna/sensors/DallasSensor.h
@@ -140,9 +140,16 @@ class DallasSensor : public BaseSensor {
                 _previous = _gpio;
             }

+           // Make sure the buffers have invalid CRCs
+           for (size_t index = 0; index < _devices.size(); ++index) {
+             _devices[index].data[0] ^= 0xcc;
+             _devices[index].data[1] ^= 0xcc;
+           }
+
             _last_reading = TimeSource::now();
             _ready = true;
             _dirty = false;
+           _error = SENSOR_ERROR_NOT_READY;

         }
```

### BUG 3: MovingAverage filter is a sum, not an average

The current MovingAverage filter computes a sum but never actually divides by the count to get an average.  The result is the incorrect value for the average.

```c++
[... 121 lines cut ...]
     double value() const override {
         double sum { 0.0 };
```

and to allow sensors to take advantage of this
```c++
diff --git a/code/espurna/sensor.cpp b/code/espurna/sensor.cpp
index 0d5504d5..666bebf9 100644
--- a/code/espurna/sensor.cpp
+++ b/code/espurna/sensor.cpp
@@ -4009,6 +4014,11 @@ void loop() {
             // Initial status or after report counter overflows
             bool report { ready_to_report() };

+           // If the filter is not yet fully initialized, don't report
+           if (! magnitude.filter->status() ) {
+             report = false;
+           }
+
             // In case magnitude was configured with ${name}MaxDelta, override report check
             // when the value change is greater than the delta
             if (!std::isnan(magnitude.reported) && (magnitude.max_delta > build::DefaultMaxDelta)) {
```

Thanks for your help!

mcspr, 2024-02-08:
> BUG 1: some sensors don't know their status until value()

True, but that was the intention behind sensor->pre() which is followed by sensor error check in sensor loop. I have changed most of the sensors to report value() from a pre-read value to avoid this issue specifically, dallas was overlooked so far but the commit above have moved things around.

> BUG 2: DallasSensor can report values before first reading

Should be fixed, since pre() would not allow value readings?

> BUG 3: MovingAverage filter is a sum, not an average
> SIDE NOTE: I don't think the MedianFilter is actually doing a median filter, but more like some kind of moving average. I'm not sure what was intended there, so it's hard to suggest a fix.

See https://github.com/xoseperez/espurna/blob/dev/code/test/unit/src/filters/filters.cpp
This is how it works right now, with the expected values from all filters. I would expect this would be much easier to run through some specific values compared to just waiting for the device to manifest them.

https://github.com/xoseperez/espurna/blob/37d2837937a9a882149e6ebb2188e9a17b6000cf/ci_script.sh#L9-L15

mcspr, 2024-05-15:
test/dev is merged with dev, so these are no longer lingering commits...
5e599ec2775bd308cb445e57b6c140ec0c684dd3                                                                                                                                                                           3c6160440328d90dd70cc689da6255c860f3ca3b                                                                                                                                                                           b580dd4d1c0d97d94e3393e7d0ad188ccbae1cc7                                                                                                                                                                           e92be04eea4848098a8a11675a1ece527966adee                                                                                                                                                                           6c382d7259a427fc52ea280f7868484c5c5dc6b1

'sensor->error(index)', 'sensor->value(index)' etc. is solved similar to other sensors - using separate class instance for each real sensor.

Test suggestion still stands, should the filter implementation(s) need a change.

commit 70f214e2 (2024-08-09): sns: update filter api, actually calculate median

- sync with base class, implement required override methods
- make sure last, max, min and sum filters reset() and restart() are the same
- find a real median over the specified reading 'window' vs. averaged medians
  of every 3 elements that it was previously
- preserve list of inputs for average and median filters, handle updates
  like a circular buffer and return value based on previous readings
  (todo might be to have a more efficient storage instead of vector)

resolve #2543
closed 2024-08-09 by mcspr