xoseperez/espurna #2544: Some bugs in 1.15 firmware

GitJa21, 2022-10-12:
### Device

Sonoffdual (in unnamed device)

### Version

1.15

### Bug description

Greetings!
I really liked the espurna firmware and the community feedback :)
I have to use firmware version 1.15 (1.14 does not allow me to disable unnecessary events in my unique device https://github.com/xoseperez/espurna/issues/2541)
Firmware 1.15 has some annoying bugs.
I hope this will be helpful when removing them.
1) When I change the Hostname and save - the web interface says "No changes detected". And there's no way to change that. Only through the console. set hostname <name>
2) Description - cannot be set in any way.
3) Also, the device name is not visible as in 1.14 (screenshot)
[image]
4) NTP server is not working, so I need to change it to something like time.windows.com.
But after reboot the ntp-server reverts to 195.78.244.50. Impossible to change even through the console.
5) It would be great if "Time Zone" was a drop down list. Since it's quite hard to figure out how to set POSIX TZ
(Values from https://github.com/esp8266/Arduino/blob/master/cores/esp8266/TZ.h work, but usability...)
(Time is very important to me because I need a scheduler.)
6) Annoying window saying "Connection lost with the device, click OK to refresh the page" after 5 minutes of idleness.
Firefox cheeky switches to this tab every 5 minutes! (In version 1.14, there were no such problems at all)

7) I saw someone wanted to be able to centrally manage all Espurna devices... Me too))
My previous love was the firmware https://github.com/tretyakovsa/Sonoff_WiFi_switch but the guys abandoned the project and the firmware does not work on some chips.
That firmware had a nice feature - devices see each other due to UPNP.
So, in interface of one device shows the controls of other devices with the same firmware. (In a separate interface's box. Even can set different devices in the scheduler.)
I think, this functionality would be useful to us to ))

mcspr, 2022-10-12:
1 & 2, looks like a bug in remote side 'accepting' certain key names and not any others. Just need to restore something acknowledging both of those

4 is related to your network setup, you already have auto-configured NTP server set through DHCP. If you don't want that, `set ntpDhcp 0`

5 TZ is intentionally left out. It is pita to maintain internally, plus the additional .bin size for all of those and the code managing this list

6 Seems like a page visible / invisible issue, but difficult to say without looking at. We are periodically sending keep-alive messages to see if remote responds. Just need to add handlers that notice page is in the background and stop doing that temporarily

7 For the barebones approach, you might want to look at MQTT messages we send out and use something like https://play.google.com/store/apps/details?id=net.routix.mqttdash
Every device is `<NAME>/relay/<RELAY INDEX>`, status update is `<NAME>/relay/<RELAY INDEX>/set` with `0` of `1`
Integrating WebUI / UPNP / any other discovery is much... wider topic for just a short answer :)

ElectricImpossible, 2022-10-27:
Items 1 & 2 affect me on 1.15.0-dev.gitb0af0afd+github220925

mcspr, 2022-10-28:
See commit above. Will be included in the next build.
Closing in favour of separate issues.