xoseperez/espurna #2547: Possible to prevent relay toggling on and off after restart/upgrade?

Kaan88, 2022-10-27:
### Device

V9261F with custom relay on GPIO5

### Version

espurna-1.15.0-dev.git97a4e0a8+github20220819

### Question

Searched but couldn't find anything.

Basically I would like my devices to stay on If I restart Espurna or upgrade it. Currently the relay quickly turns off and then on. 

On Tasmota if the switch was on it stays on during reboot or upgrade, same with esphome with "early_pin_init". Reference: https://github.com/esphome/esphome/pull/3144

Is something like this possible?

mcspr, 2022-10-27:
Something like this, just otoh. For example, if we have `relayGpio0 => 5`, we use `set gpioIgnore 0b100000` to ignore GPIO5 when booting (...since our boot mask is relay ID not GPIO number...)

Unlike esphome, it does not specify mode but instead ignores init altogether and expects the module to do it. In our case, 'Same' relay mode will keep status on reboot

```diff
diff --git a/code/espurna/gpio.cpp b/code/espurna/gpio.cpp
index 397e461b..f86e393c 100644
--- a/code/espurna/gpio.cpp
+++ b/code/espurna/gpio.cpp
@@ -15,6 +15,7 @@ Copyright (C) 2017-2019 by Xose Pérez <xose dot perez at gmail dot com>

 #include "mcp23s08_pin.h"

+#include "rtcmem.h"
 #include "terminal.h"
 #include "ws.h"

@@ -879,12 +880,16 @@ void pinMode(uint8_t pin, uint8_t mode) {

 // Special override for Core, allows us to skip init for certain pins when needed
 void resetPins() {
-    const auto& hardware = hardwareGpio();
+    const auto ignore = Rtcmem->pin_ignore;

+    const auto& hardware = hardwareGpio();
     for (size_t pin = 0; pin < espurna::gpio::Hardware::Pins; ++pin) {
         if (!hardware.valid(pin)) {
             continue;
         }
+        if (ignore & (1 << pin)) {
+            continue;
+        }
 #if DEBUG_SERIAL_SUPPORT
         // TODO: actually check which pin is set as TX via function select
         if (pin == 1) {
diff --git a/code/espurna/rtcmem.h b/code/espurna/rtcmem.h
index d89115e3..f3845390 100644
--- a/code/espurna/rtcmem.h
+++ b/code/espurna/rtcmem.h
@@ -25,7 +25,7 @@ Copyright (C) 2019 by Maxim Prokhorov <prokhorov dot max at outlook dot com>
 #define RTCMEM_BLOCKS 96u

 // Change this when modifying RtcmemData
-#define RTCMEM_MAGIC 0x46535076
+#define RTCMEM_MAGIC 0x46536086

 // XXX: All access must be 4-byte aligned and always at full length.
 //      Exactly like PROGMEM works. For example, using bitfields / inner structs / etc:
@@ -51,6 +51,7 @@ struct RtcmemData {
     uint32_t magic;
     uint32_t sys;
     uint32_t relay;
+    uint32_t pin_ignore;
     uint32_t mqtt;
     uint64_t light;
     RtcmemEnergy energy[4];
diff --git a/code/espurna/system.cpp b/code/espurna/system.cpp
index 067e37d5..03f7b1ed 100644
--- a/code/espurna/system.cpp
+++ b/code/espurna/system.cpp
@@ -526,6 +526,9 @@ void hardware() {
     pinMode(16, OUTPUT);
     digitalWrite(16, HIGH);
 #endif
+
+    const auto pin_ignore = getSetting("gpioIgnore", uint32_t(0));
+    Rtcmem->pin_ignore = pin_ignore;
 }

 // If the counter reaches SYSTEM_CHECK_MAX then the system is flagged as unstable
```

mcspr, 2022-10-28:
> 1 second or so

Difference in time should be this change
https://github.com/xoseperez/espurna/commit/af12762eb61f9d3a76e4465e717dc0b7a7665f89

---

Just to make sure, relay boot mode is 'Same' (`relayBoot0 => same`)?

I am not seeing this on a test POW (telnet'ed)
```
relay 0
...
> relayGpio0 => 12
> relayBoot0 => same
...
keys
> cfg => "12"
> gpioIgnore => "4096"
> relayBoot0 => "same"
> relayBootMask => "0b1"
> relayMask => "0b1"
```
Pushbutton turned the relay on, then I typed `reset` in the telnet and relay stayed ON during that time

Kaan88, 2022-10-28:
I don't have "relayMask", (adding it made no difference) everything else is the same. Relay switching is so brief light barely turns off, but I can hear it clearly. 

Just to make sure it isn't my hacky job of connecting stuff, I checked it with tasmota and it stays on during reboot.

Btw, I have to both add the changes from the patch and use gpioIgnore, right?

Kaan88, 2022-10-28:
Forgot to mention, the relay I bought is generic 5v 10a but it never switched off when fed it with 5v, it only worked with 3.3v. But I guess this doesn't have anything to do with it.

mcspr, 2022-10-28:
> Btw, I have to both add the changes from the patch and use gpioIgnore, right?

Right, it loads that setting value as bitmask and uses at the next boot. Don't really see the pattern here :/

Any difference with `stable` instead of `reset`? Both web and terminal command increase our reboot counter, which is what triggers the safe mode when it is done too frequently.
Does WebUI reboot leave a crash dump? What is `info` last reset reason: ...?

---

nevermind relayMask, just bad copy-paste

Kaan88, 2022-10-28:
No crash and reason is whichever command I used. 

What's weird is that after I used the reset command for the 3rd time, right when it goes into safe mode for the first time, the switch doesn't toggle at all. But this only happens once.

Btw, this is how I added the relay, I'm using the NO section of it. So it's off when the pin is High.

```
    // Relays
    #define RELAY1_PIN          5
    #define RELAY1_TYPE         RELAY_TYPE_INVERSE
```

It's so brief usb devices with power supplies don't even turn off.

mcspr, 2022-10-29:
Commenting / removing this might help, inverse mode causes extra write here
https://github.com/xoseperez/espurna/blob/f1d5b62146e7dff7f65c3c2e31e5affcc4a74e7c/code/espurna/relay.cpp#L1235-L1237
(for whatever reason that code is there)

Kaan88, 2022-10-29:
And it worked! So this code ignored whatever we were doing and pulled it high anyway, huh? 

It works in dev code too. I guess I can close the issue now. 

Thanks a lot! It's perfect now for what I need.

Btw, will this code be merged or be configurable? I suck at patching and did everything manually. :)

closed 2022-10-29 by Kaan88

mcspr, 2022-10-29:
Options are to
* keep this approach, also add some build flag in-place of that `uint32_t(0)` getSetting arg
* re-use our existing 'locked' gpio list (`gpio`, `gpio.locks`) and write to the pin\_ignore mask every time something locks a pin

Relay boot mode could also use an additional 'same' variant that works without flash writes by doing everything through RAM and falling back to some specific status on boot