xoseperez/espurna #2550: Issues with I2C sensors again.

davebuk, 2022-11-03:
### Device

ESP12 Dev board

### Version

https://github.com/xoseperez/espurna/commit/03ed94b771051ac1f7162f0d2d3f4cc7bdd312c0

### Bug description

Similar to #2513 building with the latest dev is not showing any I2C sensors. I was trying to add `#define BH1750_MODE  BH1750_CONTINUOUS_HIGH_RES_MODE_2` to my config for 0.5 resolution for the BH1750.

During build I get the following but it does complete and produce a .bin The file is smaller than the one built using `dev.git03ed94b7` by around 30kB though.
```
espurna\debug.cpp:143:6: warning: 'bool espurna::debug::settings::{anonymous}::sdkDebug()' defined but not used [-Wunused-function]
  143 | bool sdkDebug() {
      |      ^~~~~~~~` 
```

I've update using `pio update --dev` and Pull/re-based the latest `dev` but that hasn't helped.

davebuk, 2022-11-03:
After building with the correct definitions it works! Whoops, sorry ;-)
`i2c.locked` gives 
```
0x23
0x76
```
`gpio.locks` gives
```
* hardware GPIO13	1396:espurna\rfbridge.cpp:rfbSetup
* hardware GPIO16	1042:espurna\led.cpp:setup
* hardware GPIO2	1042:espurna\led.cpp:setup
* hardware GPIO4	1334:espurna\button.cpp:_buttonGpioPin
* hardware GPIO5	819:espurna\gpio.cpp:gpioRegister
* hardware GPIO3	362:espurna\uart.cpp:make_port
* hardware GPIO1	358:espurna\uart.cpp:make_port
```

However, the LUX value is still whole numbers, not decimal. During the correct build :-) I get various issues to do with the BH1750 definitions.
```
In file included from espurna\config/all.h:40,
                 from espurna\espurna.h:25,
                 from espurna\garland.cpp:46:
espurna\config/sensors.h:151: warning: "BH1750_MODE" redefined
  151 | #define BH1750_MODE                     BH1750_CONTINUOUS_HIGH_RES_MODE
      |
In file included from espurna\config/all.h:30,
                 from espurna\espurna.h:25,
                 from espurna\garland.cpp:46:
espurna\config/custom.h:819: note: this is the location of the previous definition
  819 |  #define BH1750_MODE  BH1750_CONTINUOUS_HIGH_RES_MODE_2
      |
In file included from espurna\config/all.h:40,
                 from espurna\espurna.h:25,
                 from espurna\gpio.cpp:9:
espurna\config/sensors.h:151: warning: "BH1750_MODE" redefined
  151 | #define BH1750_MODE                     BH1750_CONTINUOUS_HIGH_RES_MODE
      |
In file included from espurna\config/all.h:30,
                 from espurna\espurna.h:25,
                 from espurna\gpio.cpp:9:
espurna\config/custom.h:819: note: this is the location of the previous definition
  819 |  #define BH1750_MODE  BH1750_CONTINUOUS_HIGH_RES_MODE_2
```

I guess putting `#define BH1750_MODE BH1750_CONTINUOUS_HIGH_RES_MODE_2` in my `custom.h` file is the correct place?

mcspr, 2022-11-03:
iirc these are default I2C pins, used by something unknown and a button?
```
* hardware GPIO4	1334:espurna\button.cpp:_buttonGpioPin
* hardware GPIO5	819:espurna\gpio.cpp:gpioRegister
```
also need i2c to get displayed in gpio 'locks', it is definitely missing. weird that it finds devices, though

davebuk, 2022-11-03:
`custom.h` is:
```
#elif defined(DB_DEV_BOARD_1RY)

    // Info
    #define MANUFACTURER        "DEVBOARD"
    #define DEVICE              "ESP12F_BMX-BH_RELAY_X1_RFRX"

    //My config

    #define DEBUG_SERIAL_SUPPORT    0
    #define ALEXA_SUPPORT           0
    #define DOMOTICZ_SUPPORT        0
    #define HOMEASSISTANT_SUPPORT   0
    #define THINGSPEAK_SUPPORT      0
    #define NTP_TIMEZONE            TZ_Europe_London
    
    
    // Buttons
    #define BUTTON_MQTT_SEND_ALL_EVENTS     1
    #define BUTTON1_PIN         4
    #define BUTTON1_CONFIG      BUTTON_PUSHBUTTON | BUTTON_DEFAULT_HIGH | BUTTON_SET_PULLUP
    #define BUTTON1_RELAY       1
    #define BUTTON1_PRESS       BUTTON_ACTION_ON
    #define BUTTON1_CLICK       BUTTON_ACTION_OFF
    #define BUTTON1_DBLCLICK    BUTTON_ACTION_ON
    #define BUTTON1_LNGCLICK    BUTTON_ACTION_ON
    #define BUTTON1_LNGLNGCLICK BUTTON_ACTION_ON

    // Relays
    #define RELAY1_PIN          5    // Default GPIO 5 via R14
    #define RELAY1_TYPE         RELAY_TYPE_NORMAL

    // LEDs
    #define LED1_PIN            2
    #define LED1_PIN_INVERSE    1
    #define LED2_PIN            16
    #define LED2_PIN_INVERSE    1
    
    //Extras
    #define BMX280_SUPPORT 1
    #define BMX280_ADDRESS 0x76  
    #define BH1750_SUPPORT 1
    #define BH1750_ADDRESS 0x00
    #define BH1750_MODE  BH1750_CONTINUOUS_HIGH_RES_MODE_2
    #define I2C_SDA_PIN 12
    #define I2C_SCL_PIN 14
    // RF Support
    #define RFB_SUPPORT              1
    #define RFB_PROVIDER    RFB_PROVIDER_RCSWITCH
    #define RFB_RX_PIN              13
```

mcspr, 2022-11-03:
Thx. And the is nothing returned in `magnitudes` command too? Does a single sensor config work?

davebuk, 2022-11-03:
Sorry if you hadn't picked up on my initial mistake. I have two configs in my `custom.h` for the development board I'm using, one with I2C enabled and one without. I got them round the wrong way so that's why the sensors weren't showing up! Building the correct setup, as shown above worked as expected.

The issue I now have is getting the LUX value to show the 0.5 values for the BH1750 sensor.

mcspr, 2022-11-03:
Hard to track, yes :)

Rounding comes from our hard-coded number of digits. Lux does not have anything set, and default value is `0`
https://github.com/xoseperez/espurna/blob/cc90554b0e7e1c2923d3f08fd2ce1edeaec393f1/code/espurna/sensor.cpp#L1714-L1719
```diff
diff --git a/code/espurna/sensor.cpp b/code/espurna/sensor.cpp
index 93bf7668..c760c305 100644
--- a/code/espurna/sensor.cpp
+++ b/code/espurna/sensor.cpp
@@ -1700,6 +1700,8 @@ unsigned char decimals(Unit unit) {
         return 3;
     case Unit::WattSecond:
         return 0;
+    case Unit::Lux:
+        return 3;
     case Unit::CountsPerMinute:
     case Unit::MicrosievertPerHour:
         return 4;
```

Language wise, I'd like a suggestion for user-side name here. Internally it is named 'decimals' (function, and sometimes sensor side). Does not really sound right in the setting name.

davebuk, 2022-11-04:
Sorry 🫣

I could only suggest "Decimal of value, magnitude or reading" or "Fraction of ...." ?

mcspr, 2022-11-05:
`luxPrecision`? (https://simple.wikipedia.org/wiki/Arithmetic_precision)
Meaning, we have these for every magnitude available for adjustment at runtime. Or, `luxFraction`; `luxDecPoint`

Based on https://www.mouser.com/datasheet/2/348/bh1750fvi-e-186247.pdf (pg. 11), measurement here needs to be adjusted based on mode and halved in mode2
> The below formula is to calculate illuminance per 1 count.
>  * H-reslution mode : Illuminance per 1 count ( lx / count ) = 1 / 1.2 *( 69 / X )
>  * H-reslution mode2 : Illuminance per 1 count ( lx / count ) = 1 / 1.2 *( 69 / X ) / 2 

> 1.2 : Measurement accuracy
69 : Default value of MTreg ( dec )
X : MTreg value

https://github.com/xoseperez/espurna/blob/079c4305c0d00dc5ca940a8a556c59dc9fe0d3be/code/espurna/sensors/BH1750Sensor.h#L122

davebuk, 2022-11-06:
If other magnitudes use `***Precision` go with that. Otherwise `luxDecPoint` sounds good.

davebuk, 2022-11-06:
From the spec sheet it looks like the smallest decimal value is 0.5. Can espurna make the resolution 0.1?

- H-reslution mode2 : Illuminance per 1 count ( lx / count ) = 1 / 1.2 *( 69 / X ) / 10

The device I'm using switches a light ON when it gets dark during the day and back OFF again when it gets light again. The problem is around dusk when the light comes ON, it increases the LUX level and turns the light back OFF until its dark enough to stay ON. I'm hoping with the decimal value I can get a better "range of values" during intermediate brightness levels.

commit b8921b9a (2022-12-04): bh1750: more decimals for mode2

return half of raw value in both modes involving it
resolve #2550 (just for the sensor, no global setting yet)
closed 2022-12-04 by mcspr