xoseperez/espurna #2555: Problem dht11 Module

limecat-ss, 2022-11-13:
### Device

Wemos d1 Mini

### Version

1.15.0-dev

### Bug description

Hi,

I can't make dht11 work with the module. I always get timeout. It doesn't seem to be a hardware problem because with the Arduino IDE it works perfectly.

### Build tools used

PlatformIO

### Any relevant log output (when available)

```shell
[970722] [SENSOR] Could not read from DHT11 @ GPIO14 (Timeout)
[980728] [SENSOR] Could not read from DHT11 @ GPIO14 (Timeout)
[990736] [SENSOR] Could not read from DHT11 @ GPIO14 (Timeout)
```

mcspr, 2022-11-13:
What is the IDE example code & library?
Did you set `DHT_TYPE` to `DHT_CHIP_DHT11`?

limecat-ss, 2022-11-13:
I used the example library for dht11 in the arduino IDE.

My configuration in platformio.ini is:

```
build_src_flags = 
     -DWEMOS_D1_MINI -DNOWSAUTH
     -DDHT_SUPPORT=1 -DDHT_PIN=14 -DHT_TYPE=DHT_CHIP_DHT11

```

mcspr, 2022-11-13:
> 1.15.0-dev

How did you get the source? Did you just get dev branch .zip?

Does using DHTSensor.h from here -> https://raw.githubusercontent.com/xoseperez/espurna/f44aab5fdbf4558c620ba5d4c5a997cff3e980fe/code/espurna/sensors/DHTSensor.h <- change anything?

Does it work if these lines are removed?
https://github.com/xoseperez/espurna/blob/1169be25a55401d42e3b3a465dbb1dc9f8c2f1df/code/espurna/gpio.cpp#L843-L899

mcspr, 2022-11-13:
...although, looking at the .h I mentioned, what about this change?
(open code/espurna/system.cpp and find the corresponding `void blockingDelay` func, and change `false` to `true`)
```diff
diff --git a/code/espurna/system.cpp b/code/espurna/system.cpp                                                                                                              index 067e37d5..8080ad34 100644                                                                                                                                             --- a/code/espurna/system.cpp                                                                                                                                               +++ b/code/espurna/system.cpp                                                                                                                                               @@ -315,7 +315,7 @@ bool tryDelay(CoreClock::time_point start, CoreClock::duration timeout, CoreCloc                                                                                                                                                                                                                                                     void blockingDelay(CoreClock::duration timeout, CoreClock::duration interval) {                                                                                                 blockingDelay(timeout, interval, []() {                                                                                                                                -        return false;                                                                                                                                                      +        return true;                                                                                                                                                            });                                                                                                                                                                     }                                                                                                                                                                           
```
Does not seem it waits for DHT properly

limecat-ss, 2022-11-14:
It works ! Changing false to true makes me get Temperature and Humidity correctly. Thanks !

commit 22791cfc (2022-11-16): system: blocking delay should actually call delay()

avoid optimizing out function call when `&&` comparison is always false
resolve #2555
closed 2022-11-14 by limecat-ss