xoseperez/espurna #2563: PUTTY broken on snapshot builds

mcspr, 2022-12-05:
See #2562 
(and possibly any other 'interactive' client that splits data packets like PUTTY does)

Server processing ignores special TELNET newlines sent by PUTTY (fixed in PUTTY config, though)
Server processing ignores packets without newlines (either `\r\n` or `\n`)
https://github.com/xoseperez/espurna/blob/988a9724842ddcd7c9a7c53283875b51f35bd6d7/code/espurna/telnet.cpp#L538-L540
https://github.com/xoseperez/espurna/blob/988a9724842ddcd7c9a7c53283875b51f35bd6d7/code/espurna/telnet.cpp#L569-L574

Instead, we get two packets instead of one as with something like nmap-nc
```
121	20:55:15,033230	HOST	ESP8266	TELNET	59	Telnet Data ...
> hello
122	20:55:15,033380	HOST	ESP8266	TELNET	56	Telnet Data ...
> \r\n
```
I see no option in putty to not do this kind of stuff, but it is possible there is something

Algorithm above could either feed internal parser repeatedly, or keep our pbufs for some amount of time and repeat parsing line-like data once more... data is available

Server also always sends `\n`, but this is mostly a config issue (implicit CR after LF)

closed 2022-12-06 by mcspr

Sag06, 2022-12-11:
I'm sorry to say it is still broken.
1.- No possible to connect with Telnet password. Always rejected.
2.- Not a big issue, but when Telnet connected (without authentication), first command doesn´t work, I have to repeat it.
```
[855725] [TELNET] Connected 10.12.16.188:51381
                                              get hostname
-ERROR: TERMINAL: UnterminatedQuote
                                   get hostname
> hostname => "ESPURNA-5BCBEE-Garaje"
                                     +OK
```
```
[241055] [TELNET] Connected 10.12.16.188:52197
                                              ota http://10.12.16.188:8080/espurna.BLITZWOLF_BWSHPX.221208.gz.bin
-ERROR: TERMINAL: UnterminatedQuote
                                   ota http://10.12.16.188:8080/espurna.BLITZWOLF_BWSHPX.221208.gz.bin
[263766] [OTA] Connecting to 10.12.16.188:8080
                                              +OK
                                                 [263774] [EEPROM] Disabling EEPROM rotation
            [263774] [OTA] Downloading /espurna.BLITZWOLF_BWSHPX.221208.gz.bin
                                                                              [2[272192] [OTA] Progress:  420480 bytes
         [272262] [OTA] Success:  428262 bytes
                                              [272263] [MAIN] Requested reset: Reboot after a successful OTA update
                                   [272266] [OTA] Disconnected
                                   ```
                                   Tried on three devices.

mcspr, 2022-12-11:
iirc PuTTY should be in RAW socket and not telnet to avoid a bunch of assumptions it makes about the data. Plus, Terminal > Implicit cr in every lf would make output much nicer.
Could you share `HKEY_CURRENT_USER\Software\SimonTatham\PuTTY\Sessions\{YOUR SESSION NAME}`?
(<kbd>Win</kbd>+<kbd>R</kbd> -> `regedit.exe` -> export; either c/p text contents of .reg or attach it here)

Sag06, 2022-12-11:
```
Windows Registry Editor Version 5.00

[HKEY_CURRENT_USER\SOFTWARE\SimonTatham\PuTTY\Sessions\zero]
"Present"=dword:00000001
"HostName"="10.12.16.160"
"LogFileName"="putty.log"
"LogType"=dword:00000000
"LogFileClash"=dword:ffffffff
"LogFlush"=dword:00000001
"LogHeader"=dword:00000001
"SSHLogOmitPasswords"=dword:00000001
"SSHLogOmitData"=dword:00000000
"Protocol"="telnet"
"PortNumber"=dword:00000017
"CloseOnExit"=dword:00000001
"WarnOnClose"=dword:00000001
"PingInterval"=dword:00000000
"PingIntervalSecs"=dword:00000000
"TCPNoDelay"=dword:00000001
"TCPKeepalives"=dword:00000000
"TerminalType"="xterm"
"TerminalSpeed"="38400,38400"
"TerminalModes"="CS7=A,CS8=A,DISCARD=A,DSUSP=A,ECHO=A,ECHOCTL=A,ECHOE=A,ECHOK=A,ECHOKE=A,ECHONL=A,EOF=A,EOL=A,EOL2=A,ERASE=A,FLUSH=A,ICANON=A,ICRNL=A,IEXTEN=A,IGNCR=A,IGNPAR=A,IMAXBEL=A,INLCR=A,INPCK=A,INTR=A,ISIG=A,ISTRIP=A,IUCLC=A,IUTF8=A,IXANY=A,IXOFF=A,IXON=A,KILL=A,LNEXT=A,NOFLSH=A,OCRNL=A,OLCUC=A,ONLCR=A,ONLRET=A,ONOCR=A,OPOST=A,PARENB=A,PARMRK=A,PARODD=A,PENDIN=A,QUIT=A,REPRINT=A,START=A,STATUS=A,STOP=A,SUSP=A,SWTCH=A,TOSTOP=A,WERASE=A,XCASE=A"
"AddressFamily"=dword:00000000
"ProxyExcludeList"=""
"ProxyDNS"=dword:00000001
"ProxyLocalhost"=dword:00000000
"ProxyMethod"=dword:00000000
"ProxyHost"="proxy"
"ProxyPort"=dword:00000050
"ProxyUsername"=""
"ProxyPassword"=""
"ProxyTelnetCommand"="connect %host %port\\n"
"ProxyLogToTerm"=dword:00000001
"Environment"=""
"UserName"=""
"UserNameFromEnvironment"=dword:00000000
"LocalUserName"=""
"NoPTY"=dword:00000000
"Compression"=dword:00000000
"TryAgent"=dword:00000001
"AgentFwd"=dword:00000000
"GssapiFwd"=dword:00000000
"ChangeUsername"=dword:00000000
"Cipher"="aes,chacha20,3des,WARN,des,blowfish,arcfour"
"KEX"="ecdh,dh-gex-sha1,dh-group14-sha1,rsa,WARN,dh-group1-sha1"
"HostKey"="ed25519,ecdsa,rsa,dsa,WARN"
"PreferKnownHostKeys"=dword:00000001
"RekeyTime"=dword:0000003c
"GssapiRekey"=dword:00000002
"RekeyBytes"="1G"
"SshNoAuth"=dword:00000000
"SshBanner"=dword:00000001
"AuthTIS"=dword:00000000
"AuthKI"=dword:00000001
"AuthGSSAPI"=dword:00000001
"AuthGSSAPIKEX"=dword:00000001
"GSSLibs"="gssapi32,sspi,custom"
"GSSCustom"=""
"SshNoShell"=dword:00000000
"SshProt"=dword:00000003
"LogHost"=""
"SSH2DES"=dword:00000000
"PublicKeyFile"=""
"RemoteCommand"=""
"RFCEnviron"=dword:00000000
"PassiveTelnet"=dword:00000000
"BackspaceIsDelete"=dword:00000001
"RXVTHomeEnd"=dword:00000000
[... 140 lines cut ...]
"BoldFontCharSet"=dword:00000000
"BoldFontHeight"=dword:00000000
"WideFont"=""
"WideFontIsBold"=dword:00000000
"WideFontCharSet"=dword:00000000
"WideFontHeight"=dword:00000000
"WideBoldFont"=""
"WideBoldFontIsBold"=dword:00000000
"WideBoldFontCharSet"=dword:00000000
"WideBoldFontHeight"=dword:00000000
"ShadowBold"=dword:00000000
"ShadowBoldOffset"=dword:00000001
"SerialLine"="COM1"
"SerialSpeed"=dword:00002580
"SerialDataBits"=dword:00000008
"SerialStopHalfbits"=dword:00000002
"SerialParity"=dword:00000000
"SerialFlowControl"=dword:00000001
"WindowClass"=""
"ConnectionSharing"=dword:00000000
"ConnectionSharingUpstream"=dword:00000001
"ConnectionSharingDownstream"=dword:00000001
"SSHManualHostKeys"=""

```

mcspr, 2022-12-12:
> > iirc PuTTY should be in RAW socket and not telnet to avoid a bunch of assumptions it makes about the data. Plus, Terminal
> Yes! It works on Raw mode, 23 Port.! Wonderful!

So, settings I mentioned above are enough?
https://github.com/xoseperez/espurna/wiki/Telnet#description now mentions raw mode for putty btw

commit 542188dd (2022-12-06): telnet: buffer when eol is missing

resolve #2563
commit f99b7b2d (2024-06-13): telnet: best-effort skip of command data

usually a good idea for clients like PuTTY, since these are sent by default
in the very first packet even before user types anything else
also restore iac+eof handling, which call close() on the stream

resolve #2563
resolve #2606
commit fa7c1875 (2024-06-15): telnet: best-effort skip of command data

usually a good idea for clients like PuTTY, since these are sent by default
in the very first packet even before user types anything else
also restore iac+eof handling, which call close() on the stream

resolve #2563
resolve #2606