xoseperez/espurna #2565: Tuya generic dimmer not working for all incoming values.

davebuk, 2022-12-14:
### Device

Tuya Generic Dimmer

### Version

1.15.0-dev.git3b489d0f

### Question

 I have updated all my devices to the latest dev build but the `TUYA DIMMER` dimming via mqtt is not working correctly.  I can adjust the 0-255 value on the webUI and that works. When I send a new brightness command from openHAB, the value set 0-100% gets received by the espurna device from openHAB as a value between 0-255.

The issue seems to be most values are received as a decimal. These are the ones that don't change the brightness value physically or in the webUI.

Starting at 40% (102) I step through 5% increases, but nothing happens until I get to 60% (153). I have `tuyaFilter 2` set 
```
[953745] [TUYA] =>: 55aa00000000ff
[953769] [TUYA] <=: 55aa000000010101
[955842] [MQTT] Received TUYA001/brightness/set => 102
[962753] [TUYA] =>: 55aa00000000ff
[962777] [TUYA] <=: 55aa000000010101
[964278] [MQTT] Received TUYA001/brightness/set => 114.75
[965918] [MQTT] Received TUYA001/brightness/set => 127.5
[966876] [MQTT] Received TUYA001/brightness/set => 140.25
[967844] [MQTT] Received TUYA001/brightness/set => 153
[967849] [LIGHT] Scheduled transition for 2000 (ms) every 100 (ms)
[967850] [LIGHT] Transition from 102.00 to 153 (step 2.55, 20 times)
[967955] [MQTT] Sending TUYA001/channel/0 => 153 (PID 1)
[967956] [MQTT] Sending TUYA001/brightness => 153 (PID 1)
[967957] [MQTT] Sending TUYA001/light => 1 (PID 1)
[967959] [TUYA] =>: 55aa0006000802020004000000687d
[967992] [TUYA] <=: 55aa0007000802020004000000687e
[968004] [TUYA] <=: 55aa0007000501010001010f
[968027] [TUYA] <=: 55aa0007000802020004000000788e
[968060] [TUYA] =>: 55aa00060008020200040000006b80
[968094] [TUYA] <=: 55aa00070008020200040000006b81
[968107] [TUYA] <=: 55aa0007000501010001010f
[968131] [TUYA] <=: 55aa00070008020200040000007b91
[968155] [TUYA] =>: 55aa00060008020200040000006d82
[968190] [TUYA] <=: 55aa00070008020200040000006d83
[968255] [TUYA] =>: 55aa00060008020200040000007085
[968290] [TUYA] <=: 55aa00070008020200040000007086
[968303] [TUYA] <=: 55aa0007000501010001010f
[968317] [TUYA] <=: 55aa00070008020200040000008096
[968353] [TUYA] =>: 55aa00060008020200040000007287
[968387] [TUYA] <=: 55aa00070008020200040000007288
[968452] [TUYA] =>: 55aa0006000802020004000000758a
[968487] [TUYA] <=: 55aa0007000802020004000000758b
[968501] [TUYA] <=: 55aa0007000501010001010f
[968514] [TUYA] <=: 55aa00070008020200040000008399
[968559] [TUYA] =>: 55aa0006000802020004000000778c
[968594] [TUYA] <=: 55aa0007000802020004000000778d
[968607] [TUYA] <=: 55aa0007000501010001010f
[968621] [TUYA] <=: 55aa0007000802020004000000869c
[968655] [TUYA] =>: 55aa00060008020200040000007a8f
[968690] [TUYA] <=: 55aa00070008020200040000007a90
[968703] [TUYA] <=: 55aa0007000501010001010f
[968717] [TUYA] <=: 55aa0007000802020004000000899f
[968751] [TUYA] =>: 55aa00060008020200040000007c91
[968785] [TUYA] <=: 55aa00070008020200040000007c92
[968799] [TUYA] <=: 55aa0007000501010001010f
[968813] [TUYA] <=: 55aa00070008020200040000008ca2
[968857] [TUYA] =>: 55aa00060008020200040000007f94
[968891] [TUYA] <=: 55aa00070008020200040000007f95
[968905] [TUYA] <=: 55aa0007000501010001010f
[968918] [TUYA] <=: 55aa00070008020200040000008ea4
[968952] [TUYA] =>: 55aa00060008020200040000008297
[968987] [TUYA] <=: 55aa00070008020200040000008298
[969000] [TUYA] <=: 55aa0007000501010001010f
[969014] [TUYA] <=: 55aa000700080202000400000091a7
[969058] [TUYA] =>: 55aa00060008020200040000008499
[969093] [TUYA] <=: 55aa0007000802020004000000849a
[969106] [TUYA] <=: 55aa0007000501010001010f
[969120] [TUYA] <=: 55aa000700080202000400000094aa
[969155] [TUYA] =>: 55aa0006000802020004000000879c
[969189] [TUYA] <=: 55aa0007000802020004000000879d
[969203] [TUYA] <=: 55aa0007000501010001010f
[969216] [TUYA] <=: 55aa000700080202000400000097ad
[969251] [TUYA] =>: 55aa0006000802020004000000899e
[969285] [TUYA] <=: 55aa0007000802020004000000899f
[969350] [TUYA] =>: 55aa00060008020200040000008ca1
[969385] [TUYA] <=: 55aa00070008020200040000008ca2
[969399] [TUYA] <=: 55aa0007000501010001010f
[969413] [TUYA] <=: 55aa00070008020200040000009cb2
[969457] [TUYA] =>: 55aa00060008020200040000008ea3
[969492] [TUYA] <=: 55aa00070008020200040000008ea4
[969557] [TUYA] =>: 55aa000600080202000400000091a6
[969592] [TUYA] <=: 55aa000700080202000400000091a7
[969606] [TUYA] <=: 55aa0007000501010001010f
[969620] [TUYA] <=: 55aa00070008020200040000009fb5
[969655] [TUYA] =>: 55aa000600080202000400000093a8
[969690] [TUYA] <=: 55aa000700080202000400000093a9
[969703] [TUYA] <=: 55aa0007000501010001010f
[969717] [TUYA] <=: 55aa0007000802020004000000a2b8
[969751] [TUYA] =>: 55aa000600080202000400000096ab
[969786] [TUYA] <=: 55aa000700080202000400000096ac
[969799] [TUYA] <=: 55aa0007000501010001010f
[969813] [TUYA] <=: 55aa0007000802020004000000a5bb
[969858] [TUYA] =>: 55aa000600080202000400000099ae
[969893] [TUYA] <=: 55aa000700080202000400000099af
[969906] [TUYA] <=: 55aa0007000501010001010f
[969920] [TUYA] <=: 55aa0007000802020004000000a8be
[971755] [TUYA] =>: 55aa00000000ff
[971779] [TUYA] <=: 55aa000000010101
```
`keys` are:
```
keys
> adminPass => "***"
> brightness => "153"
> btnGpio0 => "0"
> cfg => "13"
> ch0 => "255"
> dns0 => "192.168.1.1"
> gw0 => "192.168.1.1"
> ip0 => "192.168.1.***"
> ledGpio0 => "14"
> ledMode0 => "2"
> ltRelay => "1"
> ltSave => "0"
> ltStep => "100"
> ltTime => "2000"
> mask0 => "255.255.255.0"
> mireds => "326"
> mqttClientID => "Tuya-001"
> mqttEnabled => "1"
> mqttPassword => "***"
> mqttServer => "***"
> mqttTopic => "TUYA001"
> mqttUser => "***"
> pass0 => "***"
> ssid0 => "***"
> tuyaChanState => "1"
> tuyaFilter => "2"
> tuyaSwitch0 => "1"
> useCCT => "0"
> useColor => "0"
> useTransitions => "1"
> useWhite => "0"
> wifiScan => "1"
Number of keys: 32
```

I was under the impression the `tuyaFilter` should allow the incoming values to be set to non-decimal numbers. Is that correct?

mcspr, 2022-12-14:
Looking at logs... value parsing was not as strict. Not allowing it to encounter the '.' should help

```diff
diff --git a/code/espurna/light.cpp b/code/espurna/light.cpp
index 8aa3f98a..6ff302fa 100644
--- a/code/espurna/light.cpp
+++ b/code/espurna/light.cpp
@@ -1571,6 +1571,11 @@ String _lightGroupPayload() {
 // +offset, -offset or the new value

 long _lightAdjustValue(long value, espurna::StringView operation) {
+    const auto dot = std::find(operation.begin(), operation.end(), '.');
+    if (dot != operation.end()) {
+        operation = espurna::StringView(operation.begin(), dot);
+    }
+
     if (operation.length()) {
         switch (operation[0]) {
         case '+':
```

> tuyaFilter => "2"

> I was under the impression the tuyaFilter should allow the incoming values to be set to non-decimal numbers. Is that correct?

Value is boolean since 8ceeebdb24aa9bd863fa28676a0364497ec193ae, as I understand it makes us ignore every value coming from the device when set to `true` instead of the old `2` meaning 'ignore INT values' (before the commit)
https://github.com/xoseperez/espurna/blob/3b489d0f25f78d97ee8a374a9b7de2d67f1e6a73/code/espurna/tuya.cpp#L674-L675
https://github.com/xoseperez/espurna/blob/3b489d0f25f78d97ee8a374a9b7de2d67f1e6a73/code/espurna/tuya.cpp#L346-L347
Also a case of strict parser, only `true`, `y`, `yes`, `on` or `1` would mean `true`. Set to `2` it means `false`, so every value is 'updated'.

davebuk, 2022-12-15:
That seems to work.

'tuyaFilter' key removed from `keys`.

40-60% now gives:
```
[130135] [MQTT] Received TUYA001/brightness/set => 114.75
[130154] [TUYA] =>: 55aa00060008020200040000007287
[130189] [TUYA] <=: 55aa00070008020200040000007288
[130203] [TUYA] <=: 55aa0007000501010001010f
[130226] [TUYA] <=: 55aa00070008020200040000008096
[130249] [MQTT] Sending TUYA001/channel/0 => 114 (PID 1)
[130250] [MQTT] Sending TUYA001/brightness => 114 (PID 1)
[130686] [MQTT] Received TUYA001/brightness/set => 127.5
[130709] [TUYA] =>: 55aa00060008020200040000007f94
[130744] [TUYA] <=: 55aa00070008020200040000007f95
[130758] [TUYA] <=: 55aa0007000501010001010f
[130772] [TUYA] <=: 55aa00070008020200040000008ea4
[130807] [MQTT] Sending TUYA001/channel/0 => 127 (PID 1)
[130808] [MQTT] Sending TUYA001/brightness => 127 (PID 1)
[131162] [MQTT] Received TUYA001/brightness/set => 140.25
[131178] [TUYA] =>: 55aa00060008020200040000008ca1
[131213] [TUYA] <=: 55aa00070008020200040000008ca2
[131226] [TUYA] <=: 55aa0007000501010001010f
[131249] [TUYA] <=: 55aa00070008020200040000009cb2
[131274] [MQTT] Sending TUYA001/channel/0 => 140 (PID 1)
[131275] [MQTT] Sending TUYA001/brightness => 140 (PID 1)
[131702] [MQTT] Received TUYA001/brightness/set => 153
[131724] [TUYA] =>: 55aa000600080202000400000099ae
[131758] [TUYA] <=: 55aa000700080202000400000099af
[131772] [TUYA] <=: 55aa0007000501010001010f
[131795] [TUYA] <=: 55aa0007000802020004000000a8be
[131820] [MQTT] Sending TUYA001/channel/0 => 153 (PID 1)
[131821] [MQTT] Sending TUYA001/brightness => 153 (PID 1)
[133941] [TUYA] =>: 55aa00000000ff
[133965] [TUYA] <=: 55aa000000010101
```

closed 2022-12-19 by mcspr