xoseperez/espurna #2568: Relay is missing on MINI_SMART_SOCKET

kuppe234, 2022-12-29:
### Device

MINI_SMART_SOCKET

### Version

1.15.0-dev.git68310b6d

### Question

Hello,
I build the firmware for MINI_SMART_SOCKET by myself. But I´m missing the Relay. I can switch the LEDs and change the color, but can´t switch the relay. The Website only show one switch for the LEDs. 
I have standard config and this custom.h:
```
#undef ALEXA_SUPPORT
#define ALEXA_SUPPORT           0

#undef DOMOTICZ_SUPPORT
#define DOMOTICZ_SUPPORT        0

#undef THINGSPEAK_SUPPORT
#define THINGSPEAK_SUPPORT      0

#undef API_SUPPORT
#define API_SUPPORT             0
```
The relay is on GPIO15. 
If I call "gpio 15" in DEBUG or telnet, I get 
`   INPUT @ GPIO15 (LOW)`
The GPIO15 must be an output. What´s wrong?

How can a get back the relay-switch? Any idea?

Greetings, kuppe

mcspr, 2022-12-29:
Maybe something claiming it first? check out `gpio.locks` command output

kuppe234, 2022-12-29:
gpio.locks
* hardware GPIO14	88:/scratch/Projekte/espurna/code/espurna/sensors/BaseSensor.h:_attach
* hardware GPIO5	88:/scratch/Projekte/espurna/code/espurna/sensors/BaseSensor.h:_attach
* hardware GPIO13	1334:/scratch/Projekte/espurna/code/espurna/button.cpp:_buttonGpioPin
* hardware GPIO2	405:/scratch/Projekte/espurna/code/espurna/pwm.cpp:init
* hardware GPIO4	405:/scratch/Projekte/espurna/code/espurna/pwm.cpp:init
* hardware GPIO0	405:/scratch/Projekte/espurna/code/espurna/pwm.cpp:init
* hardware GPIO3	362:/scratch/Projekte/espurna/code/espurna/uart.cpp:make_port
* hardware GPIO1	358:/scratch/Projekte/espurna/code/espurna/uart.cpp:make_port

mcspr, 2022-12-29:
`get relayGpio0`?

kuppe234, 2022-12-29:
get relayGpio0
> relayGpio0 => 15 (default)

mcspr, 2022-12-29:
Ah, I got it. `set ltRelay 0` and `reset` (or, see LIGHTS setting related to relays in webui)

The gist of it is that relay settings are simply not used. At boot, Lights create a virtual relay (ID 0). Then, when Relays init the relay ID 0 is already taken so no relays are generated. If there as `RELAY2_PIN 15` (or `relayGpio1 15` setting), then it would've worked

Our .h(s) should just do `#define LIGHT_RELAY_ENABLED 0` to set this as default at build time, don't really see any reason to keep `1` as default (besides backwards compat). Old versions needed relay to toggle lights, new ones don't

kuppe234, 2022-12-30:
`set ltRelay 0` and `reset` is working. But if I enable `Light state switch` in LIGHTS the relay is gone.
Then I modifyed `RELAY1_PIN` to `RELAY2_PIN` and `RELAY1_TYPE` to `RELAY2_TYPE` in hardware.h, because I need the `Light state switch`. It is enabled and I have 2 switches in STATUS like in old versions.
If I now disable `Light state switch`, the relay-switch is gone. Then I must call 'del ltRelay` in DEBUG, because there is no web option anymore.
Thank you for your support. But I think in future versions the relay function must have priority over the light function.

mcspr, 2022-12-30:
True. But, on the other hand, lights would be ordered last. Either way we have a 'hidden' relay injected but something else

Making this an explicit config is also what I had in mind
```cpp
#define RELAY1_PROVIDER RELAY_PRIVIDER_LIGHT_STATE
// default is RELAY_PROVIDER_GPIO for hardware pins
```
Or, just rely on existing dummy ones
```cpp
#define RELAY1_PROVIDER RELAY_PRIVIDER_DUMMY
#define LIGHT_STATE_RELAY 1 // like buttons are linked 
```

Hidden WebUI setting looks like a bug, we don't announce relay support to UI without any relays already configured :oops:

commit b5a03dac (2023-01-11): light: experiment with relay state provider

* dynamic relays should not overlap with configuration
* prefer explicit relay configuration vs. dynamic addition

resolve #2568
closed 2023-01-11 by mcspr