xoseperez/espurna #2571: ad-hoc network available even if connected to infrastructure

adoeller, 2023-01-07:
### Device

itead-s20

### Version

1.15.0-dev.git312c3ef2+github221212

### Bug description

ad-hoc network available even if connected to infrastructure

### Steps to reproduce

device is in network ESSID A
stronger network ESDID A offered
switch off stronger netzwork A
restart original network A, i.e. for a minutes there is no network whatsoever available

=> device will connect to original network A but still offer ad-hoc wifi

mcspr, 2023-01-07:
By offered you mean it decides to re-connect and in the middle of it there is a disconnect? Why do we need 2, and is it different with just one A disappearing?
Fallback AP may appear, but there is a 60s timeout if there are no clients connected (see `wifi` and `wifi.stations` commands output)

adoeller, 2023-01-08:
No, I mean that ad-hoc and infrastructure will be permanently available. Even after 15 minutes.

mcspr, 2023-01-09:
Sounds like a bug in timers, then. Just not yet seeing the trigger...

commit f387b864 (2023-01-11): relay: scheduling from system timer does not work

also queue fallback check instead of resolving it in the timer
resolve #2571
closed 2023-01-11 by mcspr