xoseperez/espurna #2572: HA discover adds random sensors from time to time

Sag06, 2023-01-11:
### Device

BLITZWOLF BW-SHP6

### Version

dev_68310b6  SDK version 2.2.2-dev(38a443e). Core version 3.0.2

### Bug description

HA discover adds random sensors from time to time to home assistant (MQTT broker).
Sensor names are mostly "Unknown XXX" name, but can be other as "Analog XXX", "Voc98", etc.
Example (Name State) sensors of one of my B-:SHP6:
Analog 103 Unknown
Analog 128  Unknown
Apparent 0  0 VA
Energy 0  0,353 kWh
Energy_delta 0  0 J
Factor 0  0%
Power 0  0 W
Pressure 0  Unknown
Reactive 0  0 VAR
Unknown 0  Unknown
Unknown 118 Unknown
Unknown 13  Unknown
Unknown 187  Unknown
Unknown 23  Unknown
Unknown 46  Unknown
Unknown 52  Unknown
Unknown 67  Unknown
Unknown 71  Unknown
Unknown 72  Unknown
Unknown 93  Unknown
Voc 98  Unknown
Voltage 0  0 V
Unknown 208  Unknown

Dev compilation on 2022/12/20. 
Tried on four BW-SHP6 and two different houses/routers/Home assistant servers.
I've deleted devices and added again BW-SHP6 from Home Assistant. Same results.  New sensors appear from time to time.
Never happened before with core <3.
Not sure how to debug to help

[mqtt-0ef2bff31883feba54d29e3b1cd29370-BW_SHP6_ESPURNA_48ED71-736521b70981214631fcfc04171e95e7.json (1).txt](https://github.com/xoseperez/espurna/files/10393016/mqtt-0ef2bff31883feba54d29e3b1cd29370-BW_SHP6_ESPURNA_48ED71-736521b70981214631fcfc04171e95e7.json.1.txt)
[mqtt-0ef2bff31883feba54d29e3b1cd29370-ESPURNA_5D1273-cf79da4d77b9bc2b246282207c327079.json.txt](https://github.com/xoseperez/espurna/files/10393017/mqtt-0ef2bff31883feba54d29e3b1cd29370-ESPURNA_5D1273-cf79da4d77b9bc2b246282207c327079.json.txt)
[mqtt-0ef2bff31883feba54d29e3b1cd29370-BW_SHP6_ESPURNA_1568A8_Frigo-b84bb75d1fb41de58046951359b56b87.json.txt](https://github.com/xoseperez/espurna/files/10393018/mqtt-0ef2bff31883feba54d29e3b1cd29370-BW_SHP6_ESPURNA_1568A8_Frigo-b84bb75d1fb41de58046951359b56b87.json.txt)

### Steps to reproduce

Delete and add again BW-SHP6 from Home Assistant. HA discover. MQTT.

### Build tools used

Arduino IDE

mcspr, 2023-01-11:
Notice there is no 'Current', I'd guess we just send some random data instead of the very first magnitude. Quick fix should be
```diff
diff --git a/code/espurna/homeassistant.cpp b/code/espurna/homeassistant.cpp
index 0358bcda..74122636 100644
--- a/code/espurna/homeassistant.cpp
+++ b/code/espurna/homeassistant.cpp
@@ -776,7 +776,7 @@ private:

     unsigned char _magnitudes { 0u };
     unsigned char _index { 0u };
-    sensor::Info _info;
+    sensor::Info _info { magnitudeInfo(_index) };

     String _unique_id;
     String _name;
```

Sag06, 2023-01-11:
Yes. That did the trick. No random sensors this afternoon. Thank you again!

commit d798052d (2023-01-12): ha: fix uninitialized sensor info

and double-check that we really have magnitudes available
fix #2572
closed 2023-01-11 by Sag06