xoseperez/espurna #2574: SONOFF_T1_2CH relay switches lock up

tita2000, 2023-01-13:
[image]

my configuration for the relays looks like this

[image]

the moment I reboot, the first switch switches on, pulses for 30 seconds and switches off, although I configured boot mode off. Then after a short while maybe about 5 seconds both switches turn grey.

Regards Timo

tita2000, 2023-01-13:
here is the debug output of the issue, hope it is useful 

`[026447] [WEBSOCKET] #1 connected, ip: 192.168.100.1, url: /ws
[060197] [MAIN] Resetting stability counter
[084105] [MQTT] Sending ESPROL08/relay/0 => 1 (PID 7)
[084107] [MQTT] Sending ESPROL08/relay/0/set => 1 (PID 8)
[084110] [RELAY] #0 pulse OFF scheduled in 30000 (ms)
[084111] [RELAY] #0 set to ON
[084159] [MQTT] Received ESPROL08/relay/0 => 1
[084165] [MQTT] Received ESPROL08/relay/0/set => 1
[085120] [RELAY] Relay mask: 0b1
[114174] [MQTT] Sending ESPROL08/relay/0 => 0 (PID 9)
[114176] [MQTT] Sending ESPROL08/relay/0/set => 0 (PID 10)
[114180] [RELAY] #0 set to OFF
[114227] [MQTT] Received ESPROL08/relay/0 => 0
[114233] [MQTT] Received ESPROL08/relay/0/set => 0
[115189] [RELAY] Relay mask: 0b0`

Regards

mcspr, 2023-01-13:
> the moment I reboot, the first switch switches on,

Do you mean on boot it automatically switches relay to ON without any interaction?

> Ok, i figured out, that I have to set the INTERLOCK DELAY greater than 0 for normal operation. If this is a feature, feel free to close this issue.

Certainly a bug, if it fails to unlock relays after locking them b/c of the sync mode.

Maybe something like this needs to happen instead
*(note to self; or, if you are able to build, to try out)*
```diff
diff --git a/code/espurna/relay.cpp b/code/espurna/relay.cpp
index 48bc377e..0271d3f9 100644
--- a/code/espurna/relay.cpp
+++ b/code/espurna/relay.cpp
@@ -1681,10 +1681,10 @@ void _relaySyncUnlock() {
 #endif
         };

-        if (all_off) {
-            _relay_sync_timer.schedule(_relay_delay_interlock, action);
-        } else {
+        if (!all_off || !_relay_delay_interlock.count()) {
             action();
+        } else {
+            _relay_sync_timer.schedule(_relay_delay_interlock, action);
         }
     }
 }
```

Still prone to some timing issues, but at least locks would be disabled