xoseperez/espurna #2581: Espurna style WB2S - BK7231 chips support.

davebuk, 2023-03-12:
### Description

It's seems quite a few smart devices are moving away from ESP controllers and use BK7231 chips instead. Would there likely be espurna style firmware in the future?

### Solution

The following site has one solution: 
https://github.com/openshwprojects/OpenBK7231T_App

### Alternatives

It seems the BK7231 chips can be directly replaced with ESP-02S boards. This involves de-soldering the old chip board and soldering in a new one. At least they could be flashed as normal putting espurna firmware on them.

mcspr, 2023-03-14:
I'd also add a link to https://github.com/kuba2k2/libretuya and the accompanying https://docs.libretuya.ml/docs/projects/esphome/ which supports both BK72xx and Realtek chips used in Tuya devices.

Not sure I am in the market of buying any of those, though, and developing anything without a hw around is fairly non-exiting to even think about. Port should be fairly straightforward, these do not add any extra hw features otherwise and can be treated as almost a esp8266 clone.
(have to look at actual differences some more)

iprak, 2023-03-17:
The [OpenBK7231T_App](https://github.com/openshwprojects/OpenBK7231T_App) firmware is not as polished but it targets a variety of variants and is quite functional. I would give it a try. I understand Libretuya only handles a specific set of features found in esphome.

kuba2k2, 2023-03-18:
We're still enhancing the ESPHome support in LibreTuya, for example the remote receiver component was fixed lately.

davebuk, 2023-03-18:
I've not bought any new devices yet and I have just stock piled a few ESP WiFi sockets.

I will carry on using Espurna on my home made D1mini devices, but was looking to the future when the standard plug in WiFi sockets all come with other chips.

For me, as long as the devices support MQTT, thats all I generally need. I think OpenBK7231T now has MQTT support but its still early days where as Espurna has been working solidly for me for a few years.

I'll keep an eye on progress on github.

kuba2k2, 2023-03-18:
OpenBK7231 is around a year old now, so I presume it's stable enough for daily usage. If you need MQTT, then ESPHome-LibreTuya won't work for you yet, because MQTT still has issues.

iprak, 2023-03-18:
OpenBK7231T_App is quite decent but also a work in progress. There is active support for various chips and sensors, lights. Also the project supports not just BK7231T but also variants like W600/W800/etc ([device database](https://openbekeniot.github.io/webapp/devicesList.html))