xoseperez/espurna #2582: espurna 1.15  does not connect to a wi-fi hotspot with a fixed ip

limecat-ss, 2023-03-13:
### Device

Wemos d1 mini

### Version

espurna-1.15.0-dev.git909949b5

### Bug description

Hi,

Espurna 1.15  does not connect to a wi-fi hotspot with a fixed ip. Version 1.14.1 connects normally on the same network.

### Any relevant log output (when available)

```shell
[235257] [WIFI] State WaitConnected -> Timeout
[235257] [WIFI] Next connection attempt in 3000 (ms)
[235258] [WIFI] Connection timeout
[235271] [WIFI] State Timeout -> Idle
[238273] [WIFI] State Idle -> Connect
[238276] [WIFI] Connecting
[238290] [WIFI] State Connect -> WaitConnected
[242148] [WIFI] State WaitConnected -> Timeout
[242161] [WIFI] State Timeout -> Fallback
[242162] [WIFI] Next connection attempt in 120000 (ms)
[242163] [WIFI] Reconnecting
[242176] [WIFI] State Fallback -> Idle
```

mcspr, 2023-03-13:
Any relevant info from hotspot logs?
Device can also be asked to log SDK routines; `set dbgSDK 1` and `reset` in console and see additional serial output

works-for-me(c), with dhcp or without on the same AP
```
network
st0   up global ip 192.168.1.2 gateway 192.168.1.1 mask 255.255.255.0
dns 192.168.1.1
```
```
wifi 0
> ssid0 => ...
> pass0 => ...
> ip0 => 192.168.1.2
> gw0 => 192.168.1.1
> mask0 => 255.255.255.0
> dns0 => 192.168.1.1
> bssid0 => 00:00:00:00:00:00
> chan0 => 0
```

limecat-ss, 2023-03-13:
After a while it seems to connect, but I don't have access to the Webui, nor does it respond to ping. This ip that appears in the log is not what I configured

```
[000097] [MAIN] ESPURNA 1.15.0-dev built 2023-03-13 09:27:09
[000098] [MAIN] xose.perez@gmail.com
[000098] [MAIN] http://tinkerman.cat
[000098] [MAIN] CPU chip ID: AC0BFBDC94C2 frequency: 80MHz
[000103] [MAIN] Device: WEMOS_D1_MINI
[000107] [TELNET] Listening on port 23
[000111] [WEBSERVER] Webserver running on port 80
[000122] [BUTTON] Number of buttons: 1
[000123] [LED] Number of leds: 1
[000124] [MQTT] AsyncMqttClient Buffer size 1024 (bytes)
state: 2 -> 3 (0)
state: 3 -> 2 (3a0)
state: 2 -> 2 (1a0)
state: 2 -> 0 (4)
[004985] [WIFI] State WaitConnected -> Timeout
[004985] [WIFI] Next connection attempt in 3000 (ms)
[004986] [WIFI] Connection timeout
[004996] [WIFI] State Timeout -> Idle
[008000] [WIFI] State Idle -> Connect
[008003] [WIFI] Connecting
[008014] [WIFI] State Connect -> WaitConnected
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 3
cnt 

connected with SERPRO-IoT, channel 8
dhcp client start...
pm open,type:0 0
ip:169.254.195.148,mask:255.255.0.0,gw:0.0.0.0
[259389] [WIFI] State WaitConnected -> Connected
[259392] [MDNS] Started with hostname ESPURNA-DC94C2
[259396] [WIFI] Connected to BSSID 6C:AA:B3:4D:CF:59 SSID SERPRO-IoT
[259408] [WIFI] State Connected -> Idle
[300176] [MAIN] Uptime: 00y 00d 00h 05m 00s
[300177] [MAIN] Heap: initial 41032 available 26464 contiguous 26200
[300178] [MAIN] VCC: 3028 mV

```

limecat-ss, 2023-03-13:
Hi,

It is working now. I discovered that it is necessary to configure the "DNS IP" in this version. In version 1.14.1 it is not necessary (I have five devices configured like this).

thanks !

closed 2023-03-13 by limecat-ss

mcspr, 2023-03-13:
This line has to be removed then
https://github.com/xoseperez/espurna/blob/d86be5697486aa584a4ed473ac96b8d47250a7cd/code/espurna/wifi.cpp#L767

If you think this should work regardless of DNS setting
https://github.com/xoseperez/espurna/blob/d86be5697486aa584a4ed473ac96b8d47250a7cd/code/espurna/wifi.cpp#L1475
(afaik IPAddress{} default is 0.0.0.0, so everything else should work)