xoseperez/espurna #2587: RF signal repeat 4 times

gianniocchipinti, 2023-05-20:
### Device

Sonoff RF Bridge R2 v2

### Version

espurna-1.14.1-itead-sonoff-rfbridge

### Question

Hi everyone
I installed espruna on sonoff rf bridge R2 v2.
When I try to send a signal it sends it 4 times despite having set: Repeat 1.
I saved all the settings I'm sure. I reloaded the page to see if the settings were saved correctly.

Debug console:
[140092] [WEBSOCKET] Requested action: relay
[140093] [RELAY] #0 scheduled ON in 0 ms
[140100] [RELAY] #0 set to ON
[140102] [RF] Enqueuing MESSAGE '64C003380A5A255F61' 4 time(s)
[140594] [RF] Action 0xA0
[141094] [RF] Action 0xA0
[141103] [RELAY] Setting relay mask: 0b1
[141594] [RF] Action 0xA0
[142094] [RF] Action 0xA0

you can see pic here:
[image]
[image]
[image]

Help me please....

Thank you!

gianniocchipinti, 2023-05-20:
Hi .. 
after a reboot, the signal is sent once, but when the controller receives the signal, the led strip turns on and off.
I also cloned a universal remote.. (small black that you see in the video). That works.

The original remote control is the white one. (initial movie)

1st movie
https://youtu.be/A-peBvrwV-s

2nd movie
https://youtube.com/shorts/4GBHlif6tkg

pulse is disabled

Thank you

mcspr, 2023-05-20:
> after a reboot, the signal is sent once

Yep, setting only applies at boot, is an oversight.

Usually that works just fine with repeats b/c devices would filter repeating codes (in case the message gets lost the first couple of times or something like remote button is held down)

btw, same code for ON and OFF also implicitly sets repeats to 1 and ignores the setting value

gianniocchipinti, 2023-05-20:
> Yep, setting only applies at boot, is an oversight.
>
> Usually that works just fine with repeats b/c devices would filter repeating codes (in case the message gets lost the first couple of times or [...]

I still have the problem.
 It is as if the remote control button were pressed for a long time and the LEDs turn on and off immediately.

Have you seen the movie?

mcspr, 2023-05-20:
Sonoff has ESP8266 and a 2nd chip (EFM8BB1 aka EFM8 aka 8051) that actually controls the RF part. From our side, repeats mean how many times we ask it to send things. iirc firmware on the 2nd chip *also* repeats in a small time window

gianniocchipinti, 2023-05-20:
> Sonoff has ESP8266 and a 2nd chip (EFM8BB1 aka EFM8 aka 8051) that actually controls the RF part. From our side, repeats mean how many times we ask it to send things. iirc firmware on the 2nd chip _also_ repeats in a small time window, which we don't control

Ok thank you.

i have this:
https://user-images.githubusercontent.com/28188533/136151668-d4d0ee83-9e93-4b0e-aec2-10f5b0648f20.jpg

is there a solution?

Thank you

mcspr, 2023-05-20:
One way is to take control of the RF, see https://github.com/xoseperez/espurna/wiki/Hardware-Itead-Sonoff-RF-Bridge---Direct-Hack

(?) another one is to try to use a oss firmware for the efm8 as-is and / or try to tweak its own hard-coded repeat options
https://github.com/halfbakery/RF-Bridge-EFM8BB1/tree/rebaked_sniffer
https://github.com/halfbakery/RF-Bridge-EFM8BB1/blob/7a0bac6052a3c4e7fd854231f27292231b571498/inc/RF_Protocols.h#L36

Using either https://github.com/lhartmann/c2_prog_wifi or Tasmota WebUI to upload .hex, connecting ESP8266 to the C2 interface pins

gianniocchipinti, 2023-05-20:
> Un modo è prendere il controllo della RF, vedi https://github.com/xoseperez/espurna/wiki/Hardware-Itead-Sonoff-RF-Bridge---Direct-Hack
> 
> (?) un altro è provare a utilizzare un firmware oss per efm8 così com'è e / o provare a modificare le proprie opzioni di ripetizione codificate [https://github.com/halfbakery/RF-Bridge-EFM8BB1/tree/ rebaked_sniffer ](https://github.com/halfbakery/RF-Bridge-EFM8BB1/tree/rebaked_sniffer) https://github.com/halfbakery/RF-Bridge-EFM8BB1/blob/7a0bac6052a3c4e7fd854231f27292231b571498/inc/RF_Protocols.h#L36
> 
> Utilizzando https://github.com/lhartmann/c2_prog_wifi o Tasmota WebUI per caricare .hex, collegando ESP8266 ai pin dell'interfaccia C2

Thank you

I had read that there was no need to change.
Can you confirm that after making this change, and flashing the firmware directly, everything works?
I had already tried tasmota too but it was worse.

mcspr, 2023-05-20:
No idea, just looking at the symptoms and assuming receiver does not like repeating codes :shrug:
Or, it does not like the way original fw sends data, which is why I suggested to try out a different fw for EFM8

Comparing RF signals from us and the original remote might shed some light on the issue, here we just guessing

gianniocchipinti, 2023-05-20:
> No idea, just looking at the symptoms and assuming receiver does not like repeating codes 🤷 Or, it does not like the way original fw sends data, which is why I suggested to try out a different fw for EFM8
> 
> Comparing RF signals from us and the original remote might shed some light on the issue, here we just guessing

OK thank you.  I've never compiled that firmware.  Wouldn't you have a file to install directly with repeat set to 1? 

 Another question.. What happens if I hack?  Will arche still work for me if I don't solve the signal problem?

mcspr, 2023-05-21:
.hex from above can also work as-is.
No modified firmware for now, would have to download the studio and efm8 toolchain to build it first.
(...where silabs studio installer does not even have download size or time estimate after starting it. and download speed is pretty slow atm :/)

Hack simply connects RF transmitter and RF receiver pins to ESP8266 instead of EFM8. Plus, we stop sending commands and receiving data from UART connection. Any standalone ESP8266 plus a generic PT2262 receiver & transmitter could also work btw, here they just happen to be in the same package.

gianniocchipinti, 2023-05-21:
> .hex from above can also work as-is. No modified firmware for now, would have to download the studio and efm8 toolchain to build it first. (...where silabs studio installer does not even have [...]

I have found this for R2 v2.2
can i follow this?
https://community.home-assistant.io/t/new-sonoff-rf-bridge-board-need-flashing-help/344326/17

I didn't understand where to download the .hex file to flash with tasmota webui
To flash the hex, do I have to do something with the pins? C2?? could you explain yourself better?

Thank you

gianniocchipinti, 2023-05-21:
> .hex from above can also work as-is. No modified firmware for now, would have to download the studio and efm8 toolchain to build it first. (...where silabs studio installer does not even have [...]

Hi,

[image]

[image]

It's ok?
tomorrow I have to buy the 2 resistors.

Let me know if the tracks are good

Thank you

gianniocchipinti, 2023-05-22:
I solved.  Hardware hacks + esphome

closed 2023-05-22 by gianniocchipinti

mcspr, 2023-05-23:
Can you post the config for esphome rf?

gianniocchipinti, 2023-05-23:
> Can you post the config for esphome rf? And did try to use our 1.14.1 or 1.15 builds for direct rf, or both did not work?

perhaps with espurna it was not sending the signals correctly.
With esphome the same thing.
Finally I fixed it by decoding the signal with a flipper zero and adding it to esphome (RAW CODE)

`esphome:
  name: sonoff-rf
  platform: ESP8266
  board: esp8285
  build_path: ./build/sonoff-rf

wifi:
  ssid: your_ssid
  password: your_pwd
  fast_connect: true

api:
logger: 
ota:
web_server:

status_led:
  pin: GPIO13

# USBRX = GPIO4 ---> receiver
# USBTX = GPIO5 ---> trasmitter

# receiver = pin 5 of the 8-legged chip (the one closer to the wifi antenna)
# transmitter = pin 4 of the 6-legged chip (closest to r12)

remote_receiver:
  pin:
    number: GPIO04
    inverted: true
  dump: rc_switch
  tolerance: 50%
  filter: 35us
  idle: 2ms

remote_transmitter:
  pin: GPIO05
  carrier_duty_percent: 100%

# this will log received commands, and can also transmit. Read up here:
# https://esphome.io/components/remote_transmitter.html#remote-setting-up-rf

button:
  - platform: template
    id: strip_led_ingresso_on_off
    name: ON/OFF
    on_press:
      - remote_transmitter.transmit_raw:
          code: [1335, -13132, 197, -528, 133, -66, 101, -732, 1017, -132, 97, -1224, 979, -522, 327, -166, 165, -966, 65, -100, 659, -134, 261, -524, 961, -560, 133, -332, 131, -100, 167, -17782, 133, -3738, 165, -1116, 497, -594, 129, -558, 329, -66, 163, -17424, 397, -98, 167, -200, 65, -498, 131, -198, 327, -394, 131, -196, 129, -362, 263, -132, 67, -166, 97, -432, 333, -68, 501, -894, 729, -264, 197, -98, 3277, -9944, 485, -12572, 383, -1268, 407, -1250, 1243, -450, 415, -1260, 381, -1272, 1229, -448, 407, -1286, 1211, -454, 413, -1270, 1239, -452, 415, -1280, 1241, -452, 1239, -450, 1271, -450, 1245, -484, 1229, -448, 409, -1288, 1249, -450, 1231, -484, 387, -1282, 379, -1266, 409, -1252, 383, -1280, 1233, -480, 381, -12648, 381, -1298, 363, -1284, 1239, -450, 415, -1266, 383, -1276, 1227, -480, 369, -1278, 1243, -456, 415, -1278, 1239, -450, 409, -1286, 1217, -462, 1235, -484, 1241, -484, 1233, -484, 1239, -450, 415, -1268, 1239, -456, 1273, -464, 379, -1268, 409, -1280, 375, -1280, 381, -1276, 1229, -480, 379, -12682, 369, -1294, 385, -1244, 1261, -450, 393, -1280, 381, -1280, 1249, -468, 381, -1270, 1231, -480, 379, -1290, 1249, -480, 379, -1286, 1215, -484, 1229, -484, 1253, -450, 1247, -472, 1243, -488, 385, -1278, 1239, -486, 1205, -482, 379, -1300, 383, -1304, 379, -1284, 343, -1306, 1237, -462, 383, -12662, 417, -1264, 389, -1284, 1207, -480, 365, -1312, 387, -1274, 1243, -446, 395, -1284, 1251, -450, 377, -1312, 1209, -486, 383, -1280, 1237, -486, 1241, -448, 135, -210, 177, -177548, 229, -996, 231, -166, 629, -166, 929, -17312, 563, -1088, 265, -24900, 99, -98, 133, -66, 231, -200, 133, -532, 301, -232, 399, -132, 399, -398, 599, -300, 265, -166, 65, -394, 399, -17284, 65, -334, 299, -98, 229, -862, 987, -918, 555, -66, 227, -98, 133, -466, 233, -68, 495, -17338, 197, -264, 101, -166, 765, -68, 165, -266, 67, -100, 165, -100, 133, -98, 495, -98, 99, -130, 65, -66, 165, -496, 99, -66, 231, -236, 297, -464, 65, -166, 1091, -132, 983, -12746, 129, -1360, 165, -232, 165, -498, 65, -66, 331, -100, 197, -66, 97, -230, 197, -594, 427, -132, 163, -64, 227, -756, 457, -16524, 297, -898, 197, -102, 203, -270, 165, -100, 97, -464, 131, -66, 731, -134, 65, -1060, 629, -230, 263, -264, 885, -132, 887, -11012, 233, -1752, 165, -522, 163, -164, 261, -166, 265, -234, 1391, -464, 265, -134, 165]
          repeat:
            times: 1
            wait_time: 100ms

`