xoseperez/espurna #2590: Sonoff S31 device loses wifi connection to wifi router

cohoman1, 2023-06-22:
### Device

Sonoff S31

### Version

espurna-1.15.0-dev.gitb409dc27+github230330-itead-sonoff-s31.bin

### Question

I installed the ESPerna firmware on two different Sonoff S31 Outlet devices and and these devices seem to work fine for about 12 hours then they are unreachable by the web browser GUI. I can't bring up the GUI nor send http GET commands to the devices. When I send a ping to the device's IP address,  I get nothing but occasionally I'll get a ping response with very large ping times. My ASUS wifi router shows the devices are connected to the router, and I've set the 2.4 GHz frequency to a specific channel number. The only way I can get these devices working again is to unplug them and plug them back in the AC power.

Has anyone run into this situation and know of a fix?

Thanks.

mcspr, 2023-06-23:
iirc the problem with our DHCP client (seeing 12h time, pretty common DHCP lease time)
Builds would have to switch to 3.1.x Arduino Core version where this is fixed, but I was slacking about that... :)

In the meantime, static ip would probably work around this issue?
Fetching our repo and building locally with `esp8266-1m-git-base` and `-DITEAD_SONOFF_S31` flag is another one; you'd test my theory, plus it would be possible to play around with our other build options

mcspr, 2023-07-27:
...also noticed in https://github.com/esp8266/Arduino/issues/8950

ASUS routers are either trying to throw us into 5Ghz with so called 'band steering' (disabled routers-side), or simply rejecting certain connections when in 802.11n and requiring 802.11g mode instead (can be us, or router setting)