xoseperez/espurna #2592: Build fail in latest dev when defining Software Serial.

davebuk, 2023-09-17:
### Device

D1 mini

### Version

4f22960a90408ebce1f4a38530a290818172ceab

### Bug description

Hi

I'm trying to build the latest 'dev' branch for one of my devices defined below:

```
  #define MANUFACTURER            "WEMOS"
    #define DEVICE                  "PZEM004TV3"    // Software Serial

    // My config
    #define ALEXA_SUPPORT           0
    #define DOMOTICZ_SUPPORT        0
    #define HOMEASSISTANT_SUPPORT   0
    #define THINGSPEAK_SUPPORT      0
    #define NTP_TIMEZONE            TZ_Europe_London
    
    // Buttons
    #define BUTTON1_PIN             0    // D3
    #define BUTTON1_CONFIG          BUTTON_PUSHBUTTON | BUTTON_DEFAULT_HIGH | BUTTON_SET_PULLUP
    
    // LEDs
    #define LED1_PIN                2 // D4
    #define LED1_PIN_INVERSE        1
    
    // Extras
	
    #define PZEM004TV30_ADDRESS                0xF8
    #define UART_SUPPORT                1
    #define UART_SOFTWARE_SUPPORT       1
    #define UART1_BAUDRATE              9600
    #define UART1_TX_PIN                14
    #define UART1_RX_PIN                12
    #define PZEM004TV30_SUPPORT         1
    #define PZEM004TV30_PORT            1
		
    #define SENSOR_POWER_CHECK_STATUS    0
```
It fails to build with the error below. I tried changing line 322 and 337 to `SoftwareSerial` but that didn't help. Is there an issue with `espurna\uart.cpp`?

```
espurna\uart.cpp:322:13: error: 'SoftwareSerialConfig' in namespace '::' does not name a type; did you mean 'SoftwareSerial'?
  322 | constexpr ::SoftwareSerialConfig from_config(Config config) {
      |             ^~~~~~~~~~~~~~~~~~~~
      |             SoftwareSerial
espurna\uart.cpp: In function 'espurna::driver::uart::{anonymous}::BasePortPtr espurna::driver::uart::{anonymous}::software_serial_port(uint32_t, uint8_t, uint8_t, espurna::driver::uart::{anonymous}::Config, bool)':
espurna\uart.cpp:337:40: error: '::SoftwareSerialConfig' has not been declared; did you mean 'SoftwareSerial'?
  337 |     ptr->begin(baudrate, from_config<::SoftwareSerialConfig>(config));
      |                                        ^~~~~~~~~~~~~~~~~~~~
      |                                        SoftwareSerial
espurna\uart.cpp:337:68: error: no matching function for call to 'from_config<<expression error> >(espurna::driver::uart::{anonymous}::Config&)'
  337 |     ptr->begin(baudrate, from_config<::SoftwareSerialConfig>(config));
      |                                                                    ^
espurna\uart.cpp:173:13: note: candidate: 'template<class T> constexpr T espurna::driver::uart::{anonymous}::from_config(espurna::driver::uart::{anonymous}::Config)'
  173 | constexpr T from_config(Config);
      |             ^~~~~~~~~~~
espurna\uart.cpp:173:13: note:   template argument deduction/substitution failed:
espurna\uart.cpp:337:68: error: template argument 1 is invalid
  337 |     ptr->begin(baudrate, from_config<::SoftwareSerialConfig>(config));
      |                                                                    ^
```

### Build tools used

--------------------------  ------------------------------------------------------
PlatformIO Core             6.1.12a1
Python                      3.9.10-final.0
System Type                 windows_amd64
Platform                    Windows-10
File System Encoding        utf-8
Locale Encoding             cp1252
PlatformIO Core Directory   C:\Users\***\.platformio
PlatformIO Core Executable  C:\Users\***\.platformio\penv\Scripts\platformio.exe
Python Executable           C:\Users\***\.platformio\penv\Scripts\python.exe
Global Libraries            0
Development Platforms       4
Tools & Toolchains          19
--------------------------  ------------------------------------------------------

mcspr, 2023-09-18:
SoftwareSerial update shenanigans...

`-git` environments use latest version of esp8266/Arduino, where SoftwareSerial had some breaking changes merged into master branch not so long ago
```diff
diff --git a/code/espurna/uart.cpp b/code/espurna/uart.cpp
index b3ef2b43..18ef0f9f 100644
--- a/code/espurna/uart.cpp
+++ b/code/espurna/uart.cpp
@@ -15,6 +15,7 @@ Copyright (C) 2022 by Maxim Prokhorov <prokhorov dot max at outlook dot com>

 #if UART_SOFTWARE_SUPPORT
 #include <SoftwareSerial.h>
+using SoftwareSerialConfig = EspSoftwareSerial::Config;
 #endif

 #include <array>
```

...
I'll need to make a reminder to apply fix above to the lib in esp8266/Arduino to preserve backwards compatibility

davebuk, 2023-09-18:
Yes, that now builds and runs on the device but the sensor isn't seen.

`[SENSOR] Could not read from PZEM004TV30 @ 0xf8 (Other / Unknown Error)`

UART gives:
`0 - SOFTWARE{tx=y rx=y}`

mcspr, 2023-09-19:
`pzemv30Debug => 1` for some more logs. Disconnected from pzem, does it send and receive with e.g. windows com port configured as 9600 8n1 and something like `echo hello > COM5`? Using a separate serial<->usb connected to esp
(see `mode` for ports list)

...
*And another reminder for a better sensor error log text*

davebuk, 2023-09-19:
With `1.15.0-dev.gitc86b4c37` running the device works, its been running fine for a year or so. DEBUG gives:

`[PZEM004TV3] Received: F8041409C1032A0000004C0000018D000001F40004000050F8 (25 bytes)`

I wanted to update to the latest version so with `1.15.0-dev.git4f22960a`

DEBUG gives:
```
[PZEM004TV3] ERROR: Expected 25 bytes, got 0
[SENSOR] Could not read from PZEM004TV30 @ 0xf8 (Other / Unknown Error)
```

davebuk, 2023-09-19:
I had set 
```
    #define UART1_TX_PIN                14
    #define UART1_RX_PIN                12
```
but `gpio` gives:
```
*  INPUT @ GPIO00 (HIGH)
   INPUT @ GPIO01 (HIGH)
*  INPUT @ GPIO02 (HIGH)
   INPUT @ GPIO03 (HIGH)
   INPUT @ GPIO04 (HIGH)
   INPUT @ GPIO05 (HIGH)
*  INPUT @ GPIO12 (HIGH)
   INPUT @ GPIO13 (HIGH)
*  INPUT @ GPIO14 (HIGH)
   INPUT @ GPIO15 (LOW)
   INPUT @ GPIO16 (LOW)
```

mcspr, 2023-09-19:
Wonder if it is software serial itself and not uart code here? Quick test with replacing debug port to pins 12 & 14, latest version always sends out corrupted lines. Might be the issue.

See `C:\Users\NAME\.platformio\packages\framework-arduinoespressif8266` with `.git` inside
(likely, the one that ends with `@src-*`. but, may be just the name above)

```cmd
> cd FRAMEWORK DIR ABOVE
> cd libraries\SoftwareSerial
> git fetch --tags
> git checkout 6.17.1
```

And rebuild

davebuk, 2023-09-19:
Not sure I'm doing the correct task but within VS Code, platformIO CLI I get a fatal error:
```
PS C:\Users\David\.platformio\packages\framework-arduinoespressif8266\libraries\SoftwareSerial> git fetch --tags
fatal: not a git repository (or any of the parent directories): .git
PS C:\Users\David\.platformio\packages\framework-arduinoespressif8266\libraries\SoftwareSerial>
```
I also have:
```
C:\Users\David\.platformio\packages\framework-arduinoespressif8266@3.20704.0
and
C:\Users\David\.platformio\packages\framework-arduinoespressif8266@3.30002.0
```
directories but they give the same error.

mcspr, 2023-09-19:
Hm, my mistake, first one is the right one. Instead of git, moving it some place safe and replacing it with https://github.com/plerup/espsoftwareserial/archive/refs/tags/6.17.1.zip would do the trick

mcspr, 2023-09-19:
Or, using a generally safer method with platformio.ini
```diff
diff --git a/code/platformio.ini b/code/platformio.ini
index 3a7c6a21..51108fec 100644
--- a/code/platformio.ini
+++ b/code/platformio.ini
@@ -138,6 +138,7 @@ debug_flags =

 shared_lib_dir = libraries/
 shared_lib_deps =
+    https://github.com/plerup/espsoftwareserial/archive/refs/tags/6.17.1.zip
     https://github.com/me-no-dev/ESPAsyncTCP.git#7e9ed22
     bblanchon/ArduinoJson@^5.13.4
     marvinroger/AsyncMqttClient@^0.8.1
```

Should show up like this when `pio run --verbose` is used (idk what IDE flag that might be)

```console
|-- EspSoftwareSerial @ 6.17.1 (License: LGPL-2.1+, URI: https://github.com/plerup/espsoftwareserial/archive/refs/tags/6.17.1.zip, Path: /home/builder/dev/espurna/code/libraries/EspSoftwareSerial)
```

davebuk, 2023-09-20:
Built using the `platformio.ini` method. Error again in `uart.cpp`.
```
Building in release mode
Compiling .pio\build\db-wemos-pzemv3\src\uart.cpp.o
Archiving .pio\build\db-wemos-pzemv3\lib57c\libESP8266WiFi.a
Archiving .pio\build\db-wemos-pzemv3\lib6a7\libFauxmoESP.a
Compiling .pio\build\db-wemos-pzemv3\lib2eb\Adafruit NeoPixel\kendyte_k210.c.o
Compiling .pio\build\db-wemos-pzemv3\liba1f\Wire\Wire.cpp.o
Compiling .pio\build\db-wemos-pzemv3\lib505\IRremoteESP8266\IRac.cpp.o
Compiling .pio\build\db-wemos-pzemv3\lib505\IRremoteESP8266\IRrecv.cpp.o
Archiving .pio\build\db-wemos-pzemv3\lib2eb\libAdafruit NeoPixel.a
Compiling .pio\build\db-wemos-pzemv3\lib505\IRremoteESP8266\IRsend.cpp.o
Compiling .pio\build\db-wemos-pzemv3\lib505\IRremoteESP8266\IRtext.cpp.o
Archiving .pio\build\db-wemos-pzemv3\liba1f\libWire.a
Compiling .pio\build\db-wemos-pzemv3\lib505\IRremoteESP8266\IRtimer.cpp.o
Compiling .pio\build\db-wemos-pzemv3\lib505\IRremoteESP8266\IRutils.cpp.o
Compiling .pio\build\db-wemos-pzemv3\lib505\IRremoteESP8266\ir_Airton.cpp.o
Compiling .pio\build\db-wemos-pzemv3\lib505\IRremoteESP8266\ir_Airwell.cpp.o
espurna\uart.cpp:18:30: error: 'EspSoftwareSerial' does not name a type; did you mean 'SoftwareSerial'?
   18 | using SoftwareSerialConfig = EspSoftwareSerial::Config;
      |                              ^~~~~~~~~~~~~~~~~
      |                              SoftwareSerial
Compiling .pio\build\db-wemos-pzemv3\lib505\IRremoteESP8266\ir_Aiwa.cpp.o
*** [.pio\build\db-wemos-pzemv3\src\uart.cpp.o] Error 1
```

mcspr, 2023-09-20:
https://github.com/xoseperez/espurna/issues/2592#issuecomment-1723203000 `using = ...` addition should be removed when trying .zip

davebuk, 2023-09-20:
`//using = ...` saved to file and now it builds but there still isn't any data and returns 0 bytes with the same GPIO keys:
```
Dependency Graph
|-- ArduinoJson @ 5.13.4
|-- FauxmoESP @ 3.4.0
|-- ESPAsyncTCP @ 1.2.0+sha.7e9ed22
|-- ESP8266WiFi @ 1.0
|-- Adafruit NeoPixel @ 1.11.0
|-- Wire @ 1.0
|-- IRremoteESP8266 @ 2.8.2
|-- my92xx @ 3.0.3
|-- ESP8266LLMNR @ 1.0
|-- SPI @ 1.0
|-- ESP8266mDNS @ 1.2
|-- AsyncMqttClient @ 0.8.2
|-- MQTT @ 2.4.7+sha.196556b
|-- PubSubClient @ 2.8.0
|-- ESP8266NetBIOS @ 1.0
|-- NoFUSS @ 0.4.0+sha.bec6a8c
|-- ArduinoOTA @ 1.0
|-- ESP8266WebServer @ 1.0
|-- ESP8266HTTPClient @ 1.2
|-- ESP8266httpUpdate @ 1.3
|-- rc-switch @ 2.6.2+sha.1140265
|-- RFM69 @ 1.5.2
|-- RPNlib @ 0.24.1
|-- ESP8266SSDP @ 1.0
|-- EEPROM_Rotate @ 0.9.2
|-- ESP8266_SSD1306 @ 4.1.0+sha.3398c97
|-- EspSoftwareSerial @ 6.17.1
...
```

```
*  INPUT @ GPIO00 (HIGH)
   INPUT @ GPIO01 (HIGH)
*  INPUT @ GPIO02 (HIGH)
   INPUT @ GPIO03 (HIGH)
   INPUT @ GPIO04 (HIGH)
   INPUT @ GPIO05 (HIGH)
*  INPUT @ GPIO12 (HIGH)
   INPUT @ GPIO13 (HIGH)
*  INPUT @ GPIO14 (HIGH)
   INPUT @ GPIO15 (LOW)
   INPUT @ GPIO16 (LOW)
```

mcspr, 2023-09-20:
Aah... TX and RX are just swapped.
```diff
diff --git a/code/espurna/uart.cpp b/code/espurna/uart.cpp
index b3ef2b43..e4238f39 100644
--- a/code/espurna/uart.cpp
+++ b/code/espurna/uart.cpp
@@ -333,7 +333,7 @@ BasePortPtr software_serial_port(
     const int8_t tx_pin = (tx == GPIO_NONE) ? -1 : tx;
     const int8_t rx_pin = (rx == GPIO_NONE) ? -1 : rx;

-    auto* ptr = new SoftwareSerial(tx_pin, rx_pin, invert);
+    auto* ptr = new SoftwareSerial(rx_pin, tx_pin, invert);
     ptr->begin(baudrate, from_config<::SoftwareSerialConfig>(config));

     return std::make_unique<BasePort>(
```

davebuk, 2023-09-21:
Nice one, that works.

TX RX changed with EspSoftwareSerial @ 6.17.1 added.

Should I try reverting `uart.cpp` back to normal and just change the TX RX code line?

closed 2023-09-22 by mcspr

mcspr, 2023-09-22:
Pins order was wrong, fixed in dev

swserial v8 *might* be ok to use, but idk how it may compare to the old version
(6.17.1 was what previously used, so known to work)
`using = ...` fix is need to test that

davebuk, 2023-09-22:
All files pulled from `1.15.0-dev.git3a34f06c` but it wouldn't build.

`using SoftwareSerialConfig = EspSoftwareSerial::Config;` added to `uart.cpp` now allows it to build and the device works correctly.

commit e8c93778 (2023-09-22): uart: fix swserial pin order

fix #2592
commit 06e49ccf (2023-10-16): uart: swserial workaround for recent Core version

fix #2592 (for real this time)