xoseperez/espurna #2597: Sonoff Pow 3 (POWR316D) support

vreesie18, 2023-11-06:
### Description

Is there any chance for support of the Sonoff  Pow 3 (POWR316D) device on a short notice ?

mcspr, 2023-11-12:
Check out https://templates.blakadder.com/sonoff_POWR316D.html
This is ESP32 device, we are still on ESP8266 builds. Pending #687 (...and since it has to do with a major software stack changes)

vreesie18, 2023-12-04:
Thanks, already running Tasmota now on this device (seems only option for now). Actually I left Tasmota on all my devices in favor of Espurna. Unexplainable hick-ups (power drops) in Tasmota made me decide to do this.

closed 2023-12-04 by vreesie18