xoseperez/espurna #2598: Possible issue with LIGHTS and MQTT.

davebuk, 2023-11-10:
### Device

MAGICHOME Single LED and other 'LIGHTS'

### Version

1.15.0-dev.gitf3c3efca

### Bug description

I have updated all my devices to `1.15.0-dev.gitf3c3efca` but the three that have `LIGHTS` (Single LED, RGB LED and TUYA dimmer) do not respond correctly either using MQTT or the webUI. I think that although `channel0` receives the command, it doesn't get updated to the GPIO. The webUI slider moves WRT the MQTT command but the MQTT sending always says `0`. Any ideas?

Thanks.

```
[111151] [MQTT] Received LED001/relay/0/set => 1
[112154] [MQTT] Received LED001/channel/0/set => 51
[112264] [MQTT] Sending LED001/channel/0 => 0 (PID 228)
[112265] [MQTT] Sending LED001/brightness => 255 (PID 229)
[112266] [MQTT] Sending LED001/light => 0 (PID 230)
[205163] [MQTT] Sending LED001/status => 1 (PID 231)
[205165] [MQTT] Sending LED001/host => ESPURNA-46 (PID 232)
[205166] [MQTT] Sending LED001/desc => Dining Room Cupboards (PID 233)
[205170] [MQTT] Sending LED001/ssid => BrownNetV (PID 234)
[205174] [MQTT] Sending LED001/ip => 192.168.1.46 (PID 235)
[205179] [MQTT] Sending LED001/rssi => -57 (PID 236)
[205184] [MQTT] Sending LED001/channel/0 => 0 (PID 237)
[205191] [MQTT] Sending LED001/brightness => 255 (PID 238)
[205194] [MQTT] Sending LED001/light => 0 (PID 239)
[324260] [MQTT] Received LED001/relay/0/set => 1
[325264] [MQTT] Received LED001/channel/0/set => 51
[327419] [MQTT] Received LED001/channel/0/set => 63.75
[327531] [MQTT] Sending LED001/channel/0 => 0 (PID 240)
[327532] [MQTT] Sending LED001/brightness => 255 (PID 241)
[327532] [MQTT] Sending LED001/light => 0 (PID 242)
[344019] [MQTT] Received LED001/relay/0/set => 0
[345369] [MQTT] Received LED001/relay/0/set => 1
[346366] [MQTT] Received LED001/channel/0/set => 51
[346483] [MQTT] Sending LED001/channel/0 => 0 (PID 243)
[346484] [MQTT] Sending LED001/brightness => 255 (PID 244)
[346484] [MQTT] Sending LED001/light => 0 (PID 245)
[354865] [MQTT] Received LED001/channel/0/set => 114.75
[354977] [MQTT] Sending LED001/channel/0 => 0 (PID 246)
[354978] [MQTT] Sending LED001/brightness => 255 (PID 247)
[354978] [MQTT] Sending LED001/light => 0 (PID 248)
[358837] [MQTT] Received LED001/channel/0/set => 178.5
[358948] [MQTT] Sending LED001/channel/0 => 0 (PID 249)
[358949] [MQTT] Sending LED001/brightness => 255 (PID 250)
[358950] [MQTT] Sending LED001/light => 0 (PID 251)
[367033] [MQTT] Received LED001/channel/0/set => 216.75
[367147] [MQTT] Sending LED001/channel/0 => 0 (PID 252)
[367148] [MQTT] Sending LED001/brightness => 255 (PID 253)
[367148] [MQTT] Sending LED001/light => 0 (PID 254)
```

mcspr, 2023-11-11:
> [111151] [MQTT] Received LED001/relay/0/set => 1

> [112266] [MQTT] Sending LED001/light => 0 (PID 230)

`LED001/light/set`?

b5a03dac728a4eef34a5c03da5bb395b4d564ed6 removed `LIGHT_RELAY_ENABLED`, and cfg has to have explicit relay provider being set (see commit changes)

davebuk, 2023-11-11:
I had to change my custom.h from 
`#define RELAY_PROVIDER    RELAY_PROVIDER_LIGHT`
to
`#define RELAY1_PROVIDER    RELAY_PROVIDER_LIGHT_STATE`.

That has fixed the issue. I also removed `#define LIGHT_CHANNELS    1`.

All working now.

Thanks.

closed 2023-11-11 by davebuk