xoseperez/espurna #2603: Position KingArt Q4

4139GGN, 2024-01-21:
### Device

KingArt Q4

### Version

1.15.0-dev.git909949b5+github230112

### Question

Hello!
I have had several KingArt Q4s running with Espurna in Home Assistant for months. But it only indicates when it is "open" or "closed". If it stops in another position, it does not indicate its position (%).

`  cover:
    - name: "Persiana Salón"
      device_class: shutter
      command_topic: "PersianaSalon/curtain/set"
      position_topic: "PersianaSalon/curtain"
      set_position_topic: "PersianaSalon/curtain/set"
      state_topic: "PersianaSalon/curtain"
      qos: 0
      retain: false
      payload_open: "0"
      payload_close: "100"
      payload_stop: "pause"
      position_open: 0
      position_closed: 100      
      state_open: "open"
      state_opening: "opening"
      state_closed: "close"
      state_stopped: "closing"
      state_closing: "stop" 
      optimistic: true `

mcspr, 2024-01-21:
Something to do with the HA config? Kingart hw does indeed report on / off / pause, but we only ever send out 0...100 value
https://github.com/xoseperez/espurna/blob/238dc130af023d6b950c5c0ce190021a8d0526ad/code/espurna/curtain_kingart.cpp#L318-L326

Also note that incoming payload is also limited to on / off / pause / numeric-value
https://github.com/xoseperez/espurna/blob/238dc130af023d6b950c5c0ce190021a8d0526ad/code/espurna/curtain_kingart.cpp#L358-L368

4139GGN, 2024-01-22:
> Something to do with the HA config? Kingart hw does indeed report on / off / pause, but we only ever send out 0...100 value [...]

Thank you! But there I lose the truth...
Yes, I can say that if I indicate position on the HA card (from 0 to 100), it does respond to that position and registers that it has been open or closed (depending on its last move).
So, I can't see the % position in HA?
Another question I have is: Why does HA appear as unknown in the registry until you make a move?

Thank you!!!

mcspr, 2024-01-22:
> Another question I have is: Why does HA appear as unknown in the registry until you make a move?

We publish `<DEVICE>/status` topic as last-will-at-testament aka lwt, which tracks mqtt broker connection status of the device. `<DEVICE>/curtain` tracks percentage

```yaml
availability:
  - topic: "PersianaSalon/status"
payload_available: "1"
payload_not_available: "0"
```

`<DEVICE>/curtain` only publishes numeric value, it would never get 'state' payload as described here.
```yaml
      state_topic: "PersianaSalon/curtain"
      state_open: "open"
      state_opening: "opening"
      state_closed: "close"
      state_stopped: "closing"
      state_closing: "stop" 
```

WebUI does indeed show the status, but MQTT publish is missing in current implementation. Not sure what would be the correct solution here, though. Either publish special .json `{state: "opening", position: 50}` or have something like `<DEVICE>/curtain_state`

> So, I can't see the % position in HA?

I am not actively using HA, just testing it with your configuration atm.
Don't see it either :shrug: , so I would only assume this is how the default widget is intended to work. Maybe some template magic / separate widget reading entity status can solve this.

4139GGN, 2024-02-09:
With this configuration (configuration.yaml), I only get the status "Open/Closed".
How could I get the numeric value that I do see being output in mqtt?:
BlindSalon/curtain

> 

    cover:

    - name: "Persiana Salón"
      device_class: shutter
      command_topic: "PersianaSalon/curtain/set"
      position_topic: "PersianaSalon/curtain"
      set_position_topic: "PersianaSalon/curtain/set"
      state_topic: "PersianaSalon/curtain"
      qos: 0
      retain: false
      payload_open: "0"
      payload_close: "100"
      payload_stop: "pause"
      position_open: 0
      position_closed: 100
      state_open: "0"
      state_closed: "100"
      state_stopped: "Pausa"
      optimistic: true