xoseperez/espurna #2610: UI  should limite the number of schedules to the internal limit

brunetton, 2024-06-23:
### Device

itead-sonoff-dual-r2

### Version

1.16.0-dev.git7536a6ed+github240615

### Bug description

It's possible via UI to add more schedules that the number of schedules that can be memorized in EPROM.

### Steps to reproduce

## Context

Today I wanted to add schedules, one schedule for each month (yes I'm aware of #2417 but I need to close before sunset during spring and after during summer).

So I created 12 schedules (that's takes some time), 

## Steps to reproduce

- add more than 10 schedules
- click "Save"
- UI says "Changes saved"
- go back to schedules page
- => "only" 10 schedules are saved

## Suggestion

The UI should either:
- prevent adding more than 10 schedules (grayed "Add" button)
- warn user about 10 limitation (on the text at the top of the page)
- warn user when trying to add more than 10 schedules in the alert box

Would you be opened to PR for this ?

mcspr, 2024-06-24:
ref. commit above (4f520a8)

> warn user about 10 limitation (on the text at the top of the page)
warn user when trying to add more than 10 schedules in the alert box

Apparently where were two ways to create such lines in settings code, one warning about the limit and the other wasnt doing that (e.g. in case of schedules). Commit above removes the broken entries, now everything goes through a single handler.

> prevent adding more than 10 schedules (grayed "Add" button)

Might be nicer, though. If you have a better idea than an just alert(), by all means please submit a PR

brunetton, 2024-07-10:
I didn't tested yet but this seems to answer the question :) 

Thanks !

closed 2024-07-10 by brunetton