xoseperez/espurna #2616: LWT "online" apprently published with retain on even though retain toggle in UI is turned off.

CrappyTan, 2024-09-01:
### Device

All

### Version

1.16.0-dev.git7536a6ed+github240615

### Bug description

- I have a sonoff with latest build on (1.16.0-dev.git7536a6ed+github240615)
- I have MQTT connected to Mosquitto
- I have Retained turned off. 
- I have Keep-Alive set to 15 seconds. 

**Expected**
When the device powers up, it publishes a /status message of 1.  
15 seconds of loss, the LWT is triggered and I see 0. Expected. 

**Unexpected.** 
A new subscription to the topic - whilst device is still offline - however gets a published message of 1. 
Expected behaviour should be that new subscriptions don't receive any messages or the LWT = 0 message. 
Any new clients will assume it is online which it's not. 

Am I misunderstanding the implementation / intention of LWT?

### Steps to reproduce

Added Above.
MQTTSpy is used in this as well as custom Arduino code which highlighted issue. 

### Build tools used

N/A

### Any relevant log output (when available)

```shell
Added above
```

### Decoded stack trace (when available)

```markdown
N/A
```

CrappyTan, 2024-09-01:
So having a dig through the code, it looks like I am correct. 
mqtt.cpp, line 1642, method mqttSendStatus
`void mqttSendStatus() {
    mqttSendRaw(_mqtt_settings.will.c_str(), _mqtt_payload_online.c_str(), true);
}`

For the "retain" flag, it passes hardcoded true and not _mqtt_settings.retain as set. 

Is this correct? 
Does LWT need to be retain? (the offline version is not set as retained)

mcspr, 2024-09-02:
> So having a dig through the code, it looks like I am correct. mqtt.cpp, line 1642, method mqttSendStatus `void mqttSendStatus() { mqttSendRaw(_mqtt_settings.will.c_str(), _mqtt_payload_online.c_str(), true); }`
> 
> For the "retain" flag, it passes hardcoded true and not _mqtt_settings.retain as set.
> 
> Is this correct? Does LWT need to be retain? (the offline version is not set as retained) Can it be set to _mqtt_settings.retain and pull from the settings?

Check out the setup part
https://github.com/xoseperez/espurna/blob/330ed1cc2476fce74eb0c270374b7a4c7ef1c3cf/code/espurna/mqtt.cpp#L598-L602

LWT settings are passed to the broker, and mqttSendStatus() is in the wrong here
https://docs.oasis-open.org/mqtt/mqtt/v3.1.1/os/mqtt-v3.1.1-os.html#_Toc385349234 ('3.1.2.7 Will Retain')

CrappyTan, 2024-09-02:
So if I understand you correctly, you're saying the implementation is correct? 

I played around with it a bit more last night and think there is something amiss in the logic flow.

My understand, from behaviour I've seen with this firmware, is, LWT is published on connection and then rapidly a status:1 is published. 
When the subscriber disconnects, the LWT is published. This is all good. 

However

When a new client connects and the original publisher is offline, it receives a status:1 message. This is not the LWT but the "status update" sent from previous which has the retain flag set even though the retain toggle is off in my instance. 

That thinking chimes  with the code in line 1642 does it not? sendStatus
https://github.com/xoseperez/espurna/blob/330ed1cc2476fce74eb0c270374b7a4c7ef1c3cf/code/espurna/mqtt.cpp#L1642C1-L1644C2

So the issue is not the LWT but rather the sendStatus?

CrappyTan, 2024-09-02:
Not sure why it was closed? 
Whilst we've (I) have understood it's not the LWT it certainly is not behaving as it should and this is easily workable. 

1. Turn a Sonoff on and connect to a MQTT broker. 
2. Disable Retain switch. 
3. Restart Mosquitto to clear everything (clearly, a few things need to be right for this).
4. Using mqttSpy, subscribe to the correct topic and wait for status=1. 
5. Turn the sonoff off. Wait for the LWT to publish to 0. 
6. Disconnect mqtt spy and resubscribe to the topic. A value of 1 is published. This is not correct. 

Expected behaviour: On new subscription, no value should be received by mqttSpy.

Not entirely sure of the etiquette here - shall I rename this bug to something else now that is more appropriate?

mcspr, 2024-09-02:
Have you seen https://github.com/xoseperez/espurna/commit/4a2a2f2cb7ea71cf503436890efe7ed1cb5e1368?

CrappyTan, 2024-09-02:
Oh snap! Thanks. I did not see that and did not realise you closed it with that. Noob :)

Thank you! I'll pull it and build it.

mcspr, 2024-09-03:
No worries. Assuming everything works ok now?
While looking at lwt spec... I wonder, should there be a separate setting for LWT retain independent of the global one?

CrappyTan, 2024-09-03:
I've not had a chance to do it yet. Refactoring my other code which led me down a rabbit hole :)
Will hopefully today or tomorrow. 

Yes, I think that is a very good idea. 
Is it make sense to separate the LWT and status topics? If you implement your idea then you need to always use lwt retain settings for the status setting too.

Conversely, the status without every having the ability to switch from 1 to zero is also misleading. 

Perhaps the best option is to allow their own retain settings and the status topic uses that setting and not the global setting. 
Thoughts?

mcspr, 2024-09-03:
> Is it make sense to separate the LWT and status topics? If you implement your idea then you need to always use lwt retain settings for the status setting too.

Not really, as lwt is status and status is lwt. Server is given a payload for 'disconnected' state, so we publish the opposite 'connected' (when connected :). Otherwise, it would be at odds with the real device status for a different reason - being unable to switch from zero to one

Right now topic is swappable, payload as well. QoS and Retain are not.

mcspr, 2024-09-10:
Separate will topic, qos and retain added in 1f93ce4.

Topic would be either default `<ROOT>/status`, or the user-defined one. Original code was replacing the 'status' text, so it is a bit different now for both will and JSON

CrappyTan, 2024-09-11:
thank you. That's great. I have still not got a chance to take a look but will in the coming weeks and provide some feedback on it's use.

thanks again

commit 4a2a2f2c (2024-09-02): mqtt: use global retain flag for status

resolve #2616