xoseperez/espurna #2617: Enable/ disable scheduler(s) via MQTT

bopnrgi, 2024-09-03:
### Description

Is it possible to add topics on MQTT to enable or disable schedulers via MQTT ?

For iot behind 4G router without home assistant it can manage water boiler easily (for example)

mcspr, 2024-09-03:
Do you mean something akin to HTTP API, where settings are overwritten and enabled / disabled state persists? Would it be ok if 'enabled' / 'disabled' state is not persistent and would reset after device reboots?

Does this also mean there has to be a way to know the current state?
Probably other portions of the config as well? 'Restore' and 'Time'

---

Right now, MQTT terminal option can be enabled. Then, publishing to `$device/cmd/set` payloads like `set schType0 disabled` to enable, `set schType0 calendar` to turn it back to 'Calendar' type.

bopnrgi, 2024-09-05:
Hello, thanks for your reply :)
Context : My device is behind a 4G box, so it's not possible to use HTTP API (4G terminal don't have public IP on 4G Network). That's why i wanna use MQTT to control it. This exclude HTTP API... 

I wanna use a MQTT topic to turn on/off any scheduler (and persist state at reboot, but the firmware do it now..) like a switch.
[image]
It could be : `<root>/scheduler/#/set` (replace # by id of scheduler)

`set schType0 disabled` seems not work

mcspr, 2024-09-06:
> I wanna use a MQTT topic to turn on/off any scheduler (and persist state at reboot, but the firmware do it now..) like a switch.It could be : <root>/scheduler/#/set (replace # by id of scheduler)

That is exactly why I am asking. MQTT is not request -> response exactly, but publisher -> subscriber w/ possibly multiple publishers and e.g. 'retain' flag that may introduce some quirky side effects.

> set schType0 disabled seems not work

Should work with 1.16.0+ **and** with [`TERMINAL_MQTT_SUPPORT` flag](https://github.com/xoseperez/espurna/blob/4a2a2f2cb7ea71cf503436890efe7ed1cb5e1368/code/espurna/config/general.h#L180)
As there is no ENABLED / DISABLED state anymore. But, there are more options for time control, if that is what you actually need. See the Scheduler Wiki

closed 2024-09-10 by mcspr