xoseperez/espurna #2620: Multiple Virtual Analog Ports

JavierAder, 2024-09-19:
### Description

Hello. It's been a while since I wrote something. Greetings to everyone.

Motivation: The ESP8266 has only one analog port, which does not allow measuring more than one analog sensor. Personally, I need to measure multiple temperatures using NTC sensors (the digital sensors supported by Espurna haven't worked well for me); but beyond that, this modification would allow the use of multiple analog sensors of any kind (pressure sensors, humidity sensors, etc.).

Proposal: Allow Espurna to use "multiple virtual analog ports" with the help of an external analog multiplexer and the use of digital ports.

The general idea can be seen here:
https://www.youtube.com/watch?v=OgaeEiHemU4
or here (9.2 Typical Application):
https://www.ti.com/lit/ds/symlink/cd4052b.pdf?ts=1726728167254

I suppose the main modification should be made in AnalogSensor, allowing several instances of it, each associated with different "virtual analog ports." What AnalogSensor should do is: before reading the real analog input (pin 0), set the corresponding input of the multiplexer, wait for a small delay, and then read the analog value (see _rawRead() and analogRead(pin)).

At the configuration level, to enable this functionality, one should set something like:

#EnableVAP=1  // enables virtual analog ports
#DVAP= 1 or 2 or 3  // Number of digital ports connected to the multiplexer
#DVAP0= digital port to use for specifying bit 0 of the multiplexer port address
#DVAP1= digital port to use for specifying bit 1 of the multiplexer port address
#DVAP2= digital port to use for specifying bit 2 of the multiplexer port address

With these modifications, it would be possible to read from up to 8 analog devices.

mcspr, 2024-09-23:
> I suppose the main modification should be made in AnalogSensor, allowing several instances of it, each associated with different "virtual analog ports." What AnalogSensor should do is: before [...]

So, from the sensor configuration side, it would be enough to just provide it with TYPE of analog source and PIN / CHANNEL / some-kind-of-a-name-for-ID ?

Right now something similar is happening with GPIO pins and. e.g., relays w/ mcp pins, rfb pins, sonoff pins. Relay config asks for a TYPE of pin, provider gives out a pin handler that API consumer promises to use it for reads & writes.

I do not really like the word 'virtual' here, as it does continue to use real hardware channel and just requires some extra work before doing so :) Source does change, though.

> With these modifications, it would be possible to read from up to 8 analog devices.

fwiw cd4052b pdf linked describes a general purpose multiplexer, so generic pin read & write can also use those as inputs & outputs and not just analog.

JavierAder, 2024-09-24:
Hi; 
my idea is something like this.
sensor.cpp
```

#ifdef NTC_VIRTUAL_SUPPORT

NTCSensor createVirtualNTCSensor(int NTCx,size_t NTCx_SAMPLES,
      Delay NTCx_DELAY,unsigned long NTCx_R_UP,unsigned long NTCx_R_DOWN, double NTCx_INPUT_VOLTAGE,
        unsigned long NTCx_BETA ,unsigned long NTCx_R0,double NTCx_T0)
{

        auto* sensor = new NTCSensor();
        sensor->setSamples(NTCx_SAMPLES);
        sensor->setDelay(NTCx_DELAY) ;
        sensor->setUpstreamResistor(NTCx_R_UP);
        sensor->setDownstreamResistor(NTCx_R_DOWN);
        sensor->setInputVoltage(NTCx_INPUT_VOLTAGE);
        sensor->setBeta(NTCx_BETA);
        sensor->setR0(NTCx_R0);
        sensor->setT0(NTCx_T0);
        sensor->setVirtualPort(NTCx);
        return sensor;

}
#endif

#if NTC_SUPPORT
    #ifndef NTC_VIRTUAL_SUPPORT
    {
        auto* sensor = new NTCSensor();
        sensor->setSamples(NTC_SAMPLES);
        sensor->setDelay(NTC_DELAY) ;
        sensor->setUpstreamResistor(NTC_R_UP);
        sensor->setDownstreamResistor(NTC_R_DOWN);
        sensor->setInputVoltage(NTC_INPUT_VOLTAGE);
        sensor->setBeta(NTC_BETA);
        sensor->setR0(NTC_R0);
        sensor->setT0(NTC_T0);
        add(sensor);
    }
    #else
     {
        #if NTC0 
        {add(createVirtualNTCSensor(NTC0,NTC0_SAMPLES,NTC0_DELAY,NTC0_R_UP,NTC0_R_DOWN,NTC0_INPUT_VOLTAGE,
        NTC0_BETA ,NTC0_R0,NTC0_T0)); 
         }
        #endif
        #if NTC1 
        {add(createVirtualNTCSensor(NTC1,NTC1_SAMPLES,NTC1_DELAY,NTC1_R_UP,NTC1_R_DOWN,NTC1_INPUT_VOLTAGE,
        NTC1_BETA ,NTC1_R0,NTC1_T0)); 
         }
        #endif
        //TODO: do the same for NTC2...NTC7 or use macros while.....
        
        
    }
   
    #endif
#endif

```

AnalogSensor.h
```

       void  setVirtualPort(int  vport)
        {
            _vport = vport;
        }

......

   protected:
        int _vport= -1;

        static unsigned int _rawRead(uint8_t pin, size_t samples, Delay delay) {
            // TODO: system_adc_read_fast()? current implementation is using system_adc_read()
            // (which is even more sampling on top of ours)
            unsigned int last { 0 };
            unsigned int result { 0 };
            for (size_t sample = 0; sample < samples; ++sample) {
                const auto value = ::analogRead(pin);
                result = result + value - last;
                last = value;
                if (sample > 0) {
                    espurna::time::critical::delay(delay);
                    yield();
                }
            }

            return result;
        }

        unsigned int _rawRead() const {
             #if VAP_SUPPORT
             {
                if (_vport>=0)
                {s
                    setVirtualPort();
                    //TODO: delay for multiplexer?

                }
            } 
             #endif

            return _rawRead(0, _samples, _delay);
        }

        #if VAP_SUPPORT
        void setVirtualPort() const{
            //TODO: using _vport and #DVAP0,#DVAP1 and #DVAP2 set digital outputs

        }
        #endif

````
Names are tentative

JavierAder, 2024-09-24:
> fwiw cd4052b pdf linked describes a general purpose multiplexer, so generic pin read & write can also use those as inputs & outputs and not just analog.

Yes, multiplexing is general, but my main restriction is that there is only one analog port.

mcspr, 2024-09-24:
Right, sensor specific code is possible. What I mean is to separate things ever so slightly.

Just to play around with this... You can already override `analogRead(uint8_t)` definition

```cpp
// in any .cpp file, global namespace
int multiplexer_read(uint8_t pin) {
  ???
}

extern "C" int analogRead(uint8_t pin) {
  switch (pin) {
  case A0: // 17
    return system_adc_read();

  case 0 ... 7: // or some other unused numbers in the 0...255 range
    return multiplexer_read(pin);
  }

  return 0;
}
```

Extend NTCSensor code to allow PIN value changes, make a setup() code to instantiate multiplexer and then change `analogRead` implementation to access the multiplexer pin. Any implementation details related to PIN switching timing would be apparent, e.g. is there a need for delayMicroseconds / delay from our side or not, etc.

What I meant is to integrate multiplexer system-wide, not sensor specifically.
Have you seen relay and button code related to providers?

- system initiates multiplexer elsewhere. api extended to provide a type of analog data source. real hw analog source is a type, multiplexer is a type.
- sensor code made aware of multiplexer through this 'type'. this not a virtual thing, just a type of analog data source.
- sensor code also extended to some kind of ID (pin number, channel number). configuration assigns both TYPE and ID, sensor gets analog source data though the new api instead of using analogRead directly

JavierAder, 2024-09-26:
> Right, sensor specific code is possible. What I mean is to separate things ever so slightly.
> 
> Just to play around with this... You can already override `analogRead(uint8_t)` definition
> 

> ```c++
> // in any .cpp file, global namespace
> int multiplexer_read(uint8_t pin) {
>   ???
> }
> 
> extern "C" int analogRead(uint8_t pin) {
>   switch (pin) {
>   case A0: // 17
>     return system_adc_read();
> 
>   case 0 ... 7: // or some other unused numbers in the 0...255 range
>     return multiplexer_read(pin);
>   }
> 
>   return 0;
> }
> ```
> 
> Extend NTCSensor code to allow PIN value changes, make a setup() code to instantiate multiplexer and then change `analogRead` implementation to access the multiplexer pin. Any implementation details related to PIN switching timing would be apparent, e.g. is there a need for delayMicroseconds / delay from our side or not, etc.
> 
Nice. It's not really necessary modify AnalogSensor, extending and overriding NTCSensor is enough; anyway I like modify AnalogSensor because my idea was support multiple analog sensor in general, not only ntc sensors (MICS2710 sensor for example)

> What I meant is to integrate multiplexer system-wide, not sensor specifically. Have you seen relay and button code related to providers?
> 
No much. It is related to the class DigitalPin?
 
> * system initiates multiplexer elsewhere. api extended to provide a type of analog data source. real hw analog source is a type, multiplexer is a type.
> * sensor code made aware of multiplexer [...]

I think I understand your idea, but it seems to me that making this extension for the entire system is too complex. The problem is that there are sensors that use more than one port, but for relays, buttons, and analog sensor, yes, because they use only one port.
With this extension many more buttons, relays and analog sensors could be supported.

JavierAder, 2024-09-26:
> What I meant is to integrate multiplexer system-wide, not sensor specifically. Have you seen relay and button code related to providers?
>
> * system initiates multiplexer elsewhere. api extended to [...]

Now I think I'm understanding your idea; I didn't know about providers support. To multiplex buttons I propose two new types of providers
BUTTON_PROVIDER_GPIO_MUX = 3
BUTTON_PROVIDER_ANALOG_MUX = 4
Also, in configuration, the following keys
muxAddress0= GPIO connected to bit 0 of mux address
muxAddress1= GPIO connected to bit 1 of mux address
....
muxAddressN= GPIO connected to bit N of mux address

Then, to define a digital multiplexed button, say, number 5, you would specify the following entries in the configuration

btnProv5=3
btnGpio5= (the 'real' port; the output of the mux)
btnMuxAddress5= (the address to set  in muxAddress0, muxAddress1... muxAdressN before reading the 'real' port)
(everything else keys, the same)

Ok, but what code should be modified/extended? button.cpp?

JavierAder, 2024-09-28:
As a concrete example; using CD405xB and only 4 GPIO for support 8 buttons.
[image]

In runtime config:

```
muxAddress0= 5 //GPIO connected to Pin A of multiplexer
muxAddress1= 4 //GPIO connected to Pin B of multiplexer
muxAddress2= 0 //GPIO connected to Pin C of multiplexer

```
For BTN1
```
btnProv1=3 //BUTTON_PROVIDER_GPIO_MUX
btnGpio1= 2 //GPIO connected to the output mulltiplexer, Pin COM
btnMuxAddress1 = 0 //Multiplexer channel to which the button is connected
.....

```

For BTN2

```
btnProv2=3 //BUTTON_PROVIDER_GPIO_MUX
btnGpio2= 2 //GPIO connected to the output mulltiplexer, Pin COM,SAME of BTN1
btnMuxAddress1 = 1 //Multiplexer channel to which the button is connected
.....
```

The same for BTN3... BTN8 changing btnMuxAddressX.

mcspr, 2024-09-28:
> Now I think I'm understanding your idea; I didn't know about providers support. To multiplex buttons I propose two new types of providers
BUTTON_PROVIDER_GPIO_MUX = 3
BUTTON_PROVIDER_ANALOG_MUX = 4
Also, in configuration, the following keys
muxAddress0= GPIO connected to bit 0 of mux address
muxAddress1= GPIO connected to bit 1 of mux address
....
muxAddressN= GPIO connected to bit N of mux address

Not quite the same as e.g. After enabling MCP support flag it adds extra type for pin, but button continues to use GPIO provider and digital reads.

https://github.com/xoseperez/espurna/blob/ac01afc8f02b2525c9ce86c431f783ea7576caf3/code/espurna/config/hardware.h#L5161-L5162
https://github.com/xoseperez/espurna/blob/ac01afc8f02b2525c9ce86c431f783ea7576caf3/code/espurna/config/hardware.h#L5175-L5176

btnMuxAddress aka 'Multiplexer channel to which the hardware is connected' is btnGpio. Since the main use-case is digital access
Meaning, BUTTON config only knows about the MUX pin and only MUX config knowns about the hardware pins it controls. There is a limitation of available keywords, though, but I presumed it would be enough of a an abstraction.

```
muxType => cb450xb
muxComGpio => 2
muxGpio0 => 5
muxGpio1 => 4
muxGpio2 => 0
```
```
btnGpioType1 => cd450xb
btnGpio1 => 0
```
btnGpioType2 => cd450xb
btnGpio1 => 1
```

Analog buttons in this case also re-use the same config, changing btnProv to analog would still be able to access type and pin number which in turn would use a different proxy class to read specific MUX channel on ADC

JavierAder, 2024-09-28:
Nice.
Possible problems I see

https://github.com/xoseperez/espurna/blob/ac01afc8f02b2525c9ce86c431f783ea7576caf3/code/espurna/button.h#L20

With multiplexers the number of buttons can potentially exceed 32

https://github.com/xoseperez/espurna/blob/ac01afc8f02b2525c9ce86c431f783ea7576caf3/code/espurna/gpio.h#L85

Port conflict logic changes when there is a multiplexer

JavierAder, 2024-09-28:
On the other hand, to extend not only buttons, but also relays or LEDs (in general, to use a multiplexer as output), I think it is necessary to use not only a multiplexer, but also a buffer; this buffer is enabled using the additional port
(I'll upload a schematic later).
Say
`muxBufferGPIO= //GPIO connected to the pin ENABLE of ouput buffer`

JavierAder, 2024-09-30:
Hi. Apologies, but I've been rethinking these ideas and I think:
- for digital input or output expansion, using multiplexers/demultiplexers seems unnecessarily complex to me. It seems much more elegant and simple to do it by supporting shift registers; that has the added advantage that only 3 ports need to be used. For example
https://resources.altium.com/p/how-expand-input-and-output-microcontroller
"A more elegant solution is to use serial clocking shift registers like the 74HC595 for output and 74HC165 for input. These ICs can be cascaded to each other with the limitation being the latency to shift the bytes to all the ICs. Using shift registers only involves three I/O pins on the microcontroller, regardless of the number of ICs."

- for what I do think is almost insurmountable to use a multiplexer is to expand the number of analog inputs, which was my original problem.

Following up on this last idea, a perhaps simple way for future analog sensors to make use of multiplexing is to use a special encoding to specify their "pin" when calling analogRead(uint8_t pin):
pin=0 (0x00) analogRead works the same as now
pin=0x1pppppppp=128+ Address in the analog multiplexer
That is, the most significant bit is used to distinguish the standard analog reading from the reading using the multiplexer (ok, this is the same as you proposed before, but differentiating 0 as the normal pin; the current code uses analogRead(0) not analogRead(A0)).

To configure the analog multiplexer as a whole, the keys you proposed can be used, except muxComGpio (the analog multiplexer output must always be connected to pin A0 of the microcontroller).
Later I will try to define a sensor called NTCMuxSensor that simply extends NTCSensor by modifying the code that performs the reading to exemplify these ideas (obviously, many instances of NTCMuxSensor will be allowed).

mcspr, 2024-10-02:
> for digital input or output expansion, using multiplexers/demultiplexers seems unnecessarily complex to me. It seems much more elegant and simple to do it by supporting shift registers; that has the added advantage that only 3 ports need to be used. For example
https://resources.altium.com/p/how-expand-input-and-output-microcontroller
"A more elegant solution is to use serial clocking shift registers like the 74HC595 for output and 74HC165 for input. These ICs can be cascaded to each other with the limitation being the latency to shift the bytes to all the ICs. Using shift registers only involves three I/O pins on the microcontroller, regardless of the number of ICs."

Also true. Still, generic input support is a possibility? It does run into the case of not-the-best-tool-for-the-job, yes. Suppose, such multiplexer api can be cut out from allowing OUTPUTs, limiting pin abstraction to INPUTs only. Expander, shift registers, etc. can be allowed to support both.

> pin=0 (0x00) analogRead works the same as now
pin=0x1pppppppp=128+ Address in the analog multiplexer

Ah. So, the Arduino side [supports both by checking whether input is `pin == 0 || pin == 17 (aka A0)`](https://github.com/esp8266/Arduino/blob/ccea72823ac50290bc05c67350d2be6626e65547/cores/esp8266/core_esp8266_wiring_analog.cpp#L34). I was reading espurna analog button code at that time, where I incidentally only added the A0 check.

Note that analogRead replacement is intended for 'variant' / 'board' / 'only-works-on-this-hw' override. e.g. https://github.com/esp8266/Arduino/blob/ccea72823ac50290bc05c67350d2be6626e65547/variants/wifi_slot/analogRead.cpp#L6

I do still lean to the idea of separating mux + analog and just analog through gpio type... But, still have to think about it some more.

JavierAder, 2024-10-06:
Well, here is my first attempt. Note that I added not only support for multiple NTCs but also for multiple Emon sensors (current support for multiple emon sensors requires additional hardware more complex than a simple multiplexer). I think this would be a good general scheme to support any other type of analog sensor in which more instances are required.
Hardwired in the code, using a single multiplexer, 3 NTC sensors and 2 Emon sensors would be supported, using a single multiplexer.
Obviously, feel totally free of any type of correction or suggestion.

New variables of preprocessor:

AnalogMux_SUPPORT
NTCMuxSensor_SUPPORT
EmonAnalogMuxSensor_SUPPORT

New settings:
TODO

Files:
sensor.cpp modified
AnalogMux.h added
NTCMuxSensor.h added
EmonAnalogMuxSensor.h added 

AnalogMux.h
```
#include "espurna.h"

class AnalogMux{
    private:
    //TODO: use an array/vector for GPIOs associated with address pins of Multiplexer
    static uint8_t _muxGPIO0;
    static uint8_t _muxGPIO1;
    static uint8_t _muxGPIO2;

    public:
    static void setup(){
     //TODO
      //Using settings or defines, set GPIOs for addressing the multiplexer and configure
      // and locks pins as digital outs
      _muxGPIO0=5;
      _muxGPIO1=4;
      _muxGPIO2=0;

      
    }

    static void setAddressMux(uint8_t pin)
    {
        //TODO: decode pin as 1's and 0's, use that for set GPIOs, and then use the analogRead of system
        
        //Digital write to GPI0
        //Digital write to GPI1
        //...

        //delay before analog Read?
        //delay(?)

        
    }
};

```
NTCMuxSensor.h

```
#pragma once

#include "AnalogMux.h"
#include "NTCSensor.h"

class NTCMuxSensor : public NTCSensor {
    private:
    //Instances of all NTCMux created in setup()
    static  std::vector<NTCMuxSensor> _insts;

    public:
    static  std::vector<NTCMuxSensor> getSensors()
    {
      return _insts;
    }
    static void setup(){
     //TODO
      //Using settings or defines, set GPIOs for addressing the multiplexer and configure
      // and locks pins as digital outs
      //For now, for we create 3 ntc sensor
      NTCMuxSensor* sensor = new NTCMuxSensor();
      sensor->setAnalogPin(0);
      sensor->setSamples(1); //TODO: find values for real ntc sensors
      sensor->setDelay(0);
      sensor->setUpstreamResistor(100);
      sensor->setDownstreamResistor(100);
      sensor->setInputVoltage(1);
      sensor->setBeta(1);
      sensor->setR0(10);
      sensor->setT0(10);
      _insts.push_back(*sensor);
      sensor = new NTCMuxSensor();
      sensor->setAnalogPin(1);
      sensor->setSamples(1); //TODO: find values for real ntc sensors
      sensor->setDelay(0);
      sensor->setUpstreamResistor(100);
      sensor->setDownstreamResistor(100);
      sensor->setInputVoltage(1);
      sensor->setBeta(1);
      sensor->setR0(10);
      sensor->setT0(10);
      _insts.push_back(*sensor);
      sensor = new NTCMuxSensor();
      sensor->setAnalogPin(2);
      sensor->setSamples(1); //TODO: find values for real ntc sensors
      sensor->setDelay(0);
      sensor->setUpstreamResistor(100);
      sensor->setDownstreamResistor(100);
      sensor->setInputVoltage(1);
      sensor->setBeta(1);
      sensor->setR0(10);
      sensor->setT0(10);
      _insts.push_back(*sensor);

    }

    void setAnalogPin(uint8_t analogPin)
    {
      _analogPin=analogPin;   
    }
    
    // Descriptive name of the sensor
    String description() const override {
      //TODO, use _analogPin for description?
      return F("NTCMux @ TOUT");
    } 

    void pre() override {
      //Before read trhow A0 set the multiplexer 
      AnalogMux::setAddressMux(_analogPin);
      
      NTCSensor::pre();
    }
    protected:

    uint8_t _analogPin;  

};

```

EmonAnalogMuxSensor.h
```

#pragma once

#include "AnalogMux.h"

#include "EmonAnalogSensor.h"

class EmonAnalogMuxSensor : public EmonAnalogSensor{

    private:
    //Instances of all EmonAnalogMuxSensor created in setup()
    static  std::vector<EmonAnalogMuxSensor> _insts;

    public:
    static  std::vector<EmonAnalogMuxSensor> getSensors()
    {
      return _insts;
    }
    static void setup(){
     //TODO
      //Using settings or defines, set GPIOs for addressing the multiplexer and configure
      // and locks pins as digital outs
      //For now, for we create 2 sensor, one for current and one for voltage
      //
      EmonAnalogMuxSensor* sensor = new EmonAnalogMuxSensor();
      sensor->setAnalogPin(3); //0, 1 and 2 is used for NTC sensors exsmples
        //TODO: se values for real sensors
      _insts.push_back(*sensor);

      sensor = new EmonAnalogMuxSensor();
      sensor->setAnalogPin(4);

      _insts.push_back(*sensor);
    }

    void setAnalogPin(uint8_t analogPin)
    {
      _analogPin=analogPin;   
    }
    
    unsigned int analogRead() override {
    
         //Before read trhow A0 set the multiplexer 
        AnalogMux::setAddressMux(_analogPin);

        return EmonAnalogSensor::analogRead();
    }
 
    protected:

    uint8_t _analogPin;  

};

```

JavierAder, 2024-10-06:
Ah... 
sensor.cpp:225
```
#if AnalogMux_SUPPORT
    #include "sensors/AnalogMux.h"
#endif

#if NTCMuxSensor_SUPPORT
    #include "sensors/NTCMuxSensor.h"
#endif

#if EmonAnalogMuxSensor_SUPPORT
    #include "sensors/EmonAnalogMuxSensor.h"
#endif

```
sensor.cpp:2619
```
#if AnalogMux_SUPPORT
    {
       AnalogMux::setup();
    }
#endif

#if NTCMuxSensor_SUPPORT
    {
       NTCMuxSensor::setup();
             //Add all sensors NTC 
       std::vector<NTCMuxSensor> sensors= NTCMuxSensor::getSensors();
       for (std::size_t i = 0; i < sensors.size(); i++) {
            NTCMuxSensor sensor = sensors[i];
            add(sensor);
        }
 
    }
#endif

#if EmonAnalogMuxSensor_SUPPORT
    {
       EmonAnalogMuxSensor::setup();
        //Add all sensors NTC 
       std::vector<EmonAnalogMuxSensor> sensors= NTCMuxSensor::getSensors();
       for (std::size_t i = 0; i < sensors.size(); i++) {
            NTCMuxSensor sensor = sensors[i];
            add(sensor);
        }

    }
#endif

```

JavierAder, 2024-10-15:
Hi. a more concrete version:
-Code was added to specify the NTC sensor data in a compact way in runtime settings; only one string for sensor
-Code was added to manage the multiplexer before the analog reading through the virtual pins (setAddressBeforeReading)
-TODO: get config strings from espurna

AnalogMux.h

```
#include "espurna.h"
#include "../gpio.h"

class AnalogMux
{
private:

    using Delay = espurna::duration::critical::Microseconds;
    static Delay _delay;
    // TODO: use an array/vector for GPIOs associated with address pins of Multiplexer
    static uint8_t _muxGPIO0;
    static uint8_t _muxGPIO1;
    static uint8_t _muxGPIO2;

    static std::vector<uint8_t> gpios;

    static int _error;

public:
    static void setup()
    {
        // TODO
        // Using settings or defines, set GPIOs for addressing the multiplexer and configure
        //  and locks pins as digital outs, and set delay
        _delay = Delay{ 100 };
        _muxGPIO0 = 5;
        _muxGPIO1 = 4;
        _muxGPIO2 = 0;

        // error, TODO debug print
        _error = SENSOR_ERROR_OK;

        // locks gpio and mode
        if (!gpioLock(_muxGPIO0))
        {
            // error, TODO debug print
            _error = SENSOR_ERROR_GPIO_USED;
        }
        pinMode(_muxGPIO0, OUTPUT);
        gpios.push_back(_muxGPIO0);

        if (!gpioLock(_muxGPIO1))
        {
            // error, TODO debug print
            _error = SENSOR_ERROR_GPIO_USED;
        }
        pinMode(_muxGPIO1, OUTPUT);
        gpios.push_back(_muxGPIO1);

        if (!gpioLock(_muxGPIO2))
        {

            // error, TODO debug print
            _error = SENSOR_ERROR_GPIO_USED;
        }
        pinMode(_muxGPIO2, OUTPUT);
        gpios.push_back(_muxGPIO2);
    }

    static void setAddressMuxBeforeRead(uint8_t pin)
    {
        if (_error)
            return;

        for (int i = 0; i < gpios.size(); i++)
        {
            uint8_t gpio = gpios[i];
            int bit = pin & 0x01;
            if (bit){
                digitalWrite(gpio, HIGH);
            }else{
                digitalWrite(gpio, LOW);

            }
            pin = pin >>1;
        }
        // TODO: 
        //If pin >0 -> error

        // delay before analog Read?
        espurna::time::critical::delay(_delay);
    }
};

```

NTCMuxSensor.h
```

#pragma once

#include "AnalogMux.h"
#include "NTCSensor.h"
#include "../settings_convert.h"

class NTCMuxSensor : public NTCSensor {
    private:
    //Instances of all NTCMux created in setup()
    static  std::vector<NTCMuxSensor> _insts;
    //helper

    // Helper
    static std::vector<String> splitConfig(String str)
    {
      std::vector<String> strings;
      char separator = ',';
      int startIndex = 0, endIndex = 0;
      for (int i = 0; i <= str.length(); i++)
      {

        // If we reached the end of the word or the end of the input.
        if (str[i] == separator || i == str.length())
        {
          endIndex = i;
          String temp;
          temp = str.substring(startIndex, endIndex);
          strings.push_back(temp);
          startIndex = endIndex + 1;
        }
      }
      return strings;
    }

    public:
    static  std::vector<NTCMuxSensor> getSensors()
    {
      return _insts;
    }

    static void setup(){
     //TODO: get allConfigs from settings

      //Using settings or defines, set GPIOs for addressing the multiplexer and configure
      // and locks pins as digital outs
      //For now, for we create 3 ntc sensor
      //FORMAT: 9 items separated by commas
      //AnalogPin:int,samples:int,delay:int,upResistor:long,downResistor:long,inputVoltage:double,beta:long,r0:long,t0:long
      //Examples: 10 k upResistor, voltage 1.0, beta 3799,R0 10000,T0 298.15 (25 C in Kelvin)
      String s1="0,1,0,10000,0,1.0,3799,10000,298.15";
      String s2="1,1,0,10000,0,1.0,3799,10000,298.15";
      String s3="2,1,0,10000,0,1.0,3799,10000,298.15";
      std::vector<String> allConfigs;
      allConfigs.push_back(s1);
      allConfigs.push_back(s2);
      allConfigs.push_back(s3);

      using namespace espurna::settings::internal;

      for (auto config: allConfigs) {
        std::vector<String> dataSensor = splitConfig(config);
        if (dataSensor.size() != 9)
        {
          //error:wrong number of items in config string
          //TODO: print debug?
          continue; 
        }
        NTCMuxSensor* sensor = new NTCMuxSensor();
        //TODO:check values
        int pin = convert<int>(dataSensor[0]);
        int samples = convert<int>(dataSensor[1]);
        int delay = convert<int>(dataSensor[2]);
        long rUp = convert<long>(dataSensor[3]);
        long rDown =convert<long>(dataSensor[4]);
        double voltage = convert<double>(dataSensor[5]);
        long beta = convert<long>(dataSensor[6]);
        long r0 = convert<long>(dataSensor[7]);
        double t0 = convert<double>(dataSensor[8]);

        sensor->setAnalogPin(pin);
        sensor->setSamples(samples); 
        sensor->setDelay(delay);
        sensor->setUpstreamResistor(rUp);
        sensor->setDownstreamResistor(rDown);
        sensor->setInputVoltage(voltage);
        sensor->setBeta(beta);
        sensor->setR0(r0);
        sensor->setT0(t0);
        _insts.push_back(*sensor);
      }

    }

    //API sensor

    void setAnalogPin(uint8_t analogPin)
    {
      _analogPin=analogPin;   
    }
    
    // Descriptive name of the sensor
    String description() const override {
      //TODO, use _analogPin for description?
      return F("NTCMux @ TOUT");
    } 

    void pre() override {
      //Before read trhow A0 set the multiplexer 
      AnalogMux::setAddressMuxBeforeRead(_analogPin);
      
      NTCSensor::pre();
    }
    protected:

    uint8_t _analogPin;  

};

```

I think the code is almost usable, although I'm not sure if I'm using the APIs correctly. Any suggestion is appreciated.

JavierAder, 2024-11-05:
Ok, a running version:
[image]
In terminal:
```
set analogMux 50,14,15,16
set ntcMux1 0,1,0,10000,0,1.0,3799,10000,298.15
set ntcMux2 1,1,0,10000,0,1.0,3799,10000,298.15
set ntcMux3 2,1,0,10000,0,1.0,3799,10000,298.15
set ntcMux4 3,1,0,10000,0,1.0,3799,10000,298.15

```
config\custom.h
```

// ------------------------------------------------------------------------------
// Example file for custom.h
// Either copy and paste this file then rename removing the .example or create your
// own file: 'custom.h'
// This file allows users to create their own configurations.
// See 'code/espurna/config/general.h' for default settings.
//
// See: https://github.com/xoseperez/espurna/wiki/Software-features#enabling-features
// and 'code/platformio_override.ini.example' for more details.
// ------------------------------------------------------------------------------

//LOLIN with AnalogMux support
#if defined(NODEMCU_LOLIN_AM)

    // Info
    #define MANUFACTURER        "NODEMCU"
    #define DEVICE              "LOLIN_AM"

    // Buttons
    #define BUTTON1_PIN         0
    #define BUTTON1_CONFIG      BUTTON_PUSHBUTTON | BUTTON_DEFAULT_HIGH
    #define BUTTON1_RELAY       1

    // Hidden button will enter AP mode if dblclick and reset the device when long-long-clicked
    #define RELAY1_PIN          12
    #define RELAY1_TYPE         RELAY_TYPE_NORMAL

    // Light
    #define LED1_PIN            2
    #define LED1_PIN_INVERSE    1
	
	//AnalogMux and NTCMux
	#define SENSOR_SUPPORT     	1
	#define AnalogMux_SUPPORT 	1
	#define NTCMuxSensor_SUPPORT 1
	

#endif

```
And finally, diff

```
diff --git a/code/espurna/config/sensors.h b/code/espurna/config/sensors.h
index 727183e7..66babd7b 100644
--- a/code/espurna/config/sensors.h
+++ b/code/espurna/config/sensors.h
@@ -1510,7 +1510,8 @@
     MICS2710_SUPPORT || \
     MICS5525_SUPPORT || \
     NTC_SUPPORT || \
-    TMP3X_SUPPORT \
+    TMP3X_SUPPORT || \
+    AnalogMux_SUPPORT \
 )
 #undef ADC_MODE_VALUE
[... 436 lines cut ...]
+      sensor->setInputVoltage(voltage);
+      sensor->setBeta(beta);
+      sensor->setR0(r0);
+      sensor->setT0(t0);
+      ntcSensors.push_back(sensor);
+    }
+    return ntcSensors;
+  }
+};
+
+// std::vector<NTCMuxSensor *> *NTCMuxSensor::_ntcSensors = nullptr;
+//  STATIC METHODS
+//   Helper
diff --git a/code/platformio_override.ini b/code/platformio_override.ini
new file mode 100644
index 00000000..1b9424f3
--- /dev/null
+++ b/code/platformio_override.ini
@@ -0,0 +1,4 @@
+[env:nodemcu-lolin-analog-mux]
+extends = env:esp8266-4m-base
+build_src_flags = -DNODEMCU_LOLIN_AM -DNOWSAUTH -DUSE_CUSTOM_H
+
```
TODO: get multiples config with prefix ntcMux

JavierAder, 2024-11-28:
Well, these days I realized two things:
- the ESP8266 ADC besides having low resolution is very noisy (consecutively read temperatures could vary several degrees)
- there is a way to support both multiple analog ports and a better ADC: ADS1115. It has a 16-bit ADC and an internal multiplexer that supports up to 4 channels (I think they can also be cascaded to support even more)
https://www.ti.com/product/es-mx/ADS1115
There is already code that uses that chip (EmonADS1X15Sensor.h) although not related to NTCs.

I think using ADS1115 should be the standard way to expand analog inputs, possibly at AnalogSensor level.

mcspr, 2024-11-29:
ADC noise not related to power / specific board / lack-of stable power input?
Anything changes w/ *longer* delay between analogRead()s, not just 2ms but 100ms+, 200ms+, ...?

Does current ADS1115 work?:) I do remember tweaking read loops at some point, and non-working bitmask for old configs, but i don't really remember whether it was tested. Plus, not sure I have the part around rn

JavierAder, 2024-11-30:
> ADC noise not related to power / specific board / lack-of stable power input? Anything changes w/ _longer_ delay between analogRead()s, not just 2ms but 100ms+, 200ms+, ...?
> 
I tested it with the code where AnalogSensor calculated the averages wrong; with your latest modifications and using several samples per access it will probably generate less variations.
As for the delays, I will try them as soon as I can, but I suspect that the problem does not come from there.
> Does current ADS1115 work?:) I do remember tweaking read loops at some point, and non-working bitmask for old configs, but i don't really remember whether it was tested. Plus, not sure I have the part around rn

The code I read is related to an Emon sensor; I don't know if it works well, but accessing ADS1115 seems relatively simple.

JavierAder, 2024-11-30:
Well, this is my last idea: delegate all the analog reads to a separate class, which is independently configured (in runtime) and potentially supports several devices that generate analog reads (one of which is ESP8266 itself), each of which may have several channels/pins. On the other hand, an additional field must be added to AnalogSensor, input_device_id; When you want to do the analog reading, AnalogSensor delegates it to the new class (AnalogInputs).
I think it is the most versatile and expandable form; In the future one can support not only ADS1115, but even communication with other controllers via i2c or another protocol; and all this just by modifying a class.
espurna/analog_inputs.h:

```
#pragma once

#include "types.h"

// Device to read from
#define IN_DEVICE_ESP8266_ID 0     // standar ESP8266 analog input (pin A0)
#define IN_DEVICE_ADS1115_GND_ID 1 // ADS1115 via i2c with address 1001000 /four channels)
#define IN_DEVICE_ADS1115_VDD_ID 2 // ADS1115 via i2c with address 1001001 (four channels)
#define IN_DEVICE_ADS1115_SDA_ID 3 // ADS1115 via i2c with address 1001010 (four channels)
#define IN_DEVICE_ADS1115_SCL_ID 4 // ADS1115 via i2c with address 1001011 (four channels)

struct AnalogInputResult
{
    int raw_value;
    double voltage; // conversion form raw_value to voltage is device dependent
    bool error;     // overflow of maximum value, connection error, etc
};
// Singleton, configurated in boot time (or runtime)

struct ADS1115Config

{

    uint8_t datarate;
    uint8_t mode;
    uint8_t gain;
    bool configured;
    // TODO: delay? current channel?
};
//Singleton
class AnalogInputs
{
public:
    static AnalogInputs *createInst();

    //singleton: global instance
    static AnalogInputs *Inst();

    //hook for possible adaptation via subclasing
    static void setCustomAnalogInputs(AnalogInputs* custom)
    {
        _inst = custom;

    }
    AnalogInputs()
    {
        ads1115gnd.configured = false;
        ads1115vdd.configured = false;
        ads1115sda.configured = false;
        ads1115scl.configured = false;
    }
    // called from main with runtime config

    void setADS1115ConfigGND(ADS1115Config config)
    {
        ads1115gnd = config;
    }

    void setADS1115ConfigVDD(ADS1115Config config)
    {
        ads1115vdd = config;
    }
    void setADS1115ConfigSDA(ADS1115Config config)
    {
        ads1115sda = config;
    }
    void setADS1115ConfigSCL(ADS1115Config config)
    {
        ads1115scl = config;
    }

    // standard analog read of ESP8266
    AnalogInputResult analogRead()
    {

        return analogRead(IN_DEVICE_ESP8266_ID, 0);
    }
    AnalogInputResult analogRead(uint8_t device_id, uint8_t pin)
    {
        AnalogInputResult result;
        // TODO:Depending on the device ID, reading is forwarded to pin A0, i2c (ADS115), etc...
	//IMPORTANT: also calculate the voltage 
	//if (device_id == IN_DEVICE_ESP8266_ID)
	//	normal read via ::analogRead
	//else if (device_id == IN_DEVICE_ADS1115_GND_ID )
	//	read via i2c using pin and ads1115gnd
	//etc

        return result;
    }

protected:
    static AnalogInputs *_inst;
    ADS1115Config ads1115gnd;
    ADS1115Config ads1115vdd;
    ADS1115Config ads1115sda;
    ADS1115Config ads1115scl;
};

AnalogInputs *AnalogInputs::_inst = nullptr;

AnalogInputs *AnalogInputs::createInst()
{
    AnalogInputs::_inst = new AnalogInputs();
    return _inst;
}
AnalogInputs *AnalogInputs::Inst()
{
    return AnalogInputs::_inst;
}

AnalogInputs* _i = AnalogInputs::createInst();

```

Next week I'm going to buy an ADS1115 and I'm going to try to make a usable version.

JavierAder, 2024-12-06:
I don't understand what you mean by "the genuine part and cheaper CJ-..., GY-... , etc. variants"
My first idea was to expand Esp8266 with an analog multiplexer to allow multiple analog sensors to share the Esp8266 ADC; my current idea is to expand Esp8266 so that analog sensors can use any analog source, in particular, that they can use an external ADC (ADS1115), and not be restricted to the Esp8266 ADC (which is noisy and low resolution).
If instead of using ADS1115 you want to use an analog multiplexer and the Esp8266 ADC (as in my original idea) you can simply define a new device id, say:
IN_DEVICE_MUX_ID = 5
and add the necessary code in AnalogInputs

mcspr, 2024-12-07:
> I don't understand what you mean by "the genuine part and cheaper CJ-..., GY-... , etc. variants"

Try searching e.g. aliexpress, chips are ADS1x15 analogues

mcspr, 2024-12-07:
> If instead of using ADS1115 you want to use an analog multiplexer and the Esp8266 ADC (as in my original idea) you can simply define a new device id, say:
IN_DEVICE_MUX_ID = 5
and add the necessary code in AnalogInputs

Right, this is the implementation. Should the reader remember the id internally to pass it along to the analoginputs? In sensor class, I mean. Or terminal `adc`. Or the buttons.
There is also resolution difference btw

JavierAder, 2025-02-03:
Hello, just recently I had time to advance the code.
In summary: AnalogSensor delegates the analog readings to another class, AnalogInputs; to do the reading, you only have to pass the ID of the reading device (the same Esp8266, a generic multiplexer, or an ADS1115) and the pin/channel within it.
AnalogInputs is configured independently and at runtime.
As an example of use I added a RawAnalogSensor.
The ADS1115 access code is based on the Adafruit library.
Modified files: 
AnalogSensor.h
sensor.cpp 
Files added:
analog_inputs.h 
RawAnalogSensor.h

TODO: 
-finalize details and compile (see TODOs in code)
-modify NTCSensor to support many ntc sensors using the same scheme that RawAnalogSensor uses

JavierAder, 2025-02-03:
analog_inputs.h
```
#pragma once

#include "types.h"
#include "gpio.h"
#include "settings.h"
#include "settings_convert.h"
#include "espurna.h"
#include "i2c.h"

// Device to read from
#define IN_DEVICE_ESP8266_ID 0     // standar ESP8266 analog input (pin A0)
#define IN_DEVICE_MUX_ID 1         // generic mux using ESP8266 ADC
#define IN_DEVICE_ADS1115_GND_ID 2 // ADS1115 via i2c with address 1001000 /four channels)
#define IN_DEVICE_ADS1115_VDD_ID 3 // ADS1115 via i2c with address 1001001 (four channels)
#define IN_DEVICE_ADS1115_SDA_ID 4 // ADS1115 via i2c with address 1001010 (four channels)
#define IN_DEVICE_ADS1115_SCL_ID 5 // ADS1115 via i2c with address 1001011 (four channels)

#define ADS1115_GND_ADDRESS 0b1001000
#define ADS1115_VDD_ADDRESS 0b1001001
#define ADS1115_SDA_ADDRESS 0b1001010
#define ADS1115_SCL_ADDRESS 0b1001011

// Based in Adafruit

/*=========================================================================
    POINTER REGISTER
    -----------------------------------------------------------------------*/
#define ADS1X15_REG_POINTER_MASK (0x03)      ///< Point mask
#define ADS1X15_REG_POINTER_CONVERT (0x00)   ///< Conversion
#define ADS1X15_REG_POINTER_CONFIG (0x01)    ///< Configuration
#define ADS1X15_REG_POINTER_LOWTHRESH (0x02) ///< Low threshold
#define ADS1X15_REG_POINTER_HITHRESH (0x03)  ///< High threshold
/*=========================================================================*/

/*=========================================================================
    CONFIG REGISTER
    -----------------------------------------------------------------------*/
#define ADS1X15_REG_CONFIG_OS_MASK (0x8000) ///< OS Mask
#define ADS1X15_REG_CONFIG_OS_SINGLE \
    (0x8000) ///< Write: Set to start a single-conversion
#define ADS1X15_REG_CONFIG_OS_BUSY \
    (0x0000) ///< Read: Bit = 0 when conversion is in progress
#define ADS1X15_REG_CONFIG_OS_NOTBUSY \
    (0x8000) ///< Read: Bit = 1 when device is not performing a conversion

#define ADS1X15_REG_CONFIG_MUX_MASK (0x7000) ///< Mux Mask
#define ADS1X15_REG_CONFIG_MUX_DIFF_0_1 \
    (0x0000) ///< Differential P = AIN0, N = AIN1 (default)
#define ADS1X15_REG_CONFIG_MUX_DIFF_0_3 \
    (0x1000) ///< Differential P = AIN0, N = AIN3
#define ADS1X15_REG_CONFIG_MUX_DIFF_1_3 \
    (0x2000) ///< Differential P = AIN1, N = AIN3
#define ADS1X15_REG_CONFIG_MUX_DIFF_2_3 \
    (0x3000)                                     ///< Differential P = AIN2, N = AIN3
#define ADS1X15_REG_CONFIG_MUX_SINGLE_0 (0x4000) ///< Single-ended AIN0 100 : AINP = AIN0 and AINN = GND
#define ADS1X15_REG_CONFIG_MUX_SINGLE_1 (0x5000) ///< Single-ended AIN1 101 : AINP = AIN1 and AINN = GND
#define ADS1X15_REG_CONFIG_MUX_SINGLE_2 (0x6000) ///< Single-ended AIN2 110 : AINP = AIN2 and AINN = GND
#define ADS1X15_REG_CONFIG_MUX_SINGLE_3 (0x7000) ///< Single-ended AIN3 111 : AINP = AIN3 and AINN = GND

constexpr uint16_t MUX_BY_CHANNEL[] = {
    ADS1X15_REG_CONFIG_MUX_SINGLE_0, ///< Single-ended AIN0
    ADS1X15_REG_CONFIG_MUX_SINGLE_1, ///< Single-ended AIN1
    ADS1X15_REG_CONFIG_MUX_SINGLE_2, ///< Single-ended AIN2
    ADS1X15_REG_CONFIG_MUX_SINGLE_3  ///< Single-ended AIN3
}; ///< MUX config by channel

#define ADS1X15_REG_CONFIG_PGA_MASK (0x0E00)   ///< PGA Mask
#define ADS1X15_REG_CONFIG_PGA_6_144V (0x0000) ///< +/-6.144V range = Gain 2/3
[... 518 lines cut ...]

};

AnalogInputs *AnalogInputs::_inst = nullptr;

AnalogInputs *AnalogInputs::createInst()
{
    AnalogInputs::_inst = new AnalogInputs();
    return _inst;
}
AnalogInputs *AnalogInputs::Inst()
{
    return AnalogInputs::_inst;
}

//AnalogInputs *_i = AnalogInputs::createInst();

#ifndef __cpp_inline_variables
constexpr int AnalogInputs::RawBits8266;

constexpr double AnalogInputs::RawMin8266;
constexpr double AnalogInputs::RawMax8266;
#endif

```

JavierAder, 2025-02-03:
RawAnalogSensor.h
```
#pragma once

#include "../analog_inputs.h"
#include "../settings.h"
#include "../settings_convert.h"
#include "AnalogSensor.h"

class RawAnalogSensor : public AnalogSensor
{
public:
    RawAnalogSensor()
    {
    }

    unsigned char id() const override
    {
        return 123; // TODO
    }

    unsigned char count() const override
    {
        return 3;
    }
    // Descriptive name of the sensor
    String description() const override
    {
        return F("Raw Analog Sensor");
    }

    // Type for slot # index
    unsigned char type(unsigned char index) const override
    {
        if (index == 0)
        {
            return MAGNITUDE_COUNT;
        }

        if (index == 1)
        {
            return MAGNITUDE_VOLTAGE;
        }
        if (index == 2)
        {
            return MAGNITUDE_ANALOG;
        }

        return MAGNITUDE_NONE;
    }

    // Current value for slot # index
    double value(unsigned char index) override
    {
        if (index == 0)
        {
            return _sampledValue();
        }
        if (index == 1)
        {
            return _sampledVoltageValue();
        }
        if (index == 2)
        {
            return _sampledVoltageValue() * getMult();
        }

        return 0;
    }

    double getMult()
    {
        return _mult;
    }
    void setMult(double mult)
    {
        _mult = mult;
    }

protected:
    double _mult{1.0};
};

class RawAnalogSensorConfig
{

public:
    RawAnalogSensorConfig()
    {
    }
    std::vector<RawAnalogSensor *> getSensors()
    {
        std::vector<RawAnalogSensor *> rass;
        using namespace espurna::settings::internal;
        String configSensors;
        // FORMAT:rawAnalogSensors=analog_devide_id1,channel1,mult1;analog_devide_id2,channel2,mult2;...
        configSensors = getSetting("rawAnalogSensors");
        std::vector<String> configs = AnalogInputs::Inst()->splitConfig(configSensors, ';');

        for (auto config : configs)
        {
            std::vector<String> dataSensor = AnalogInputs::Inst()->splitConfig(config);
            if (dataSensor.size() != 3)
            {
                // error:wrong number of items in config string
                // TODO: print debug?
                continue;
            }
            int device_id = convert<int>(dataSensor[0]);
            int channel = convert<int>(dataSensor[1]);
            double mult = convert<double>(dataSensor[2]);
            RawAnalogSensor* ras = new RawAnalogSensor();
            ras->setInDevice(device_id);
            ras->setPin(channel);
            ras->setMult(mult);
            rass.push_back(ras);
        }

        return rass;
    }
};

```

JavierAder, 2025-02-03:
My current problem is what to do in

```
    // Wait for the conversion to complete
        while (!conversionCompleteADS1115(config))
        {
            ; // TODO; How long to wait before detecting a read error?
        }
```
How to implement a time out?

JavierAder, 2025-02-04:
> As a guess, `1000 / samples-per-second (datarate)` is duration-per-sample in milliseconds? For 128, 8ms. For 826, roughly 2ms. And etc.

I think so, in one shot mode the reading time is inversely proportional to the datarate plus a small additional time needed to exit sleep mode (although I would have to investigate this further).

> For non-blocking... just check the clock? Otherwise, see `blockingDelay` template and pass it `in_progress?` func as the last argument.

Would the following code work?

```
	// Wait for the conversion to complete
	const auto timeout = TimeSource::now() + getReadTimeout(config.datarate);
	//getReadTimeout returns, say, 16 mls for datarate 128, 4 mls for datarate 826
        while (!conversionCompleteADS1115(config))
        {
		
            if (TimeSource::now() > timeout)
			{
				result.error = SENSOR_ERROR_TIMEOUT;
				return result;
			}
        }
```
Although I find it not very energy efficient... Using blockingDelay is probably a better solution.

mcspr, 2025-02-04:
> ```cpp
> const auto timeout = TimeSource::now() + getReadTimeout(config.datarate);
> ...
>   if (TimeSource::now() > timeout)
> ...
> ```

If `TimeSource` is `CoreClock` and thus duration is `duration::Milliseconds` with `uint32_t` as underlying type, one should be aware that 'now() + offset' can overflow at some point. Subtraction result compared to timeout usually works better here.
https://github.com/xoseperez/espurna/blob/13667138962d8e335148f2251bca2f435cc3d2ba/code/espurna/system.h#L193-L202
(and note that `duration::` types are all unsigned underneath; delta between durations / time points is always a positive number)

JavierAder, 2025-02-06:
Nice. As for energy efficiency (the loop is continuously executing i2c reads...), I looked at blockingDelay and it seems to me that it simply ends up executing delay(); wouldn't the following code be more or less equivalent (and simpler):

```
	// Wait for the conversion to complete
	const auto timeout = TimeSource::now() + getReadTimeout(config.datarate);
	//getReadTimeout returns, say, 16 mls for datarate 128, 4 mls for datarate 826
	auto delayBetweenReads = timeout/4; 
	auto last = TimeSource::now();
        while (!conversionCompleteADS1115(config))
        {
	    auto now = TimeSource::now();
            if (now - last > timeout)
	  {
	     last = now;
	    delay(delayBetweenReads);  
	  }else
	  {
		result.error = SENSOR_ERROR_TIMEOUT;
		return result;
	  }
      }
```

mcspr, 2025-02-06:
>  I looked at blockingDelay and it seems to me that it simply ends up executing delay(); wouldn't the following code be more or less equivalent (and simpler):

Correct, but it also depends what you would prefer more - readability of the name `blockingDelay`, or readability of the variables around and before the loop and the loop itself :)

The reason for the `blockingDelay` was the first, replacing the code in several sensors (and having some 2.x.x<->3.x.x Core compatible code). But you have to watch out for public funcs in the header to be aware that they exist in the first place

JavierAder, 2025-02-07:
Okay. Incidentally, I was reading that the delay should only be used in the main loop (and not, for example, interrupt routines). Does the sensor code always run in the main loop? I think in general yes, but I have my doubts if it cannot be executed in the interrupt routine or TCP/IP stack code from the execution of commands in the debug terminal (I seem to remember that there was a command to, for example, read the ADC of Esp8266).

Btw, the previous code was wrong, I think this one would work:

 ```
   // Wait for the conversion to complete
	const auto timeout = getReadTimeout(config.datarate);
	//getReadTimeout returns, say, 16 mls for datarate 128, 4 mls for datarate 826
	auto delayBetweenChecks = timeout/4; 
	auto firstCheck = TimeSource::now();
    while (!conversionCompleteADS1115(config))
    {
	  auto now = TimeSource::now();
      if (now - firstCheck > timeout)
	  {
		 result.error = SENSOR_ERROR_TIMEOUT;
	     return result;
	  }else
	  {
	    delay(delayBetweenChecks);  
	  }
    }
```

mcspr, 2025-02-07:
Sensor methods are ok for delay()s, so are terminal commands possibly calling something with delay()s
Only worry if you expect something to happen in some other loop() func in the mean time. delay() schedules sdk / system tasks and suspends loop() at that point in code; for example, no other sensor methods would execute until the delay is done.

Both yield() and delay() are indeed made to only work within setup() & loop(). Though, calling them outside of that context would crash the board, so I'd expect that occurrence would be pretty easy to spot.

fwiw, unless you are manually writing tcp / udp servers or clients with espasynctcp or using lwip callbacks, you won't happen within networking task
another way to call something in non-delay()-able context are SystemTimer callbacks (os\_timer\_...)
interrupts are also explicit; e.g. attachInterrupt callbacks in some sensors

JavierAder, 2025-02-22:
Ok, I finally think I have a working version. I attach the diff (against development version 79fa00424d4534752a1925e66e2fc3eb87fdf63a) and the custom.h (the platformio environment to build is nodemcu-lolin-analog-inputs-raw-ntc; for some reason my platformio version stopped supporting platform-override.ini). To achieve a configuration that uses two NTC sensors on an ADS1115 (with pin direction to GND) you can run from the debug console, for example:
set ADS1115GND 100,7,2,1
set ntcMuxNumber 2
set ntcMux1 2,0,1,0,10000,0,1.0,3799,10000,298.15
set ntcMux2 2,1,1,0,10000,0,1.0,3799,10000,298.15
and reboot (see comments in code for more information about formats)
I add an image of a NodeMCU Login using a module with ADS1115 (the connection is simple, v3.3->VCC, GND->GND, SCL->SCL, SDA->SDA, GND->HDDR) and image for write to a NodeMCU board (version 3 I think)

Features:

- support multiples analog inputs using generic multiplexers or ADS1115 (up to 4); all configurations in runtime
- Added to new sensors :
- - NTCMuxSensor: similar to NTCSensor but with multiples instances support
- - RawAnalogSensor: for testing and simples sensors: returns raw value and voltage read from Analog_inputs, and a third value who is voltage multiplied by a number; useful for example to interpret the voltage read as a current (mult=resistor)

TODO:
- more test
- web support for easy configurations (Max, could you guide me in this regard?)

[firmware.zip](https://github.com/user-attachments/files/18917983/firmware.zip)

[custom.zip](https://github.com/user-attachments/files/18917979/custom.zip)

[AnalogInputs3.diff.zip](https://github.com/user-attachments/files/18917543/AnalogInputs3.diff.zip)

[image]

JavierAder, 2025-02-22:
RawAnalogSensor must have some bug; setting
set ADS1115GND 100,7,2,1
set RawAnalogSensors 2,3,2.3,10;2,4,1,10
crash the system

mcspr, 2025-02-22:
For the code, you might have easier time uploading to your forked repo instead of manually .zip'ing stuff?
i.e. `git remote add fork https://github.com/JavierAder/espurna.git` & `git push fork` to upload periodically.
(see 'Fork' button at the top of Web and follow Github documentation related to 'https://...' clone / prs / 'flow')

Any crashes can be 'decoded' by using the original .elf & stack dump as .txt, or while being attached to the serial and using pio built-in monitor w/ exception decoder filter enabled.
For exception decoder filter, `pio device monitor --echo --environment <ENV> --filter esp8266_exception_decoder`. Or, I assume it is GUI'fied somewhere in the VSCode extension monitor widget
(wiki missing this info... :/ only bug-report .yaml mentions this. besides general awareness of how esp8266/Arduino builds work)
For manual option, see https://github.com/xoseperez/espurna/blob/dev/code/scripts/decoder.py `<.elf>` `<.txt>`. Toolchain package usually located at `~/.platformio/packages/toolchain-xtensa`

WebUI is separate thing here, see code/html. Generally, .cpp pushes some key-values back in a web-socket .json object which should already exist in the .html as form inputs / spans *or* has some custom logic by registering 'key' handlers

JavierAder, 2025-02-24:
My first git repository... I guess it's ok https://github.com/JavierAder/espurna/tree/ADS1115 
As for adding web support I'm going to try to base myself on the code related to leds or relays.

JavierAder, 2025-02-28:
Ok, testing NtcMuxSensor using ADS1115 seems to be going relatively well (it doesn't read temperatures very well but I think it has more to do with the ntc parameters I'm using; everything related to i2c seems to be fine).
What's driving me crazy is because RawAnalogSensor makes the system unstable.
I don't see any big differences, except that RawAnalogSensor inherits directly from AnalogSensor and generates 3 values...
The only thing that makes me suspicious is the treatment of strings in description(); I've seen that some sensors use the F() macro, but many others give the description with what these sensors do....
It must be something simple, but I read and reread the code and I can't see why RawAnalogSensor generates errors, but NtcMuxSensor doesn't...
https://github.com/JavierAder/espurna/blob/ADS1115/code/espurna/sensors/NTCMuxSensor.h

https://github.com/JavierAder/espurna/blob/ADS1115/code/espurna/sensors/RawAnalogSensor.h

mcspr, 2025-02-28:
^ see 'decoder.py' paragraph above. no use guessing when it would tell you the code breaking line

mcspr, 2025-02-28:
any reason not to use separate configuration key-values for config? getSetting(KEY, T) already deduces T as 2nd argument and calls convert<T>(), plus you dont have to remember positional argument location and worry about the correct syntax and number of commas
and btw there are delimiter / split classes for terminal use, see libs/Delimiter.h and SplitStringView

JavierAder, 2025-03-01:
> ^ see 'decoder.py' paragraph above. no use guessing when it would tell you the code breaking line

Yes, I will try, but I don't know if system really crash, simply web interface get iresponsive (connection is lost)

> any reason not to use separate configuration key-values for config? getSetting(KEY, T) already deduces T as 2nd argument and calls convert(), plus you dont have to remember positional argument location and worry about the correct syntax and number of commas and btw there are delimiter / split classes for terminal use, see libs/Delimiter.h and SplitStringView

I think that use only one configutarion string is more easy to use, but now I"m not sure... my other concern is flash memory use (ntc needs several paremeters...).

mcspr, 2025-03-01:
> Yes, I will try, but I don't know if system really crash, simply web interface get iresponsive (connection is lost)

serial-connected device is the easiest to troubleshoot. socket-connection and 'crash' terminal command can also help, but it does not give the full stack contents
only rule is that firmware.elf with the debug info embedded **must match** the running firmware.bin for this to work correctly

if it reboots w/ a message mentioning 'wdt' though... that would be a bit harder to trace

JavierAder, 2025-03-02:
I haven't been able to debug yet, but the problem is clearly here (and it must be an error in another code)

https://github.com/JavierAder/espurna/blob/12de171f8694b8d2175a2a82625ac6b0d03ce0f9/code/espurna/sensors/RawAnalogSensor.h#L30

MAGNITUDE_VOLTAGE is problematic (why? a mystery.... but maybe the problem is in code related to web)
I made this other version of RawAnalogSensor and it works....
Now, the same code but returning MAGNITUDE_VOLTAGE in type(slot) crash the system...

```
#pragma once

#include "../analog_inputs.h"
#include "../settings.h"
#include "../settings_convert.h"
#include "AnalogSensor.h"

class RawAnalogSensor : public AnalogSensor
{
public:
    unsigned char id() const override
    {
        return SENSOR_RAW_ANALOG_ID;
    }

    unsigned char count() const override
    {
        return 1;
    }

    // is it necessary?
    void pre() override
    {
    }
    // Descriptive name of the sensor
    String description() const override
    {
        char buffer[34];
        snprintf_P(buffer, sizeof(buffer),
                   PSTR("Raw Analog Sensor Dev: %2d Pin: %2d"), _in_device, _pin);
        return String(buffer);
    }

    // Type for slot # index
    unsigned char type(unsigned char index) const override
    {
        if (index == 0)
        {
            if (_rawType == 1)
                return MAGNITUDE_COUNT;
            if (_rawType == 2)
                //return MAGNITUDE_VOLTAGE; ////TODO:THAT'S generating errors
                return MAGNITUDE_ANALOG;
            return MAGNITUDE_ANALOG;
        }

        return MAGNITUDE_NONE;
    }

    espurna::sensor::Unit units(unsigned char index) const override
    {
        if (index == 0)
        {
            if (_rawType == 1)
                return espurna::sensor::Unit::None;
            if (_rawType == 2)
                //return espurna::sensor::Unit::Volt; //TODO:THAT'S generating errors
                return espurna::sensor::Unit::None;
            return espurna::sensor::Unit::None;
        }

        return espurna::sensor::Unit::None;
    }

    // Current value for slot # index
    double value(unsigned char index) override
    {
        if (index == 0)
        {
            if (_rawType == 1)
                return _sampledValue();
            if (_rawType == 2)
                return _sampledVoltageValue();
            return _sampledVoltageValue() * getMult();
        }
        return 0;
    }

    double getMult()
    {
        return _mult;
    }
    void setMult(double mult)
    {
        _mult = mult;
    }

    int getRawType()
    {
        return _rawType;
    }
    void setRawType(uint8 rawType)
    {
        _rawType = rawType;
    }

protected:
    double _mult{1.0};
    uint8 _rawType{1};
};

class RawAnalogSensorConfig
{

public:
    RawAnalogSensorConfig()
    {
    }
    std::vector<RawAnalogSensor *> getSensors()
    {
        std::vector<RawAnalogSensor *> rawSensors;
        // TODO: comment format
        using namespace espurna::settings::internal;
        String ns = getSetting("rawAnalogNumber", "0"); // number of ntcMux in system
        DEBUG_MSG_P(PSTR("rawAnalogNumber: %s\n"), ns.c_str());
        int n = convert<int>(ns);
        if (n <= 0)
            return rawSensors;

        for (int i = 1; i <= n; i++)
        {
            RawAnalogSensor *sensor = getSensor(i);
            if (sensor == nullptr)
                continue;
            rawSensors.push_back(sensor);
        }

        return rawSensors;
    }
    RawAnalogSensor *getSensor(uint8 i)
    {
        using namespace espurna::settings::internal;

        RawAnalogSensor *sensor = new RawAnalogSensor();
        // TODO:check values
        String prefix = "raw" + String(i);
        int in_device = convert<int>(getSetting(prefix + "_dev_id", "-1"));
        if (in_device < 0)
            return nullptr;
        int pin = convert<int>(getSetting(prefix + "_pin", "-1"));
        int samples = convert<int>(getSetting(prefix + "_samples", "1"));
        int delay = convert<int>(getSetting(prefix + "_delay", "200"));
        double mult = convert<double>(getSetting(prefix + "_mult", "1.0"));
        int type = convert<double>(getSetting(prefix + "_type", "1"));

        sensor->setInDevice(in_device);
        sensor->setPin(pin);
        sensor->setSamples(samples);
        sensor->setDelay(delay);
        sensor->setMult(mult);
        sensor->setRawType(type);

        return sensor;
    }
};

```

mcspr, 2025-03-02:
Probably this one
https://github.com/xoseperez/espurna/blob/13667138962d8e335148f2251bca2f435cc3d2ba/code/espurna/sensor.cpp#L3410-L3417

`isEmon(magnitude.sensor)` missing. `->getRatio` at best is harmless, at worst - crashes the board b/c it calls something else when referencing 'sensor' class vtable

(there is also `isAnalogEmon(magnitude.sensor)` for various other stuff... but I'd rather untangle analog bases first. hopefully, with less inheritance)