xoseperez/espurna #2625: Incorrect sampling in AnalogSensor.h

JavierAder, 2024-11-07:
### Device

Lolin

### Version

1.18

### Bug description

Hi. I suppose the idea of ​​the samples variable is to achieve some kind of average between several measurements as the final result.
https://github.com/xoseperez/espurna/blob/0976bdba1030f9c26d13c4660ab79b40e4137288/code/espurna/sensors/AnalogSensor.h#L122
I think that a way to approximate the average (without fear of generating overflows) can be
```

            unsigned int last { 0 };
            unsigned int result { 0 };
            for (size_t sample = 0; sample < samples; ++sample) {
                const auto value = ::analogRead(pin);
		if (sample == 0)
		   last = value;
		 //cumulative average plus weighted difference  
                result = last + (value - last)/(sample+1);
                last = result;
                if (sample > 0) {
                    espurna::time::critical::delay(delay);
                    yield();
                }
            }

            return result;

```

Surely there are other ways to approximate the average, that's the one that occurred to me. I think it is correct, or at least it does not generate crazy results.

mcspr, 2024-11-07:
Apparently, c22b369c865aad8df1f78e438351bca4f0b43b0e changed a lot of neighbouring sensors.

Commit above does something different; instead of waiting for the sample(s) immediately, value is cached to satisfy the delayed reading timing and is sampled in tick() instead of pre() / value()
Only difference is how close in time the actual value is to the reading routine... but, I would guess it is already controlled by the reading time, or it could simply continue sampling and reset `_sample` num to zero w/ some delay in-between

JavierAder, 2024-11-08:
> Apparently, [c22b369](https://github.com/xoseperez/espurna/commit/c22b369c865aad8df1f78e438351bca4f0b43b0e) changed a lot of neighbouring sensors.
> 
> Commit above does something different; instead of waiting for the sample(s) immediately, value is cached to satisfy the delayed reading timing and is sampled in tick() instead of pre() / value() Only difference is how close in time the actual value is to the reading routine... but, I would guess it is already controlled by the reading time, or it could simply continue sampling and reset `_sample` num to zero w/ some delay in-between

But the problem I see isn't about the reading timing, but rather the way it calculates the average.

mcspr, 2024-11-09:
Commit above i.e. commit a1ffa9f, average is calculated as sum divided by sample total. Weights can be added, if necessary

I also meant to point out time spent in yield() compared to time it can spend in main task.
```cpp
espurna::time::critical::delay(delay); // cpu spin, nothing happens
yield(); // task yield to scheduler, wait sdk, resume here
```

JavierAder, 2024-11-11:
> Commit above i.e. commit [a1ffa9f](https://github.com/xoseperez/espurna/commit/a1ffa9f47cc1faeff2ea1d53e3bb61a8219951ed), average is calculated as sum divided by sample total. Weights can be added, if necessary
> 
> I also meant to point out time spent in yield() compared to time it can spend in main task.
> 
> ```c++
> espurna::time::critical::delay(delay); // cpu spin, nothing happens
> yield(); // task yield to scheduler, wait sdk, resume here
> ```
ah, sorry, I hadn't seen your last commit... You are reading the sensor in tick(). 
Does this method have timing restrictions?
I'm asking you about this because of the use I'm trying to make of analog multiplexers; Multiplexers require some delay between setting the address and reading it, and this delay has to occur every time analogRead() is called.

Btw, I built a version of my code with your modifications; I'll test it as soon as I can.

mcspr, 2024-11-17:
> Does this method have timing restrictions?

Sensor::tick() is called every Arduino loop(), so it can be treated similarly.
(no clever coroutine code, just two tasks. for now :)

The general idea was caching value for value(). Actually reading this cached value has to fit somewhere between sensor module readings which call pre() and value()