xoseperez/espurna #2626: Schedules - UI should warn when one or more schedules have incorrect syntaxes

brunetton, 2024-11-07:
### Device

itead-sonoff-dual-r2

### Version

1.18.0-gita518080a+github240830

### Bug description

I was very happy to see that #2417 have been implemented :partying_face: 

Today it's the day when I have to review my automatic door schedules. So I read the doc and tried to use combinations of months and `SUNSET` and `SUNRISE` keywords.

But I've absolutely **no idea** if my schedules strings are correct or not. The only way to have an answer is to wait for ... months :s

I presume that the string is parsed internally and if there are errors the information is available somewhere ? 

It would be really useful to let user be aware of errors on save. At least instead of the "Changes saved" alert, it would be great to have "At least one schedule have an error". Then user can search and find the error.

Thanks !

mcspr, 2024-11-07:
> I presume that the string is parsed internally and if there are errors the information is available somewhere ?

Internally, yes, but only as a yes / no answer. Can point to the specific space-separated chunk, but would have to update the returned struct in that case to replace `bool ok` with e.g. pointer to the last parsed chunk and some kind of friendly error string
https://github.com/xoseperez/espurna/blob/79fa00424d4534752a1925e66e2fc3eb87fdf63a/code/espurna/scheduler_time.re#L973-L977
https://github.com/xoseperez/espurna/blob/79fa00424d4534752a1925e66e2fc3eb87fdf63a/code/espurna/scheduler_time.re#L1016-L1036

Otherwise, as a temporary workaround, systemd-analyze calendar ... would do the trick. Obviously, excluding the extensions added here
https://www.freedesktop.org/software/systemd/man/latest/systemd-analyze.html

> 1.18.0-gita518080a+github240830

As mentioned in the previous issue - you'd have to build it locally first with this option enabled
https://github.com/xoseperez/espurna/blob/79fa00424d4534752a1925e66e2fc3eb87fdf63a/code/espurna/config/general.h#L1419

brunetton, 2024-11-08:
> I presume that the string is parsed internally and if there are errors the information is available somewhere ?
>
> > Internally, yes, but only as a yes / no answer.

This would be perfect ! I could have noticed by myself that `SUNRISE` is not valid with the default build, without having to wait until this morning to find that the shutter hadn't opened :s

> As mentioned in the previous issue - you'd have to build it locally first with this option enabled

I guessed that this morning when I seen that the scheduler wasn't working. But I thought the source of truth was in the doc: https://github.com/xoseperez/espurna/wiki/Scheduler "Build scheduler module, **1 by default** (0 to disable)"

Usually (I mean in all firmwares and softwares I've seen) the assets are built with the default provided configuration. And users who wants to activate/deactivate some functionalities can do it in their own copies of the configuration.

mcspr, 2024-11-08:
> I guessed that this morning when I seen that the scheduler wasn't working. But I thought the source of truth was in the doc: https://github.com/xoseperez/espurna/wiki/Scheduler "Build scheduler module, 1 by default (0 to disable)"

Looks like the same issue as NTP one, wrong column used for runtime and build settings? Adjusted rn

> Usually (I mean in all firmwares and softwares I've seen) the assets are built with the default provided configuration. And users who wants to activate/deactivate some functionalities can do it in their own copies of the configuration.

True, but .bin are provided as a convenience for existing devices with sort-of default fits-all config. I hope it is understandable that any code added is not free, and takes more and more space, the more things are added.

I would (very) strongly suggest to take a look at PIO (CLI or through VSCode extension) and just build locally. This also simplifies configuration on per-site basis by a lot

mcspr, 2024-12-07:
^ chain of commits above, plus c8ebbbd6975e2eb851e3f2f26344f8448b649750 to cancel validity

Experimenting with possible checks & notifications.
- `schedule check ...` terminal call for manual checks
- debug log when settings reload (terminal `reload`, webui 'save' button)
- webui notification pointing at the input elem

But, unless I am missing some obvious web api call, js cannot display any pre-localized string from the built-in set of translations. e.g. I cannot tell it to say 'value is invalid' in UI language. Which is pretty annoying, considering every other validity message is translated

brunetton, 2024-12-07:
Sound great :) Thank you, I'll play with it when I've time !

closed 2025-01-16 by mcspr