xoseperez/espurna #2628: UI - Schedules - schedules types are not correctly displayed by UI

brunetton, 2024-11-09:
### Device

itead-sonoff-dual-r2

### Version

1.18.0-gita518080a+github240830

### Bug description

After changing event type (Disabled, Calendar, Relative) and come back again to the Schedule page, events types are all displayed as "Disabled"

### Steps to reproduce

- goto Schedule page
- change one or more schedules types to "Relative" 
- save changes
- reload UI (ctrl-R on browser)
- goto Schedule page again
- => all Schedules are displayed as "Disabled"

We can confirm that it's only a display problem using terminal. Using `keys` command we can check, for relative schedules:

```
> schType0 => "3"
> schType1 => "3"
```

That corresponds to definitions in `code/espurna/scheduler.cpp`:

```c
enum class Type : int {
    Unknown = 0,
    Disabled,
    Calendar,
    Relative,
};

```

### Build tools used

PlatformIO

mcspr, 2024-12-07:
The issue was that enumerations were sent as strings, ref. `schedule 0` output. Internally, string input is converted back to enumeration. Not the other way around, though

`IndexedSettings` here is key-prefix and a function to load key-prefix + index from settings and serialize it. Replacing output for webui lists may be helpful here, as every other place would prefer the string representation
https://github.com/xoseperez/espurna/blob/fa5af86d0e8a9dea3eaeca3a3682428c80562904/code/espurna/scheduler.cpp#L1533-L1539

Another change is UI now receives these as type pairs '[[number, string], [number, string], ...]' on load, so .html does not have to embed both description (at the top of the page) and the keys themselves (in the select / option)

mcspr, 2025-01-16:
Resolved in v1.19

closed 2025-01-16 by mcspr