xoseperez/espurna #2630: How to get SonOff T1 into AP mode

Ivasar, 2024-11-15:
### Device

T1 EU 2 gang

### Version

not sure

### Question

Hello,
I am trying to get the device working again, after having lost connection to it. I am uncertain about the firmware version installed because I cannot get into it. Would love not having to disconnect it and reflash it because it's in a rather inconvenient location.
How do you get these devices back into AP mode? Or, as least, back into "factory reset" (so it offers the original AP network again, with the fibonacci password).
I have tried to look for a solution and found nothing, but apologies if I failed in this search too :-)
Kind regards and thanks!

mcspr, 2024-11-17:
> How do you get these devices back into AP mode? Or, as least, back into "factory reset" (so it offers the original AP network again, with the fibonacci password).

If you do remember the device password, you can turn off the network it usually connects to. When not being able to connect, AP is kept up in-between reconnection attempts.
If the "1st" button is working, have you tried holding it for at least 10sec? [Just make sure not to power it off an on with it held, or it would be stuck in "boot" mode](https://github.com/xoseperez/espurna/wiki/Hardware-Itead-Sonoff-T1). It should do the 'factory reset' 
 and clear the network settings.

Ivasar, 2024-11-17:
Thanks for the reply.
This one never seems to offer the "local" wifi network, never goes into AP mode. It is not connected to the home's WiFi, and yet never goes into AP mode.
I have tried everything, and it seems to work correclty besides the connectivity: both buttons work, they beep if pressed 10 seconds... but can't access the device, no matter how I try.

mcspr, 2024-11-17:
The default LED mode can at least tell us what the device thinks. If it is blinking rapidly every ~1sec, it is running the AP. If it is ~5sec of OFF then briefly on, it is connected.
Also watch out that AP is 2.4GHz and running WPA2+TKIP, which *may* be a problem if client thinks it is not good enough to connect to. 

Beep is from the built-in RF module (not controlled in sw). 1st beep to notify its recording, 2nd when its recorded. I don't have the device readily at hand, but will try to have it checked for button press actions <-> actual button presses. Probably something I am forgetting from 6y ago

Ivasar, 2024-12-27:
Just realized I had not come back here to share what my problem was, and how it was solved.

Looks like my new router and wifi extender were messing with the connection: the sonoff was somehow connected somewhere, and that's why I believe it never went into AP mode. As soon as I disabled 5Ghz. wifi in the new router, not only this device but also some others started working properly again.

Thanks for the help, and hopefully this helps some other lost soul like me at some point! :-)

Ivasar, 2024-12-27:
Missed in the previous message: problem fixed!

closed 2024-12-27 by Ivasar