xoseperez/espurna #2632: Alexa

tonymerce, 2025-01-09:
### Device

Sonoff T1 2ch-Sonoff T1 3ch

### Version

1.18.0

### Bug description

Estoy actualizando el software de los dispositivos sonoff porque ahora tengo un Alexa (Echo Spot) y quería ponerlos a funcionar con alexa. El caso es que descubre los dispositivos pero en los que son dobles o triples solo funciona uno de ellos, generalmente el último, es decir si tengo un dispositivo con 2 canales, alexa solo enciende el último canal. Pregunto porque no he encontrado mucha información al respecto aquí con la última versión. ¿ Tengo que hacer algo adicional ? ¿ Seguir algún procedimiento ? Gracias por anticipado.

mcspr, 2025-01-10:
>  I ask because I have not found much information about it here with the latest version. do I have to do something additional ? follow some procedure ? Thanks in advance.

Are you building or using .bin from releases?
Can you try to build it yourself and see if it works?

We are not supporting Alexa directly, but using an external lib. 2024 alexa apparently has some issues with device detection. No idea if related, though. 1.18.0 .bin built in august is using 3.4, builds past that should use 3.4.1 w/ the patch included
https://github.com/vintlabs/fauxmoESP/issues/259

tonymerce, 2025-01-11:
Estoy usando un .bin de la versión 1.18.0, como accedo a las últimas compilaciones posteriores a noviembre?

mcspr, 2025-01-11:
Right now, only building yourself. Or waiting some time this week for 1.19

tonymerce, 2025-01-14:
Thanks, I will wait to the 1.19 version and I will to test.

mcspr, 2025-01-14:
> Thanks, I will wait to the 1.19 version and I will to test.

Done. Let me know if it solves the issue

tonymerce, 2025-01-21:
Hi, I have installed version 1.19.0 on 33 devices and alexa discovers them without problems and works ok. They work on both alexa and Apple's Homekit.
Updated devices:
Sonoff Touch
Sonoff T1 1 ch
Sonoff T1 2 ch
Sonoff T1 3 ch
Sonoff S20
Sonoff Mini
Thank you and best regards

closed 2025-01-21 by mcspr