xoseperez/espurna #2633: DS18b20 CRC Error in Espurna 1.18

limecat-ss, 2025-01-10:
### Device

Wemos D1 Mini

### Version

ESPURNA 1.18.0

### Bug description

I'm experiencing issues with Espurna 1.18 when using a DS18B20 sensor. I constantly receive a CRC Error warning. This does not happen with Espurna 1.14 using the same hardware.

### Build tools used

[env:wemos-d1]
extends = env:esp8266-4m-base
board = d1_mini
build_src_flags = -DWEMOS_D1_MINI -DNOWSAUTH
                            -DDALLAS_SUPPORT=1 
                            -DDALLAS_PIN=14
                            -DDALLAS_PARASITE=0

### Any relevant log output (when available)

```shell
[149337] [MQTT] Sending bp-ala06-b/temperature => 24.31 (PID 1)
[169355] [MQTT] Sending bp-ala06-b/temperature => 24.31 (PID 1)
[189372] [SENSOR] Could not read from DS18B20 @ GPIO4 - CRC / Data Error
[209384] [SENSOR] Could not read from DS18B20 @ GPIO4 - CRC / Data Error
[229400] [SENSOR] Could not read from DS18B20 @ GPIO4 - CRC / Data Error
[249413] [MQTT] Sending bp-ala06-b/temperature => 24.31 (PID 1)
[269425] [MQTT] Sending bp-ala06-b/temperature => 24.31 (PID 1)
[289439] [SENSOR] Could not read from DS18B20 @ GPIO4 - CRC / Data Error
[309452] [SENSOR] Could not read from DS18B20 @ GPIO4 - CRC / Data Error
[329466] [SENSOR] Could not read from DS18B20 @ GPIO4 - CRC / Data Error
[349481] [MQTT] Sending bp-ala06-b/temperature => 24.31 (PID 1)
[369492] [SENSOR] Could not read from DS18B20 @ GPIO4 - CRC / Data Error
```

mcspr, 2025-01-10:
Do you also have `DS_PARASITE` (currently, `DALLAS_PARASITE`) set to 0 in 1.14.1 build?
https://github.com/xoseperez/espurna/blob/1.14.1/code/espurna/sensors/DallasSensor.h

iirc my time with dallas, I'd check connection / connection opts first

limecat-ss, 2025-01-10:
> Do you also have `DS_PARASITE` (currently, `DALLAS_PARASITE`) set to 0 in 1.14.1 build? https://github.com/xoseperez/espurna/blob/1.14.1/code/espurna/sensors/DallasSensor.h
> 
> iirc my time with dallas, I'd check connection / connection opts first

Yes, I also used DALLAS PARASITE set to 0 on Espurna 1.14.1. Since I'm using the same hardware (same Wemos, same sensor, same connections), I didn't think it could be related to connections...

mcspr, 2025-01-10:
> Yes, I also used DALLAS PARASITE set to 0 on Espurna 1.14.1. Since I'm using the same hardware (same Wemos, same sensor, same connections), I didn't think it could be related to connections...

Just checking. Let me check the code then, it worked at least one time based on the log

mcspr, 2025-01-10:
Same build settings as above. Data pin <-> D5 (GPIO14), GND and 3v3 from the D1MINI pins; 4.2k resistor between Dallas VDD and DATA.
```
(`set w1Debug 1` & `reset` for W1 logs; `set w1Debug 0` / `del w1Debug` afterwards to remove the noise)

...
[000231] [DALLAS] 28FFB3CB6D224F06 is port handler
[000232] [W1] Reset (Present)
[000238] [W1] Selected 28FFB3CB6D224F06
[000238] [W1] -> BE
[000243] [W1] <- 850155007FFF0C100C
[000245] [W1] Reset (Present)
[000245] [DALLAS] Conversion time is 750 (ms)
[000246] [W1] Reset (Present)
[000248] [W1] ROM skip
[000250] [W1] -> 44
[000258] [SENSOR] Finished initialization for 1 sensor(s) and 1 magnitude(s)
...
[234934] [W1] Reset (Present)
[234934] [W1] ROM skip
[234935] [W1] -> 44
[235707] [W1] Reset (Present)
[235712] [W1] Selected 28FFB3CB6D224F06
[235713] [W1] -> BE
[235718] [W1] <- 7D0155007FFF0C1001
[235719] [W1] Reset (Present)
[240956] [W1] Reset (Present)
[240957] [W1] ROM skip
[240958] [W1] -> 44
[240961] [MQTT] Sending test/ESPURNA-4A2914/temperature => 23.75 (PID 1)
[241722] [W1] Reset (Present)
[241728] [W1] Selected 28FFB3CB6D224F06
[241729] [W1] -> BE
[241734] [W1] <- 7D0155007FFF0C1001
[241735] [W1] Reset (Present)
[246966] [W1] Reset (Present)
[246967] [W1] ROM skip
[246968] [W1] -> 44
[247736] [W1] Reset (Present)
[247742] [W1] Selected 28FFB3CB6D224F06
[247743] [W1] -> BE
[247748] [W1] <- 7D0155007FFF0C1001
[247749] [W1] Reset (Present)
```

No timeouts / CRC errors to reproduce :shrug:

At least one sensor initially returned all zeroes causing CRC, though, but only because of accidentally setting GPIO14 as relay pin on boot (oops)

limecat-ss, 2025-01-11:
I built from version 21ee2d5 and didn't encounter any CRC errors (tested for over an hour). I think it's resolved. I'm closing this issue.

Thanks for fixing this! Really appreciate it. Great job!

closed 2025-01-11 by limecat-ss

mcspr, 2025-01-14:
Tnx for testing!

I am still not sure what exactly made the difference, though. As a guess - restricting communication without anything on the wire generating a presence pulse after reset. Even before the 21ee2d5 it works pretty reliably with the batch of sensors I got back in march. I'd suspect there is a timing difference, w1 logs might show something
(and note that sensors I got are not maxim / dallas -made, but various clones sold on aliexpress by various vendors)