xoseperez/espurna #2638: Incorrect temperature value below 0 C with DHT22 sensor

djsomi, 2025-02-19:
### Device

Sonoff POW R2

### Version

ESPURNA 1.19.0 built 2025-02-18 20:05:14

### Bug description

I get invalid data from DHT22 if temperature under 0 C.  

404.6°C shows when temperature is -4.6°C.

What can be the problem, and how to fix it?

### Build tools used

PlatformIO, VScode

### Any relevant log output (when available)

```shell

```

### Decoded stack trace (when available)

```markdown

```

mcspr, 2025-02-20:
Looks like the problem is with the number representation (encoding) in memory. Current code expects one thing, while the device returns something completely different when number is negative.

Some references
https://github.com/adafruit/DHT-sensor-library/issues/182
https://github.com/arendst/Tasmota/issues/14165
https://github.com/arendst/Tasmota/pull/14173
https://en.wikipedia.org/wiki/Signed_number_representations#Sign%E2%80%93magnitude

Sensor code has to
a) prefer s16 in initial conversion
b) check hardware temperature limits for the sake of sanity
c) detect sign-magnitude and swap bits for dht22s like yours in this block
https://github.com/xoseperez/espurna/blob/13667138962d8e335148f2251bca2f435cc3d2ba/code/espurna/sensors/DHTSensor.h#L283-L289

closed 2025-03-05 by mcspr

djsomi, 2025-03-07:
Thank you, Tested, now its working fine!

commit 305918d5 (2025-03-05): sns(dht): different number representations for dht22

sometimes sign-magnitude, sometimes twos-complement
resolve #2638